Joshua-Riek/ubuntu-rockchip #105: pulse audio magically stopped working

develperbayman, 2023-04-08:
opi5 went to sleep woke up and cant seem to get the pulse audio service to restart even after reboot 

developerbayman@developerbaymans-box:~$ systemctl --user enable pulseaudio
Created symlink /home/developerbayman/.config/systemd/user/default.target.wants/pulseaudio.service → /usr/lib/systemd/user/pulseaudio.service.
Created symlink /home/developerbayman/.config/systemd/user/sockets.target.wants/pulseaudio.socket → /usr/lib/systemd/user/pulseaudio.socket.
developerbayman@developerbaymans-box:~$ systemctl --user start pulseaudio
Job for pulseaudio.service failed because the control process exited with error code.
See "systemctl --user status pulseaudio.service" and "journalctl --user -xeu pulseaudio.service" for details.
developerbayman@developerbaymans-box:~$ ^C
developerbayman@developerbaymans-box:~$ systemctl --user status pulseaudio.service
× pulseaudio.service - Sound Service
     Loaded: loaded (/usr/lib/systemd/user/pulseaudio.service; enabled; vendor >
    Drop-In: /usr/lib/systemd/user/pulseaudio.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Sat 2023-04-08 12:42:39 EDT; 19s >
TriggeredBy: × pulseaudio.socket
    Process: 4528 ExecStart=/usr/bin/pulseaudio --daemonize=no --log-target=jou>
   Main PID: 4528 (code=exited, status=1/FAILURE)

Apr 08 12:42:38 developerbaymans-box systemd[1969]: pulseaudio.service: Main pr>
Apr 08 12:42:38 developerbaymans-box systemd[1969]: pulseaudio.service: Failed >
Apr 08 12:42:38 developerbaymans-box systemd[1969]: Failed to start Sound Servi>
Apr 08 12:42:39 developerbaymans-box systemd[1969]: pulseaudio.service: Schedul>
Apr 08 12:42:39 developerbaymans-box systemd[1969]: Stopped Sound Service.
Apr 08 12:42:39 developerbaymans-box systemd[1969]: pulseaudio.service: Start r>
Apr 08 12:42:39 developerbaymans-box systemd[1969]: pulseaudio.service: Failed >
Apr 08 12:42:39 developerbaymans-box systemd[1969]: Failed to start Sound Servi>
lines 1-17/17 (END)...skipping...
× pulseaudio.service - Sound Service
     Loaded: loaded (/usr/lib/systemd/user/pulseaudio.service; enabled; vendor preset: enabled)
    Drop-In: /usr/lib/systemd/user/pulseaudio.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Sat 2023-04-08 12:42:39 EDT; 19s ago
TriggeredBy: × pulseaudio.socket
    Process: 4528 ExecStart=/usr/bin/pulseaudio --daemonize=no --log-target=journal (code=exited, status=1/FAILURE)
   Main PID: 4528 (code=exited, status=1/FAILURE)

Apr 08 12:42:38 developerbaymans-box systemd[1969]: pulseaudio.service: Main process exited, code=exited, status=1/FAILURE
Apr 08 12:42:38 developerbaymans-box systemd[1969]: pulseaudio.service: Failed with result 'exit-code'.
Apr 08 12:42:38 developerbaymans-box systemd[1969]: Failed to start Sound Service.
Apr 08 12:42:39 developerbaymans-box systemd[1969]: pulseaudio.service: Scheduled restart job, restart counter is at 5.
Apr 08 12:42:39 developerbaymans-box systemd[1969]: Stopped Sound Service.
Apr 08 12:42:39 developerbaymans-box systemd[1969]: pulseaudio.service: Start request repeated too quickly.
Apr 08 12:42:39 developerbaymans-box systemd[1969]: pulseaudio.service: Failed with result 'exit-code'.
Apr 08 12:42:39 developerbaymans-box systemd[1969]: Failed to start Sound Service.
~
~
~

Joshua-Riek, 2023-04-08:
Did you recently update your packages with apt upgrade? There is another issue I'm investigating now in #104

Joshua-Riek, 2023-04-08:
Try this out and it should fix any issue https://github.com/Joshua-Riek/ubuntu-orange-pi5/issues/104#issuecomment-1500933738

closed 2023-04-08 by Joshua-Riek