Joshua-Riek/ubuntu-rockchip #106: Changing root partition to btrfs

pehala, 2023-04-09:
I want to migrate the root partition to btrfs. To do that, I suspect I will need to edit booting info (apart from /etc/fstab), and since OPI 5 is using uboot, I am kind of lost about where I should look to edit the root partition for boot. Could you please point me in the right direction?

Joshua-Riek, 2023-04-11:
It may be difficult right now to do this. I have u-boot read the initrd and kernel from the root filesystem. I doubt u-boot has support for the btrfs filesystem.

Typically operating systems on the arm architecture put all the boot-related files on the first partition (typically fat32). If I can make a few more changes and make this the case, it should be easy to change the root partition to btrfs. I have some time tonight and might be able to make the required changes.

td512, 2023-04-21:
@Joshua-Riek I would probably do what the raspberry pi foundation did with `rpikernelhack` and create a deb that redirects things like the kernel to another partition, and store the real kernel/initrd there. from there you can have uboot read the kernel and have the root fs being btrfs

From the [Raspberry Pi StackExchange](https://raspberrypi.stackexchange.com/a/94827):

> "rpikernelhack" is a fake package name and a directory name used as part of a hack (in the sense of a dirty but expedient solution to a problem) to work around the fact that the Raspberry Pi foundation decided to make /boot a fat32 partition and dpkg does not get on well with fat32. I was the one who initially came up with the idea, though it was refined later by others.
>
> dpkg will install new files onto a fat32 partition (spewing some warnings along the way), but if it tries to update an existing file on a fat32 partition it will fail (iirc it tries to make a backup of the old file by creating a hardlink and fat32 doesn't support hardlinks).
>
> When people (including me) started trying to make deb packages of Pi kernels and firmware they ran into this problem, a package would install initially but trying to upgrade it would fail, ouch.
>
> My workaround was to (ab)use the "diversion" feature in dpkg. This feature was intended to allow files to be diverted so they could be replaced with either locally modified versions or versions from another package, but I was able to use it from the maintainer scripts in such a way that dpkg would perform its installation tasks on a Linux partition and then move the file to its final location at the end.
>
> Diversions require you to either specify a "package name" or "local". If you specify a package name then the diversion will affect files owned by all packages except the one you specify (the intent here is to allow a package to divert a file owned by another package and then install its own versions). I also needed a directory to divert the files to.
>
> Using the name of the kernel package being installed would have rendered the hack ineffective. Using "local" also seemed wrong, since that is supposed to be reserved for use by the local sysadmin. So I needed a fake package name that was unlikely to conflict with anything. I came up with "rpikernelhack", I also used this same string for the directory name.

You could call it something like `opikernelhack` and have an apt repo hosted on GitHub and added to `sources.list.d`

Joshua-Riek, 2023-04-21:
I have something similar in the latest release using the flash-kernel package. The kernel and initrd are installed on the rootfs and whenever the kernel or initrd are updated the system will copy the files to the boot partition.