Joshua-Riek/ubuntu-rockchip #108: Building fails

EvilOlaf, 2023-04-10:


Joshua-Riek, 2023-04-10:
The `Re-reading the partition table failed.: Invalid argument` should not be a problem. I manually call partprobe to re-read the partition table, then use a timer to wait and ensure loopdev creation.

I see the mistake, forgot to export the BOARD variable. Fixed with commit 180343421f2e522c647c66ca1275b43d7c6f51dc.

EvilOlaf, 2023-04-10:
Aight, will try again.

EvilOlaf, 2023-04-10:
Different issue now. For some reason wiringpi now fails
```
[...]
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for initramfs-tools (0.140ubuntu13.1) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Selecting previously unselected package linux-headers-5.10.110-rockchip-rk3588.
(Reading database ... 37054 files and directories currently installed.)
Preparing to unpack .../linux-headers-5.10.110-rockchip-rk3588_5.10.110-1_arm64.deb ...
Unpacking linux-headers-5.10.110-rockchip-rk3588 (5.10.110-1) ...
Selecting previously unselected package linux-image-5.10.110-rockchip-rk3588.
Preparing to unpack .../linux-image-5.10.110-rockchip-rk3588_5.10.110-1_arm64.deb ...
Unpacking linux-image-5.10.110-rockchip-rk3588 (5.10.110-1) ...
Selecting previously unselected package linux-dtb-5.10.110-rockchip-rk3588.
Preparing to unpack .../linux-dtb-5.10.110-rockchip-rk3588_5.10.110-1_arm64.deb ...
Unpacking linux-dtb-5.10.110-rockchip-rk3588 (5.10.110-1) ...
Setting up linux-headers-5.10.110-rockchip-rk3588 (5.10.110-1) ...
Setting up linux-image-5.10.110-rockchip-rk3588 (5.10.110-1) ...
update-initramfs: Generating /boot/initrd.img-5.10.110-rockchip-rk3588
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8761a_config.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723ds_config.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723ds_fw.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723bs_config.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723b_config.bin for built-in driver btrtl
Setting up linux-dtb-5.10.110-rockchip-rk3588 (5.10.110-1) ...
update-initramfs: Generating /boot/initrd.img-5.10.110-rockchip-rk3588
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8761a_config.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723ds_config.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723ds_fw.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723bs_config.bin for built-in driver btrtl
W: Possible missing firmware /lib/firmware/rtl_bt/rtl8723b_config.bin for built-in driver btrtl
2097152+0 records in
2097152+0 records out
2147483648 bytes (2.1 GB, 2.0 GiB) copied, 7.26107 s, 296 MB/s
Setting up swapspace version 1, size = 2 GiB (2147479552 bytes)
no label, UUID=3f9cf5e3-72a1-42ef-9b3a-f296fbb650bd
dpkg-deb: building package 'orangepi-firmware' in 'orangepi-firmware_20230217.git9e0d07cc_all.deb'.
Selecting previously unselected package orangepi-firmware.
(Reading database ... 62331 files and directories currently installed.)
Preparing to unpack .../orangepi-firmware_20230217.git9e0d07cc_all.deb ...
Unpacking orangepi-firmware (20230217.git034127ab) ...
Replacing files in old package linux-firmware (20220329.git681281e4-0ubuntu3.12) ...
Setting up orangepi-firmware (20230217.git034127ab) ...
orangepi-firmware set on hold.
dpkg-deb: error: '/tmp/wiringpi_2.47.deb' is not a Debian format archive
dpkg: error processing archive /tmp/wiringpi_2.47.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /tmp/wiringpi_2.47.deb
Error: in ./scripts/build-rootfs.sh on line 224
Error: in ./build.sh on line 9
```

Full log, just FYI
https://paste.armbian.com/tayoxejena

Joshua-Riek, 2023-04-10:
Make sure you have git-lfs installed. I guess that the wiringpi deb file did not fetch properly with lfs.

EvilOlaf, 2023-04-10:
`git-lfs` was installed already but workspace was dirty. Unclean mounts and stuff. Recreating fixed this: https://paste.armbian.com/ajunexataq

It his supposed to produce both server and desktop images since I got server only?

Also noticed this close to the very end of the log file mentioned above: 
`Error: in ../scripts/build-image.sh on line 73`

Joshua-Riek, 2023-04-10:
It's supposed to produce the desktop and server images. Very weird that the script did not continue after creating the server image. I'll start a build on my TeamCity server and local environment to see if I get the same result.

`Error: in ../scripts/build-image.sh on line 73` can be ignored. It's related to `Re-reading the partition table failed.: Invalid argument`.

EvilOlaf, 2023-04-12:
Retried today, giving the builder even more memory and it worked. 

Maybe consider adjusting xz parameters to more reasonable valuables. Default with `-T` will most likely do.
Using `--extreme` and max compression is simply not worth the wasted time, energy and memory just to save a few megabytes...

closed 2023-04-12 by EvilOlaf