Joshua-Riek/ubuntu-rockchip #109: mali_kbase_device_hw_csf and chromium crash the whole system

Mreddi1, 2023-04-10:
Hello. I have an issue that crushed the whole system. I have this error for 1 month now on your release [v1.9](https://github.com/Joshua-Riek/ubuntu-orange-pi5/releases/tag/v1.9), and yesterday I have updated to [v1.11](https://github.com/Joshua-Riek/ubuntu-orange-pi5/releases/tag/v1.11). 

It often happens when I open Chromium browser and do anything, I mean, it does not depend on some actions, it crashes randomly and very often. I can reboot my Orange Pi 10 times per day, and it is annoying.

I have captured some logs:
```
Apr 10 00:42:58 eduard-desktop kernel: [   14.834055] rockchip-dmc dmc: failed to get vop bandwidth to dmc rate
Apr 10 00:42:58 eduard-desktop kernel: [   14.835223] mali fb000000.gpu: avs=0
Apr 10 00:42:58 eduard-desktop kernel: [   14.837475] rockchip-dmc dmc: failed to get vop pn to msch rl
Apr 10 00:42:58 eduard-desktop kernel: [   14.839146] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 136; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.

Apr 10 00:57:42 eduard-desktop kernel: [  904.187107] Mali<ASSERT>: In file: drivers/gpu/arm/bifrost/device/backend/mali_kbase_device_hw_csf.c line: 225 function:kbase_reg_read 

Apr 10 00:42:58 eduard-desktop kernel: [   14.919404] rockchip-pm rockchip-suspend: not set pwm-regulator-config
Apr 10 00:42:58 eduard-desktop kernel: [   14.920653] I : [File] : drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 405; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '08:14:53', on 'Mar 29 2023'.

Apr 10 00:43:08 eduard-desktop kernel: [   29.005814] dwhdmi-rockchip fde80000.hdmi: Rate 85750000 missing; compute N dynamically
Apr 10 00:43:08 eduard-desktop kernel: [   29.011412] dwhdmi-rockchip fde80000.hdmi: Rate 85750000 missing; compute N dynamically
Apr 10 00:43:09 eduard-desktop kernel: [   30.807381] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
Apr 10 00:43:09 eduard-desktop kernel: [   30.825686] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 1920x1080p60, type: 11(if:800) for vp0 dclk: 148500000
Apr 10 00:43:09 eduard-desktop kernel: [   30.825805] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:16a8c8 rate:857500
Apr 10 00:43:09 eduard-desktop kernel: [   30.826070] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
Apr 10 00:43:09 eduard-desktop kernel: [   30.826141] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] dclk_out0 div: 0 dclk_core0 div: 2
Apr 10 00:43:09 eduard-desktop kernel: [   30.826202] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:16a8c8 rate:1485000
Apr 10 00:43:09 eduard-desktop kernel: [   30.826464] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
Apr 10 00:43:09 eduard-desktop kernel: [   30.826470] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] set dclk_vop0 to 148500000, get 148500000
Apr 10 00:43:10 eduard-desktop kernel: [   30.843304] dwhdmi-rockchip fde80000.hdmi: final tmdsclk = 148500000
Apr 10 00:43:10 eduard-desktop kernel: [   30.843336] dwhdmi-rockchip fde80000.hdmi: don't use dsc mode
Apr 10 00:43:10 eduard-desktop kernel: [   30.843340] dwhdmi-rockchip fde80000.hdmi: dw hdmi qp use tmds mode
Apr 10 00:43:10 eduard-desktop kernel: [   30.843348] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: bus_width:0x16a8c8,bit_rate:1485000
Apr 10 00:43:10 eduard-desktop kernel: [   30.843534] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy lane locked!
Apr 10 00:43:25 eduard-desktop kernel: [   46.391570] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Apr 10 00:54:53 eduard-desktop kernel: [  735.187318] perf: interrupt took too long (2518 > 2500), lowering kernel.perf_event_max_sample_rate to 79200
Apr 10 00:57:42 eduard-desktop kernel: [  904.187107] Mali<ASSERT>: In file: drivers/gpu/arm/bifrost/device/backend/mali_kbase_device_hw_csf.c line: 225 function:kbase_reg_read 
Apr 10 00:57:42 eduard-desktop kernel: [  904.187112] kbdev->pm.backend.gpu_powered
Apr 10 00:57:42 eduard-desktop kernel: [  904.187128] 
Apr 10 00:57:42 eduard-desktop kernel: [  904.188010] ------------[ cut here ]------------
Apr 10 00:57:42 eduard-desktop kernel: [  904.188011] kernel BUG at drivers/gpu/arm/bifrost/device/backend/mali_kbase_device_hw_csf.c:225!
Apr 10 00:57:42 eduard-desktop kernel: [  904.188017] Internal error: Oops - BUG: 0 [#1] SMP
Apr 10 00:57:42 eduard-desktop kernel: [  904.189916] Modules linked in: bnep squashfs binfmt_misc sg iwlmvm iwlwifi dm_multipath dm_mod scsi_dh_rdac scsi_dh_emc scsi_dh_alua sch_fq_codel fuse ip_tables ipv6 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx raid1 raid0 multipath linear md_mod sd_mod
Apr 10 00:57:42 eduard-desktop kernel: [  904.192117] CPU: 6 PID: 1827 Comm: chromium-browse Not tainted 5.10.110-rockchip-rk3588 #1
Apr 10 00:57:42 eduard-desktop kernel: [  904.192836] Hardware name: Orange Pi 5 (DT)
Apr 10 00:57:42 eduard-desktop kernel: [  904.193210] pstate: 60400009 (nZCv daif +PAN -UAO -TCO BTYPE=--)
Apr 10 00:57:42 eduard-desktop kernel: [  904.193741] pc : kbase_reg_read+0x64/0x15c
Apr 10 00:57:42 eduard-desktop kernel: [  904.194102] lr : kbase_reg_read+0x64/0x15c
Apr 10 00:57:42 eduard-desktop kernel: [  904.194462] sp : ffffffc015fcbb90
Apr 10 00:57:42 eduard-desktop kernel: [  904.194758] x29: ffffffc015fcbb90 x28: ffffff81076c3130 
Apr 10 00:57:42 eduard-desktop kernel: [  904.195228] x27: ffffff81076c3198 x26: ffffff81076c3198 
Apr 10 00:57:42 eduard-desktop kernel: [  904.195696] x25: 0000000000000000 x24: 00000000000005dc 
Apr 10 00:57:42 eduard-desktop kernel: [  904.196166] x23: ffffff81076c0000 x22: ffffff81c4dd6c00 
Apr 10 00:57:42 eduard-desktop kernel: [  904.196637] x21: ffffff810104c010 x20: ffffff81076c0000 
Apr 10 00:57:42 eduard-desktop kernel: [  904.197106] x19: ffffff81ced90400 x18: 0000000000000000 
Apr 10 00:57:42 eduard-desktop kernel: [  904.197577] x17: 0000000000000000 x16: 0000000000000000 
Apr 10 00:57:42 eduard-desktop kernel: [  904.198046] x15: 0000000000000010 x14: 0000000000000000 
Apr 10 00:57:42 eduard-desktop kernel: [  904.198516] x13: ffffffc015fcb9f8 x12: ffffffffffffffff 
Apr 10 00:57:42 eduard-desktop kernel: [  904.198986] x11: 0000000000000010 x10: 0000000000000020 
Apr 10 00:57:42 eduard-desktop kernel: [  904.199456] x9 : ffffffc011038e1c x8 : 0000000000000000 
Apr 10 00:57:42 eduard-desktop kernel: [  904.199926] x7 : 00000000ffffffff x6 : ffffffc015fcb934 
Apr 10 00:57:42 eduard-desktop kernel: [  904.200397] x5 : 0000000000000030 x4 : ffffffc015fcb740 
Apr 10 00:57:42 eduard-desktop kernel: [  904.200865] x3 : 0000000000000000 x2 : 0000000000000000 
Apr 10 00:57:42 eduard-desktop kernel: [  904.201336] x1 : 0000000000000000 x0 : ffffff8144ca1e80 
Apr 10 00:57:42 eduard-desktop kernel: [  904.201806] Call trace:
Apr 10 00:57:42 eduard-desktop kernel: [  904.202027]  kbase_reg_read+0x64/0x15c
Apr 10 00:57:42 eduard-desktop kernel: [  904.202358]  kbase_backend_get_cycle_cnt+0x2c/0x6c
Apr 10 00:57:42 eduard-desktop kernel: [  904.202785]  kbase_csf_scheduler_queue_stop+0x2b0/0x530
Apr 10 00:57:42 eduard-desktop kernel: [  904.203246]  unbind_queue+0x5c/0x68
Apr 10 00:57:42 eduard-desktop kernel: [  904.203554]  kbase_csf_queue_unbind+0x70/0x74
Apr 10 00:57:42 eduard-desktop kernel: [  904.203938]  kbase_csf_user_io_pages_vm_close+0x80/0xc0
Apr 10 00:57:42 eduard-desktop kernel: [  904.204398]  remove_vma+0x3c/0x58
Apr 10 00:57:42 eduard-desktop kernel: [  904.204694]  __do_munmap+0x1c4/0x300
Apr 10 00:57:42 eduard-desktop kernel: [  904.205012]  __vm_munmap+0x7c/0xe0
Apr 10 00:57:42 eduard-desktop kernel: [  904.205318]  __arm64_sys_munmap+0x3c/0x50
Apr 10 00:57:42 eduard-desktop kernel: [  904.205680]  el0_svc_common.constprop.0+0x13c/0x1f0
Apr 10 00:57:42 eduard-desktop kernel: [  904.206105]  do_el0_svc+0x84/0xa4
Apr 10 00:57:42 eduard-desktop kernel: [  904.206404]  el0_svc+0x20/0x30
Apr 10 00:57:42 eduard-desktop kernel: [  904.206679]  el0_sync_handler+0xcc/0x154
Apr 10 00:57:42 eduard-desktop kernel: [  904.207029]  el0_sync+0x1a0/0x1c0
Apr 10 00:57:42 eduard-desktop kernel: [  904.207325] 
Apr 10 00:57:42 eduard-desktop kernel: [  904.207325] PC: 0xffffffc0108f06b8:
```

I will attach the full logs file.
[full.log](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/11193149/full.log)

I found this file https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-5.10-rk3588/drivers/gpu/arm/bifrost/device/backend/mali_kbase_device_hw_csf.c. But, unfortunately, I am not well-known with C.

Joshua-Riek, 2023-04-10:
Do you know what power adapter you are using? It should be rated for 5V/4A. I could see this issue happening if the board does not get enough power.

Mreddi1, 2023-04-10:
Yes, I bought this power adapter https://www.aliexpress.com/item/1005003160082010.html - 5V 4A Charger 20W USB Type C

Mreddi1, 2023-04-10:
But, I did not test it by measuring instrument...

Mreddi1, 2023-04-10:
Also, I have connected to the board:
- ORICO M2 SSD Enclosure NVMe 10Gbps PCIe SSD Box (USB) - https://www.aliexpress.com/item/1005004865192860.html
- KingSpec SSD M2 512GB NVME SSD (USB) - https://www.aliexpress.com/item/1005002924616022.html
- Intel WiFi 6E AX210 (M2) - https://www.aliexpress.com/item/1005004566795793.html
- And a Bluetooth mouse, a wire keyboard and a soundbar through the audio Jack

Joshua-Riek, 2023-04-10:
This is very strange. I have not had any other reports of chromium starting and causing the entire system to crash. 
I have limited kernel knowledge and is something I need to research more, however try without your USB devices attached, only a mouse and keyboard.

Mreddi1, 2023-04-10:
Okay, I will try, thank you! But, I have one other issue, I will create the new task. I finally got to the system logs :)

Mreddi1, 2023-04-10:
And a little bit more logs, maybe it can help..

```
Apr 10 00:42:58 eduard-desktop kernel: [   14.782730] pstore: Using crash dump compression: deflate
Apr 10 00:42:58 eduard-desktop kernel: [   14.782968] rga3_core0 fdb60000.rga: Adding to iommu group 2
Apr 10 00:42:58 eduard-desktop kernel: [   14.783164] rga: rga3_core0_irq, irq = 34, match scheduler
Apr 10 00:42:58 eduard-desktop kernel: [   14.783420] rga: rga3_core0 hardware loaded successfully, hw_version:3.0.76831.
Apr 10 00:42:58 eduard-desktop kernel: [   14.783439] rga: rga3_core0 probe successfully
Apr 10 00:42:58 eduard-desktop kernel: [   14.783707] rga3_core1 fdb70000.rga: Adding to iommu group 3
Apr 10 00:42:58 eduard-desktop kernel: [   14.783892] rga: rga3_core1_irq, irq = 35, match scheduler
Apr 10 00:42:58 eduard-desktop kernel: [   14.784197] rga: rga3_core1 hardware loaded successfully, hw_version:3.0.76831.
Apr 10 00:42:58 eduard-desktop kernel: [   14.784215] rga: rga3_core1 probe successfully
Apr 10 00:42:58 eduard-desktop kernel: [   14.784559] rga: rga2_irq, irq = 36, match scheduler
Apr 10 00:42:58 eduard-desktop kernel: [   14.784863] rga: rga2 hardware loaded successfully, hw_version:3.2.63318.
Apr 10 00:42:58 eduard-desktop kernel: [   14.784871] rga: rga2 probe successfully
Apr 10 00:42:58 eduard-desktop kernel: [   14.785049] rga2_mmu: IOMMU binding successfully, default mapping core[0x1]
Apr 10 00:42:58 eduard-desktop kernel: [   14.785231] rga: Module initialized. v1.2.20
Apr 10 00:42:58 eduard-desktop kernel: [   14.828183] reg-fixed-voltage combophy-avdd0v85: Looking up vin-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.828280] combophy_avdd0v85: supplied by vdd_0v85_s0
Apr 10 00:42:58 eduard-desktop kernel: [   14.828291] vdd_0v85_s0: could not add device link regulator.32: -ENOENT
Apr 10 00:42:58 eduard-desktop kernel: [   14.828631] combophy_avdd0v85: 850 mV, enabled
Apr 10 00:42:58 eduard-desktop kernel: [   14.828875] reg-fixed-voltage combophy-avdd0v85: combophy_avdd0v85 supplying 850000uV
Apr 10 00:42:58 eduard-desktop kernel: [   14.829095] reg-fixed-voltage combophy-avdd1v8: Looking up vin-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.829109] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.829181] combophy_avdd1v8: supplied by avcc_1v8_s0
Apr 10 00:42:58 eduard-desktop kernel: [   14.829192] avcc_1v8_s0: could not add device link regulator.33: -ENOENT
Apr 10 00:42:58 eduard-desktop kernel: [   14.829223] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gpu-supply property in node /power-management@fd8d8000/power-controller failed
Apr 10 00:42:58 eduard-desktop kernel: [   14.829431] combophy_avdd1v8: 1800 mV, enabled
Apr 10 00:42:58 eduard-desktop kernel: [   14.829449] mali fb000000.gpu: Kernel DDK version g13p0-01eac0
Apr 10 00:42:58 eduard-desktop kernel: [   14.829489] mali fb000000.gpu: Looking up mali-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.829668] reg-fixed-voltage combophy-avdd1v8: combophy_avdd1v8 supplying 1800000uV
Apr 10 00:42:58 eduard-desktop kernel: [   14.829757] mali fb000000.gpu: Looking up mem-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.830158] mali fb000000.gpu: Looking up mali-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.830180] vcc_3v3_sd_s0: 3300 mV, disabled
Apr 10 00:42:58 eduard-desktop kernel: [   14.830197] vdd_gpu_s0: could not add device link fb000000.gpu: -EEXIST
Apr 10 00:42:58 eduard-desktop kernel: [   14.830203] vdd_gpu_s0: Failed to create debugfs directory
Apr 10 00:42:58 eduard-desktop kernel: [   14.830213] mali fb000000.gpu: Looking up mem-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.830246] vdd_gpu_s0: could not add device link fb000000.gpu: -EEXIST
Apr 10 00:42:58 eduard-desktop kernel: [   14.830252] vdd_gpu_s0: Failed to create debugfs directory
Apr 10 00:42:58 eduard-desktop kernel: [   14.830347] reg-fixed-voltage vcc-3v3-sd-s0: Looking up vin-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.830418] vcc_3v3_sd_s0: supplied by vcc_3v3_s3
Apr 10 00:42:58 eduard-desktop kernel: [   14.830629] mali fb000000.gpu: Looking up mali-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.830667] vdd_gpu_s0: could not add device link fb000000.gpu: -EEXIST
Apr 10 00:42:58 eduard-desktop kernel: [   14.830673] vdd_gpu_s0: Failed to create debugfs directory
Apr 10 00:42:58 eduard-desktop kernel: [   14.830692] reg-fixed-voltage vcc-3v3-sd-s0: vcc_3v3_sd_s0 supplying 3300000uV
Apr 10 00:42:58 eduard-desktop kernel: [   14.831006] mali fb000000.gpu: leakage=13
Apr 10 00:42:58 eduard-desktop kernel: [   14.831135] mali fb000000.gpu: Looking up mali-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.831174] debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
Apr 10 00:42:58 eduard-desktop kernel: [   14.831595] rockchip-dmc dmc: Looking up center-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.832046] rockchip-dmc dmc: Looking up mem-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.832459] rockchip-dmc dmc: Looking up center-supply from device tree
Apr 10 00:42:58 eduard-desktop kernel: [   14.832487] vdd_ddr_s0: could not add device link dmc: -EEXIST
[... cut ...]
Apr 10 00:42:58 eduard-desktop kernel: [   14.971851] input: SONiX USB Keyboard System Control as /devices/platform/fc840000.usb/usb3/3-1/3-1:1.1/0003:09DA:2268.0002/input/input8
Apr 10 00:42:58 eduard-desktop kernel: [   14.972393] hid-generic 0003:09DA:2268.0002: input,hidraw1: USB HID v1.11 Device [SONiX USB Keyboard] on usb-fc840000.usb-1/input1
Apr 10 00:42:58 eduard-desktop kernel: [   15.113771] scsi host0: uas
Apr 10 00:42:58 eduard-desktop kernel: [   15.117645] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210B-CG      1.00 PQ: 0 ANSI: 6
```

Mreddi1, 2023-04-10:
```
[   14.486147] combophy_avdd1v8: 1800 mV, enabled
[   14.486155] mali fb000000.gpu: Looking up mali-supply from device tree
[   14.486173] vdd_gpu_s0: could not add device link fb000000.gpu: -EEXIST
[   14.486177] vdd_gpu_s0: Failed to create debugfs directory
[   14.486388] reg-fixed-voltage combophy-avdd1v8: combophy_avdd1v8 supplying 1800000uV
[   14.486445] mali fb000000.gpu: leakage=13
[   14.486472] mali fb000000.gpu: Looking up mali-supply from device tree
[   14.486494] debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
[   14.486648] vcc_3v3_sd_s0: 3300 mV, disabled
[   14.486805] reg-fixed-voltage vcc-3v3-sd-s0: Looking up vin-supply from device tree
[   14.486856] vcc_3v3_sd_s0: supplied by vcc_3v3_s3
[   14.487278] reg-fixed-voltage vcc-3v3-sd-s0: vcc_3v3_sd_s0 supplying 3300000uV
[   14.488003] mali fb000000.gpu: pvtm=877
[   14.488051] mali fb000000.gpu: pvtm-volt-sel=3
[   14.488082] rockchip-dmc dmc: Looking up center-supply from device tree
[   14.488440] rockchip-dmc dmc: Looking up mem-supply from device tree
[   14.488920] mali fb000000.gpu: avs=0
[   14.488942] rockchip-dmc dmc: Looking up center-supply from device tree
[   14.488982] vdd_ddr_s0: could not add device link dmc: -EEXIST
[   14.488989] vdd_ddr_s0: Failed to create debugfs directory
[   14.489305] rockchip-dmc dmc: Looking up mem-supply from device tree
[   14.489363] vdd_log_s0: could not add device link dmc: -EEXIST
[   14.489370] vdd_log_s0: Failed to create debugfs directory
[   14.489862] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 136; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.
[   14.490025] rockchip-dmc dmc: leakage=32
[   14.490046] rockchip-dmc dmc: leakage-volt-sel=1
[   14.490309] mali fb000000.gpu: GPU hardware issue table may need updating:
               r0p0 status 5 is unknown; treating as r0p0 status 0
[   14.490318] mali fb000000.gpu: GPU identified as 0x7 arch 10.8.6 r0p0 status 0
[   14.490390] mali fb000000.gpu: No priority control manager is configured
[   14.490397] mali fb000000.gpu: No memory group manager is configured
[   14.490421] mali fb000000.gpu: Protected memory allocator not available
[   14.490713] rockchip-dmc dmc: avs=0
[   14.490733] mali fb000000.gpu: Capping CSF_FIRMWARE_TIMEOUT to CSF_FIRMWARE_PING_TIMEOUT

rockchip-dmc dmc: failed to get vop bandwidth to dmc rate
[   14.491434] mali fb000000.gpu: Couldn't find power_model DT node matching 'arm,mali-simple-power-model'
[   14.493832] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.static-coefficient = 1*[0]
[   14.493839] rockchip-dmc dmc: failed to get vop pn to msch rl
[   14.494021] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.dynamic-coefficient = 1*[0]
[   14.496961] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.ts = 4*[0]
[   14.496982] rockchip-dmc dmc: l=10000 h=2147483647 hyst=5000 l_limit=0 h_limit=0 h_table=0
[   14.497231] mali fb000000.gpu: Error -22, no DT entry: mali-simple-power-model.thermal-zone = ''
[   14.501276] mali fb000000.gpu: Using configured power model mali-lodx-power-model, and fallback mali-simple-power-model
[   14.501474] mali fb000000.gpu: l=10000 h=85000 hyst=5000 l_limit=0 h_limit=800000000 h_table=0
[   14.502430] mali fb000000.gpu: Probed as mali0
[   14.512278] hid-generic 0003:09DA:2268.0001: input,hidraw0: USB HID v1.11 Keyboard [SONiX USB Keyboard] on usb-fc840000.usb-1/input0
[   14.512346] rockchip-dmc dmc: could not find power_model node
```

diepquynh, 2023-04-10:
@Mreddi1 a bit off-topic: which M.2 adapter do you use with your AX210? And does it require external power through USB pins? If yes, where do you plug it in GPIO headers?

Mreddi1, 2023-04-10:
@diepquynh I use Intel WiFi 6E AX210 (M2) - https://www.aliexpress.com/item/1005004566795793.html and Wireless M.2 A+E Key Slot To M.2 M Key - https://www.aliexpress.com/item/1005001274922542.html. I do not use external power for this adapter, I read that it is not required. But, maybe, it was a wrong opinion. I will investigate it, but AX210 works perfectly.

Only, what I have connected to the GPIO pins, it is a fan. https://www.aliexpress.com/item/1005005130553303.html. To the ports:
[image]

Joshua-Riek, 2023-04-10:
What the heck is going on with your board. I don't see any messages like that on my system. However, I have another board coming in sometime this week that I can try with. 

Try another OS like Armbian and set up hardware acceleration. Sadly, no other OS for this board comes with a proper hardware acceleration setup out of the box. Even the OrangePi5 Jammy image from the OrangePi company is not set up right.

Mreddi1, 2023-04-10:
Also, I will try to measure the real metrics of my power supply.

Joshua-Riek, 2023-04-10:
There are a lot of issues with monitor resolutions. For example, using an HDMI to VGA or DVI converter will not work on the system, 1080p, 1440p, and 2160p seem to be the sweet spot. However, I have minimal knowledge in this area and there is not a lot that I can do at this point.

Mreddi1, 2023-04-10:
Got it, I did not use any converter, I have just connected by HDMI cable to my TV monitor and to the board, and that is all. But, thank you a lot!

Joshua-Riek, 2023-04-10:
If you reproduce this issue on an Armbian system you can try opening an issue over there. They know a lot more than I do.

wyzdwdz, 2023-04-11:
This bug also happens on my device. I've tried this system, armbian and the official ubuntu system. It will happen with the GPU driver installed and crash after opening Chromium. However, it works well on the official debian system [](https://drive.google.com/drive/folders/1F2uc8v_EQnvsNrevDihwoymOJlFgM-dZ). I also suspect the power supply 'cause I bought the same one as you. But I don't have a measure device to prove that. If you get the measure result, please share with me.

Joshua-Riek, 2023-04-11:
I suspect you both have issues with power or the device is faulty.

Mreddi1, 2023-04-11:
@Joshua-Riek I hope it's just problems with the power adapter.. 

And I have one question, could you please tell me, when you power off the board from the Ubuntu system, in this case, your external (connected) devises also turned off? Or not? Because, my external devices continue working.. Fan, external SSD drive (just glow the diode on it).

ferrellsl, 2023-04-11:
> Someone else a while ago reported their fan continued to spin when shut down. I personally use a switch to be sure all power is off (I use this for development purposes). So I assume that it can happen, I think it might be due to a poor power delivery design by the Orange Pi company, however, I'm only a software engineer. I know very little about electrical engineering and the design of these types of devices.

Hi.  I was the guy who reported that my fan continued to spin after shutdown.  Apparently it's related to which pins you choose to power your fan.  I decided to use the 5V pins that are always powered on (even after shutting down the OS) and installed a USB-C inline power switch to fully shut the board off.  

https://www.reddit.com/r/OrangePI/comments/10padbx/fan_keeps_running_even_after_shutdown/

arun-511, 2023-04-11:
@Joshua-Riek Not just the 5V-GPIOs but all three USB A ports seem to be directly connected to the supply and they retain power even after shut down, at least for my board. Only the type-c display port properly cuts power after shut down.

Joshua-Riek, 2023-04-11:
Interesting, I guess it must be how the board was designed.

Joshua-Riek, 2023-04-29:
Hey, I had a few users talk about some system crashes in a recent discussion post. Are you still having issues? I'm trying to gather data to isolate the problem and determine the cause. 

https://github.com/Joshua-Riek/ubuntu-orange-pi5/discussions/129

arun-511, 2023-04-29:
Sometimes the OS just freezes during heavy Chrome usage and I have to forcefully shut it down using the power button. It's hard to reproduce and occurs randomly. However, yesterday system froze on the startup without Chrome running.

ferrellsl, 2023-04-29:
> Sometimes the OS just freezes during heavy Chrome usage and I have to forcefully shut it down using the power button. It's hard to reproduce and occurs randomly. However, yesterday system froze on the startup without Chrome running.

From my own experience, freezes can occur when you have modified the CPU governor settings from the stock settings to anything outside of normal operating parameters.  The other cause can be a dodgy or underpowered power supply.  If you're using a power supply that is rated at less than 5V/4A, then you're almost guaranteed to have issues.

Joshua-Riek, 2023-04-29:
> From my own experience, freezes can occur when you have modified the CPU governor settings from the stock settings to anything outside of normal operating parameters. The other cause can be a dodgy [...]

By default, I have a service that sets the CPU and GPU governor to performance every boot. Do you disable this?

arun-511, 2023-04-29:
> By default, I have a service that sets the CPU and GPU governor to performance every boot. Do you disable this?

Nope, I haven't changed anything (no need, everything runs fine with stock config) and I have a 5V/5A power brick to run the board. On top I have placed a heatsink and a small fan on the cpu, temps are below 40C even when playing 4K videos in chrome.

Everything works fine, it just happens to randomly freez sometimes.

Joshua-Riek, 2023-04-29:
I wonder if removing the service that sets the governor will prevent this. However, it will decrease performance, as there is a bug where the system cant set the CPU and GPU fast enough under a load, which is why I have this set by default in the first place. 
```
sudo disable systemctl cpu-governor-performance
sudo disable systemctl gpu-governor-performance
```

arun-511, 2023-04-29:
Interesting, I will try this right away.

Joshua-Riek, 2023-04-29:
Make sure to reboot after issuing those commands. They can be enabled again easily if you wish.
```
sudo enable systemctl cpu-governor-performance
sudo enable systemctl gpu-governor-performance
```

Joshua-Riek, 2023-04-29:
I took another look at some error logs provided by another user and noticed a CMA allocation error appearing a few times before the crash would happen.

Try to increase the CMA size to 256M with the below commands. 
```
sudo sed -i 's/cma=64M/cma=256M/g' -i /boot/firmware/boot.cmd
sudo mkimage -A arm64 -O linux -T script -C none -n "Boot Script" -d /boot/firmware/boot.cmd /boot/firmware/boot.scr
sudo reboot now
```

diepquynh, 2023-04-30:
Caught a crash while in the middle of Moonlight streaming and I got to SSH to my OPi5 to get dmesg. It also happens when I left the board idle in lock screen too
[image]

Joshua-Riek, 2023-04-30:
A page fault is very concerning. How long were you streaming for? I'm trying to reproduce this on my end.

diepquynh, 2023-04-30:
I'm afraid it's hard to reproduce, as I was streaming for about 5 hours straight at 1080p60 150Mbps on my local network, running 2 monitors (one via USB-C DP alt, one via HDMI)

diepquynh, 2023-04-30:
Initially I thought it was VOP error triggering again so I tried to unplug the secondary monitor connected via USB-C DP alt then reconnect again, and later I realized that both of my screens were frozen

Joshua-Riek, 2023-04-30:
A while ago, there was an issue with plugging in a monitor to HDMI and USB-C simultaneously where the GPU drivers would break https://github.com/Joshua-Riek/ubuntu-orange-pi5/issues/32. Since then, the patch has been added to the orangepi kernel. I wonder if it's related.

Mreddi1, 2023-05-01:
> Hey, I had a few users talk about some system crashes in a recent discussion post. Are you still having issues? I'm trying to gather data to isolate the problem and determine the cause.
> 
> #129

Hello, yes, I still have this issue. Unfortunately, I did not have a chance to get a measurement tool and measure my power supply. I tried to get this information about my power supply from the system, but it was unsuccessfully, the system did not have any information about currency, voltage, etc. I will try buying the measurement tool asap, and provide you information.

Also, I had some logs from dmesg, I wanted to investigate them, but I did not have a chance, I will attach them here, maybe it will be useful.

Besides, regarding the above messages, my system also crashes every time when I open a Chromium browser.

Joshua-Riek, 2023-05-01:
> > Hey, I had a few users talk about some system crashes in a recent discussion post. Are you still having issues? I'm trying to gather data to isolate the problem and determine the cause.
> > #129
> 
> Hello, yes, I still have this issue. Unfortunately, I did not have a chance to get a measurement tool and measure my power supply. I tried to get this information about my power supply from the system, but it was unsuccessfully, the system did not have any information about currency, voltage, etc. I will try buying the measurement tool asap, and provide you information.
> 
> Also, I had some logs from dmesg, I wanted to investigate them, but I did not have a chance, I will attach them here, maybe it will be useful.
> 
> Besides, regarding the above messages, my system also crashes every time when I open a Chromium browser.
> 
> [logs_init_system.txt](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/11363729/logs_init_system.txt)

I'm quite sure you have a power issue, there are a lot of messages about the power controller failing. However it's impossible for me to tell if its a hardware issue or your power supply is at fault.

closed 2023-05-19 by Joshua-Riek