Joshua-Riek/ubuntu-rockchip #112: Bluetooth keyboard with wifi\bluetooth m.2 pcie

davidstoll, 2023-04-11:
I activated the card with the echo command in the instructions.  It works great, but after every reboot, the keyboard doesn't reconnect.  I have to reinitiate the pairing (on both the keyboard and Ubuntu) to get it working again.

wptracy, 2023-04-13:
I have a similar problem.  It must have happened after a sudo apt update, because nothing else changed.

I can no longer pair my Bluetooth audio device.

I get this error:
bluetoothd[841]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 50:C9:71:9A:0A:4C: Protocol not available

I've tried fixing it with google searches.  Nothing I tried works.

Joshua-Riek, 2023-04-13:
I noticed an issue where some of the orange pi-specific WiFi / Bluetooth firmware are overridden by apt update. I should have fixed it in the latest release. However, I packaged the previously mentioned firmware, which may work and restore the missing files. 

Download this attached file and extract it. Then run the command `sudo dpkg -i orangepi-firmware_20230217.git9e0d07cc_all.deb`. Finally, give your system a reboot, and it might solve the issue.

[orangepi-firmware_20230217.git9e0d07cc_all.zip](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/11223418/orangepi-firmware_20230217.git9e0d07cc_all.zip)

wptracy, 2023-04-13:
Joshua,

I installed the deb file and rebooted but it didn't fix it.

I still get this error:
Apr 13 09:09:10 wptracy-desktop bluetoothd[851]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 50:C9:71:9A:0A:4C: Protocol not available

Joshua-Riek, 2023-04-13:
I don't have something to check the version, I need to add that haha. No worries, though, ill investigate when time allows.

Joshua-Riek, 2023-04-21:
Sorry for only getting back to this issue now. I have been very busy prepping for some significant changes I just released. Part of the recent changes includes a few major bug fixes for WiFi / Bluetooth firmware. Because of the large changes I made, I'm going to close this for now, please try on v1.13 and if you encounter any issues still please let me know and I will re-open the issue.

davidstoll, 2023-04-27:
I'm guessing it does because I've used a dongle, the m.2 card and a 5b built in and after a reboot nothing reconnects, but I can verify, how do I check that?  But if you don't need me to, that's ok.  I have now verified the non-reconnecting of bluetooth on v1.13 on all 3 situations:  Opi 5 usb dongle, Opi 5 m.2 card, Opi5b built in.

Joshua-Riek, 2023-04-27:
You can use `hciconfig` to find the address. In my case, I have an address of `84:14:9C:01:AA:24`, then when i reboot it changes to `32:A0:66:3C:9D:0E`.

```
ubuntu@ubuntu-desktop:~$ hciconfig 
hci0:	Type: Primary  Bus: UART
	BD Address: 84:14:9C:01:AA:24  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:916 acl:0 sco:0 events:67 errors:0
	TX bytes:3836 acl:0 sco:0 commands:67 errors:0
ubuntu@ubuntu-desktop:~$ sudo reboot now
...
ubuntu@ubuntu-desktop:~$ hciconfig 
hci0:	Type: Primary  Bus: UART
	BD Address: 32:A0:66:3C:9D:0E  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:916 acl:0 sco:0 events:67 errors:0
	TX bytes:3836 acl:0 sco:0 commands:67 errors:0
```

Joshua-Riek, 2023-04-27:
Oh, I might see the issue now. The parameter `--bd_addr_rand` makes the Bluetooth address random each time Bluetooth starts. 

https://github.com/Joshua-Riek/ubuntu-orange-pi5/blob/5c4329cac47fe872d4dd4a01b1d55511778c6f81/overlay/usr/lib/scripts/ap6275p-bluetooth.sh#L6

Try and edit the file `/usr/lib/scripts/ap6275p-bluetooth.sh`. 
Change `--bd_addr_rand` with something like `--bd_addr '89:C9:A7:15:30:59'` and reboot your system. I have an example below of my ap6275p-bluetooth.sh file looks like below. 
```bash
#!/bin/bash

bt_status=$(cat /proc/device-tree/wireless-bluetooth/status)
if [[ ${bt_status} == "okay" ]]; then
    rfkill unblock all
    brcm_patchram_plus --bd_addr '89:C9:A7:15:30:59' --enable_hci --no2bytes --use_baudrate_for_download --tosleep 200000 \
        --baudrate 1500000 --patchram /lib/firmware/ap6275p/BCM4362A2.hcd /dev/ttyS9 &
fi
```

Joshua-Riek, 2023-04-27:
Ok so, the issue is the parameter `--bd_addr_rand`. Removing it will use the hardware-assigned MAC address for Bluetooth. I will start a build on my end and double-check, on a fresh system.

Joshua-Riek, 2023-05-01:
I'm happy to say the issue is now solved, and the fix will be in the next release.

closed 2023-05-01 by Joshua-Riek

Sped0n, 2023-08-03:
> Oh, I might see the issue now. The parameter `--bd_addr_rand` makes the Bluetooth address random each time Bluetooth starts.
> 
> https://github.com/Joshua-Riek/ubuntu-orange-pi5/blob/5c4329cac47fe872d4dd4a01b1d55511778c6f81/overlay/usr/lib/scripts/ap6275p-bluetooth.sh#L6
> 
> Try and edit the file `/usr/lib/scripts/ap6275p-bluetooth.sh`. Change `--bd_addr_rand` with something like `--bd_addr '89:C9:A7:15:30:59'` and reboot your system. I have an example below of my ap6275p-bluetooth.sh file looks like below.
> 
> ```shell
> #!/bin/bash
> 
> bt_status=$(cat /proc/device-tree/wireless-bluetooth/status)
> if [[ ${bt_status} == "okay" ]]; then
>     rfkill unblock all
>     brcm_patchram_plus --bd_addr '89:C9:A7:15:30:59' --enable_hci --no2bytes --use_baudrate_for_download --tosleep 200000 \
>         --baudrate 1500000 --patchram /lib/firmware/ap6275p/BCM4362A2.hcd /dev/ttyS9 &
> fi
> ```

for those who use ubuntu image from orangepi, the file path is `/usr/lib/orangepi/orangepi-hardware-optimization`

jinglong485, 2024-06-08:
Hi,
I got the same issue on the Orange pi 5 pro using the latest Ubuntu 24.04(06-07-2024). every time I reboot the PI, there is a new mac address generated for bluetooth. Also if I use reboot directly, after rebooting , wifi is not working, dmesg gives "wlan_platdata_parse_dt: the ref_wifi_clk not found!". If I use shutdown and then turn it on, wifi works well.
Thank you.

jinglong485, 2024-06-09:
Hi, 
I deleted the --bd_addr_rand in overlay/usr/lib/systemd/system/ap6256-bluetooth.service. now bluetooth works well, however after direct rebooting wifi still not working. Thank you.