Joshua-Riek/ubuntu-rockchip #116: Fonts appear grey in Chromium (fonts which should appear black)

martinbone, 2023-04-13:
I've noticed that on a lot of web pages (eg. Google Sheets) fonts that should appear black actually appear as light grey.

[image]

If no-one else has this issue - don't worry. I understand it's early days of development. Thanks.

diepquynh, 2023-04-13:
A quick fix for this I've found so far is to enable Raw Draw feature in Chromium, either by passing a `--enable-features=RawDraw` or just simply enable it in `chrome://flags`

martinbone, 2023-04-13:
Excellent, thanks. Will try that.

diepquynh, 2023-04-14:
There's another solution too if you don't care about 4k decoding, which is commenting out `CHROMIUM_FLAGS` in `/etc/chromium-browser/default`

This way makes Chromium return to default ANGLE renderer with Panfrost backend, and imo it's much more stable than using current EGL renderer, at the  cost of hardware codecs not working at all. Even so, watching a 1080p60 video on YouTube with SW codec looks pretty stable for me

hubby4, 2023-04-18:
> A quick fix for this I've found so far is to enable Raw Draw feature in Chromium, either by passing a `--enable-features=RawDraw` or just simply enable it in `chrome://flags`

This worked for me. Thank you.

closed 2023-05-19 by Joshua-Riek