Joshua-Riek/ubuntu-rockchip #117: Unable to use armhf architecture

cantalupo555, 2023-04-14:
When adding the 32-bit architecture the system does not correctly recognize armhf when updating repositories.

`sudo dpkg --add-architecture armhf && sudo apt-get update`
                                                                                                                                                                                                 
```
Ign:43 http://ports.ubuntu.com jammy-security/main i386 Packages                                                                                                                                                                                                                 
Ign:53 http://ports.ubuntu.com jammy-security/restricted i386 Packages                                                                                                                                                                                                           
Ign:56 http://ports.ubuntu.com jammy-security/universe i386 Packages                                                                                                                                                                                                             
Ign:63 http://ports.ubuntu.com jammy-security/multiverse i386 Packages                                                                                                                                                                                                           
Ign:67 http://ports.ubuntu.com jammy-updates/main i386 Packages                                                                                                                                                                                                                  
Ign:75 http://ports.ubuntu.com jammy-updates/restricted i386 Packages                                                                                                                                                                                                            
Ign:81 http://ports.ubuntu.com jammy-updates/universe i386 Packages                                                                                                                                                                                                              
Ign:87 http://ports.ubuntu.com jammy-updates/multiverse i386 Packages                                                                                                                                                                                                            
Ign:14 http://ports.ubuntu.com jammy/main i386 Packages                                                                                                                                                                                                                          
Ign:95 http://ports.ubuntu.com jammy-backports/main i386 Packages                                                                                                                                                                                                                
Ign:104 http://ports.ubuntu.com jammy-backports/universe i386 Packages                                                                                                                                                                                                           
Ign:24 http://ports.ubuntu.com jammy/restricted i386 Packages                                                                                                                                                                                                                    
Ign:26 http://ports.ubuntu.com jammy/universe i386 Packages                                                                                                                                                                                                                      
Ign:37 http://ports.ubuntu.com jammy/multiverse i386 Packages                                                                                                                                                                                                                    
Ign:43 http://ports.ubuntu.com jammy-security/main i386 Packages                                                                                                                                                                                                                 
Ign:53 http://ports.ubuntu.com jammy-security/restricted i386 Packages                                                                                                                                                                                                           
Ign:56 http://ports.ubuntu.com jammy-security/universe i386 Packages                                                                                                                                                                                                             
Ign:63 http://ports.ubuntu.com jammy-security/multiverse i386 Packages                                                                                                                                                                                                           
Ign:67 http://ports.ubuntu.com jammy-updates/main i386 Packages                                                                                                                                                                                                                  
Ign:75 http://ports.ubuntu.com jammy-updates/restricted i386 Packages                                                                                                                                                                                                            
Ign:81 http://ports.ubuntu.com jammy-updates/universe i386 Packages                                                                                                                                                                                                              
Ign:87 http://ports.ubuntu.com jammy-updates/multiverse i386 Packages                                                                                                                                                                                                            
Err:14 http://ports.ubuntu.com jammy/main i386 Packages                                                                                                                                                                                                                          
  404  Not Found [IP: 2620:2d:4000:1::16 80]
Ign:95 http://ports.ubuntu.com jammy-backports/main i386 Packages                                                                                                                                                                                                                
Ign:104 http://ports.ubuntu.com jammy-backports/universe i386 Packages                                                                                                                                                                                                           
Ign:24 http://ports.ubuntu.com jammy/restricted i386 Packages                                                                                                                                                                                                                    
Ign:26 http://ports.ubuntu.com jammy/universe i386 Packages                                                                                                                                                                                                                      
Ign:37 http://ports.ubuntu.com jammy/multiverse i386 Packages                                                                                                                                                                                                                    
Err:43 http://ports.ubuntu.com jammy-security/main i386 Packages                                                                                                                                                                                                                 
  404  Not Found [IP: 2620:2d:4000:1::16 80]
[... cut ...]
E: Failed to fetch http://ports.ubuntu.com/dists/jammy/main/binary-i386/Packages  404  Not Found [IP: 2620:2d:4000:1::16 80]
E: Failed to fetch http://ports.ubuntu.com/dists/jammy-security/main/binary-i386/Packages  404  Not Found [IP: 2620:2d:4000:1::16 80]
E: Failed to fetch http://ports.ubuntu.com/dists/jammy-updates/main/binary-i386/Packages  404  Not Found [IP: 2620:2d:4000:1::16 80]
E: Failed to fetch http://ports.ubuntu.com/dists/jammy-backports/main/binary-i386/Packages  404  Not Found [IP: 185.125.190.36 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joshua-Riek, 2023-04-14:
Adding the armf architecture works fine for me. But from the logs provided, it's an issue with the i386 architecture. 

Try this command and it should solve your issue:
```bash
sudo dpkg --remove-architecture i386 && sudo apt-get update
```

cantalupo555, 2023-04-14:
> Adding the armf architecture works fine for me. But from the logs provided, it's an issue with the i386 architecture.
> 
> Try this command and it should solve your issue:
> 
> ```shell
> sudo dpkg --remove-architecture i386 && sudo apt-get update
> ```

Thanks, this fixed the issue on my system.

closed 2023-04-14 by Joshua-Riek