Joshua-Riek/ubuntu-rockchip #120: Overlays

Gruttebier, 2023-04-15:
The overlays=ssd-sata path is changed the code needs the new path. /mnt/overlays.txt

Joshua-Riek, 2023-04-15:
fstab will mount the boot partition to /boot/firmware. The file should be at /boot/firmware/overlays.txt. 
Could you provide me the output of the lsblk and blkid commands? Maybe the UUID is messed up.

Gruttebier, 2023-04-15:
Its only the install image. When installed its the right pad.
[image]

Gruttebier, 2023-04-15:
Flash Bootloader to SPI

For a SATA SSD, please use the below comands:

sudo dd if=/usr/share/orangepi/rkspi_loader_sata.img of=/dev/mtdblock0 conv=notrunc 
echo "overlays=ssd-sata" | sudo tee /boot/firmware/overlays.txt

Joshua-Riek, 2023-04-15:
Do you have the os installed on sda and sdb? The partitions can get messed up due to having the same name. However, I did disable zram in the kernel. Did you enable it yourself?

Joshua-Riek, 2023-04-21:
I made a few changes in the new release v1.13, and I'm now using the UUID to mount the filesystem at boot time. I have not been able to reproduce this, but using the UUID should help prevent something like this from occurring.

closed 2023-04-21 by Joshua-Riek