Joshua-Riek/ubuntu-rockchip #121: Pink display with specific HDMI monitor

pterodyne, 2023-04-16:
I posted this in reddit orange pi forum, but Im getting pink display with a specific HDMI monitor, unfortunately that is the monitor Im using for the project. Anyway I was using armbian jammy, so I thought Id try your image and see. It does the same thing. 

Does anyone here know how to modify this image so it is outputting RGB instead of Y′CBCR?  I don't know for certain that is the issue, but with other models of Orange PI this has been an issue before.  The monitor works fine with a PI4 and my laptop, and conversely the OP5 works fine on other HDMI monitors.

Long shot for sure, but you don't get answers to questions not asked.  I need to post on the armbian forums, but haven't created an account yet.

Joshua-Riek, 2023-04-16:
I looked at the Rockchip rk3588 BSP Documents available [here](https://github.com/axlrose/rkdocs) and found the document Rockchip_Developer_Guide_HDMI_CN.pdf. It has the following information.
[image]
However, the document seems outdated, or something needs to be enabled in the kernel/device tree area, as I don't see any color nodes in the sysfs.

arun-511, 2023-04-19:
Just got a Waveshare 2k display and most of the time there is a pink tint over HDMI. Randomly disconnecting and reconnecting the cable after startup sometimes fixes the colors. I use the DP port most of the time and it works fine. I have tested official and Armbian images they all have the same issue.

BlackIsBlack, 2023-04-30:
Having the exact same issue and I've been pulling my hair out trying to fix this.
[image]
Any info on this would be great.

Joshua-Riek, 2023-04-30:
Could you upload the dmesg logs in a text file? I'm curious to see if anything strange will be shown. You can enter `dmesg > log.txt` in the console to write the log to a file.

BlackIsBlack, 2023-04-30:
[log.txt](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/11360607/log.txt)

BlackIsBlack, 2023-04-30:
Let me know if there's anything else I can provide to help debug this issue.

BlackIsBlack, 2023-04-30:
I've managed to fix the issue.
It was exactly as [pterodyne](https://github.com/pterodyne) mentioned above. My fix is as follows:
1.  Change from Wayland to X11 
    - ```sudo nano /etc/gdm3/custom.conf```
    - Update the line with “WaylandEnable=true” to  “WaylandEnable=false”
    - Restart the service using  ```sudo systemctl restart gdm3```
2. Then update the "color_format" value of the monitor using
```xrandr --output HDMI-1 --set "color_format" "rgb"```

This command apparently fails, but the colours are now displaying as they should be

BlackIsBlack, 2023-04-30:
[image]

[image]

BlackIsBlack, 2023-04-30:
Though, I'm unable to change the colour space in wayland. Any ideas on how to do this?

Joshua-Riek, 2023-04-30:
Fantastic find! However, I want to warn you about using X11. The GPU drivers work best with Wayland.

If you keep Wayland enabled and issue the xrandr command, does that work?

BlackIsBlack, 2023-04-30:
Noticing some massive issues with using X11 😬. There are actually no params when running xrandr. 
Running ```xrandr --output XWAYLAND0 --set "color_format" "rgb"``` results in:
```X Error of failed request:   BadName (named color or font does not exist) 

When I run ```xrandr --verbose``` the subpixel is listed as unknown.

I think this has to be fixed in the Wayland settings, but I've spent the entire day searching and to no luck.

Joshua-Riek, 2023-04-30:
I found a website that talks about changing the color space with Wayland. No need to compile libdrm, proptest can be installed with `sudo apt  install libdrm-tests`.

https://www.brad-x.com/2017/08/07/quick-tip-setting-the-color-space-value-in-wayland/#:~:text=If%20you%E2%80%99re%20using%20Xorg%20this%20can%20be%20adjusted,xrandr%20--output%20HDMI-0%20--set%20%22Broadcast%20RGB%22%20%22Limited%2016%3A235%22

BlackIsBlack, 2023-04-30:
Unfortunately it runs through all devices and fails before saying ```no device found```

Joshua-Riek, 2023-04-30:
Right, I saw that this will need some more research.

BlackIsBlack, 2023-04-30:
Is there a way to add custom arguments to the kernel command line? I will attempt to edit the EDID. I've noticed that grub isn't accessible.

Joshua-Riek, 2023-04-30:
Yes, it's a bit annoying to do so at the moment. The next release will make adding kernel command line arguments much more accessible. 

You must edit the file `/boot/firmware/boot.cmd` with sudo. On the 6th line, you will see `env set bootargs` followed by a long string. Inside the quotes, you can add your arguments. After saving the file, you will need to re-compile boot.cmd like below.
```
sudo mkimage -A arm64 -O linux -T script -C none -n "Boot Script" -d /boot/firmware/boot.cmd /boot/firmware/boot.scr
```

BlackIsBlack, 2023-04-30:
Managed to get it working using this guide here:
https://www.wezm.net/v2/posts/2020/linux-amdgpu-pixel-format/

Joshua-Riek, 2023-04-30:
Fantastic! I'm glad it's working now.

BlackIsBlack, 2023-04-30:
Here's the edid for [this ](https://www.aliexpress.com/item/1005004285641847.html?spm=a2g0o.order_list.order_list_main.47.21ef1802yJUng6) Wisecoco OLED panel. In case anyone else stumbles across this same problem with this same monitor on this same board 😅 

[edid.zip](https://github.com/Joshua-Riek/ubuntu-orange-pi5/files/11362681/edid.zip)

BlackIsBlack, 2023-04-30:
And thanks so much for the help @Joshua-Riek !!! 😊

Joshua-Riek, 2023-04-30:
If you run into any other issues, please let me know!

closed 2023-05-01 by Joshua-Riek

pterodyne, 2023-05-02:
> 
I can confirm that this works for me to with my elcrow 1280x800 display as well.  Thanks for that, at least I know it can be done.  Ill switch back to the OrangePI 5 again :)  I was using a pi4 for my project but it feels pretty slow now that Ive used an orange pi 5.  Next Ill have to look trough the wayland info below.

pterodyne, 2023-05-02:
I thought I posted this, but now I see it's not here.  So is it then true to get wayland working I eed an edid.bin file?  Where would I put it?  I looked at the instructions for a similar problem of not outputting RGB but on AMD, and I haven't been able to figure out where.  They reference it being in a pci folder that doesn't exist on this build of orange pi ubunut.

Thanks for any pointers.

Bryan

pterodyne, 2023-05-02:
These are the instructions that came with the display for a pi, to edit config.txt

hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=87
hdmi_ignore_edid=0xa5000080
hdmi_cvt 1280x800 60 3 0 0 0
hdmi_drive=2