Joshua-Riek/ubuntu-rockchip #123: Image does not regenerate machine-id

td512, 2023-04-21:
The image should regenerate both `/etc/machine-id` and `/var/lib/dbus/machine-id` on firstboot to ensure that the system id is unique. This can be done with `rm /etc/machine-id /var/lib/dbus/machine-id && dbus-uuidgen --ensure && systemd-machine-id-setup`

Joshua-Riek, 2023-04-21:
Interesting, thanks for bringing this to my attention, I might need to have the machine-id generated on the first boot or some of the os images might share the same machine-id. I will do some testing.

Joshua-Riek, 2023-04-21:
Thanks again for making me aware of this. By default, the file `/etc/machine-id` will now be empty and systemd will automatically generate a random machine-id on the first boot.

closed 2023-04-21 by Joshua-Riek

td512, 2023-04-21:
You will likely likely also remove /var/lib/dbus/machine-id since that's where systemd gets its machine-id from

Joshua-Riek, 2023-04-22:
Yeah, i made sure to remove /var/lib/dbus/machine-id