Joshua-Riek/ubuntu-rockchip #126: If you use Ubuntu (Joshua's release) use "Remote sharing" that exists in the settings manager. Follow the instructions and it'll work out of the box

opalfruit79, 2023-04-22:
If you use Ubuntu (Joshua's release) use "Remote sharing" that exists in the settings manager. Follow the instructions and it'll work out of the box

_Originally posted by @rahul-kurup in https://github.com/Joshua-Riek/ubuntu-orange-pi5/discussions/102#discussioncomment-5541099_

opalfruit79, 2023-04-22:
I have had no luck with this at all. Fresh image with the latest image 1.13.   I have also tried Arabian. Having no RDP for this build is making me very sad.

The same error is displayed in amban. I have tried installing xrdp / VNC to no avail.

[image]

Joshua-Riek, 2023-04-22:
Strange, did you configure your desktop environment to auto-login?

opalfruit79, 2023-04-22:
No, I read on a forum somewhere that it might be the cause of the problem.

So I have tried, both logged in and logged out.

My device is plugged into a projector. 

So I have re-imaged the latest build onto and SD card only. Logged in and its the first thing I am checking I working.

It's a orange pi 5 (16gb). I have tried your build with a SATA m.2 card. I have taken it out to examin this bug. Admittadly i have not wiped the partition table in the bios (which incdentally i needed to do to get the SATA instructions working on a new device).

I have had the same problem with Armbian. To be honest I am worrying its because I am a Linux novice.

opalfruit79, 2023-04-23:
So  have had some luck with RDP on the Arabian build.

Fresh build, headless, SSH, armbian-config, follow the intrusions it works.

HOWEVER

As soon as I install the video drivers, this fails after  the login screen on RDP. It drops the connection.
So, maybe this is related to to the video drivers?

closed 2023-05-29 by Joshua-Riek