Joshua-Riek/ubuntu-rockchip #132: U-Boot SPI Boot Failure

apowers313, 2023-04-28:
After building the Orange Pi 5 U-Boot image and writing it to SPI, the board won't boot. The only difference on the serial console is that `atf-1` and similar don't seem to be loaded (see below). It appears that the Arm Trusted Firmware isn't included in the SPI image (despite being part of the build process).

## To reproduce (building on the Orange Pi 5 itself):
``` bash
git clone https://github.com/Joshua-Riek/ubuntu-orange-pi5.git
cd ubuntu-orange-pi5/
./build.sh --board=orangepi5 --uboot-only
cp ./build/u-boot-orangepi/debian/build/orangepi5/rkspi_loader.img .
sudo dd if=./rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
```

## Output when booting:
```
[...]
## Verified-boot: 0
## Checking uboot 0x00200000 ... sha256(0603bd422d...) + OK
## Checking fdt 0x00349a00 ... sha256(6a0656d703...) + OK
Total: 495.155 ms

Jumping to U-Boot(0x00200000)
```
[device hangs at this point]

## Output when booting using Orange Pi provided U-Boot (/usr/lib/linux-u-boot-legacy-orangepi5_1.1.4_arm64/rkspi_loader.img):
```
[...]
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(806278dba1...) + OK
## Checking uboot 0x00200000 ... sha256(c4e3ce0f43...) + OK
## Checking fdt 0x00349718 ... sha256(ffc773db5b...) + OK
## Checking atf-2 0x000f0000 ... sha256(c00c7fd75b...) + OK
## Checking atf-3 0xff100000 ... sha256(71c3a5841b...) + OK
## Checking atf-4 0xff001000 ... sha256(2301cf73be...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 507.3 ms

INFO:    Preloader serial: 2
[...]
```
[device continues to boot at this point]

Joshua-Riek, 2023-04-28:
I tested the latest build on my build server, and U-Boot works fine. However, think that I know the issue. U-Boot will not report an error if python2 is not installed and will continue the build process. Try downloading the below packages and compile U-Boot again, it should work without issue.
```
sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs device-tree-compiler python2 python3 \
python-is-python3 fdisk
```

apowers313, 2023-04-29:
That did the trick, thanks!

And thanks again for the awsome project and speedy help.

closed 2023-04-29 by apowers313