Joshua-Riek/ubuntu-rockchip #134: Flash Bootloader to NVMe

bikingfreak, 2023-05-04:
I have multiple times followed the instructions to flash the bootloader to the SPI, but I can't get the box to boot from the NVMe SSD when I remove the SD card. The NVMe looks identical to the SPI right now, so what did I miss, or is there an additional step not described in your procedure?

Disk /dev/nvme0n1: 476.94 GiB, 512110190592 bytes, 1000215216 sectors
Disk model: SPCC M.2 PCIe SSD                       
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: E68A88F8-15BA-468D-BB3D-D76D90992D39

Device         Start        End    Sectors   Size Type
/dev/nvme0n1p1    64       7167       7104   3.5M Linux filesystem
/dev/nvme0n1p2  7168       7679        512   256K Linux filesystem
/dev/nvme0n1p3  7680       8063        384   192K Linux filesystem
/dev/nvme0n1p4  8064       8127         64    32K Linux filesystem
/dev/nvme0n1p5  8128       8191         64    32K Linux filesystem
/dev/nvme0n1p6  8192      16383       8192     4M Linux filesystem
/dev/nvme0n1p7 16384      32734      16351     8M Linux filesystem
/dev/nvme0n1p8 32768 1000214527 1000181760 476.9G Linux filesystem

The SD card has the boot partition, but I don't see that on the SSD

Disk /dev/mmcblk1: 238.75 GiB, 256355860480 bytes, 500695040 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 8A877FFE-ECA2-40D4-A698-C053F79EF90E

Device           Start       End   Sectors   Size Type
/dev/mmcblk1p1   32768   1081343   1048576   512M Linux extended boot
/dev/mmcblk1p2 1081344 500695006 499613663 238.2G Linux filesystem

Joshua-Riek, 2023-05-04:
The system does not recognize the bootloader on the NVMe, which is why the SPI exists. This is by design.

bikingfreak, 2023-05-04:
I have copied your command into the terminal and run it at least 3 times. When I then immediately shut down the Orange Pi, remove the SD card and power up again, it does not boot.

So are you saying that it will never boot onto the SSD without the SD card in its slot? That seems to make the SSD not worthwhile. If I try to list the contents of the SSD it shows nothing.

bikingfreak, 2023-05-04:
If there were a boot-repair version for the ARM64, I could get it to write the bootloader on the SSD. I also looked for a balena-etcher for arm64 and couldn't find one.

Joshua-Riek, 2023-05-04:
You need to write an OS to your SSD first. You can download etcher here  [balena-etcher-electron_1.7.9+5945ab1f_arm64.deb](https://github.com/futurejones/balena-etcher-arm/releases/download/v1.7.9/balena-etcher-electron_1.7.9+5945ab1f_arm64.deb). To install type `sudo dpkg -i balena-etcher-electron_1.7.9+5945ab1f_arm64.deb`

bikingfreak, 2023-05-04:
I also wondered if I couldn't reinstall Ubuntu from scratch using your
repository and the image you supplied. Regular Ubuntu distros offer the
choice of a routine install or allowing you to choose which drive to
install on and how to partition the drive.

bikingfreak, 2023-05-04:
Oh, that's exactly what I was looking for. I will try that as soon as I can
- in about an hour or so.

Many thanks!

Joshua-Riek, 2023-05-04:
Right, having something like that can be very complex due to the customization factors and software limitations. It's not really something i'm planning on creating. I'm trying to stay close to the official Raspberry Pi Ubuntu OS which has a similar installation process. 

However, I expect between 2 and 3 years before an official Ubuntu distro could work on this device, and i would still expect bugs.

bikingfreak, 2023-05-04:
Joshua,

Brilliant! Many thanks  - that worked perfectly. It was exactly what I was
looking for. Now I can install all my apps again (including synaptic) and
get my environment set up. BTW, I did not need to install Bitwarden - I
just needed to install the browser extension.

Once again, my accolades to you. I don't know if you received my email
directly to your Verizon account a few days ago applauding your effort.

It would be helpful to others if you added the balena etcher link on your
Github wiki. I don't know if you're aware of this useful web posting,
below, but it also did not work in installing the SSD software.
https://www.crosstalksolutions.com/orange-pi-5-simple-overview-and-installation-with-m-2-ssd/

Kevin Jaspan

Joshua-Riek, 2023-05-04:
Thanks! I might include Etcher in the image by default to make it easy for future users. 

I'm glad that I could help. This project is a lot of fun and I have learned so much already, I only hope to continue and improve upon what i have here!

cantalupo555, 2023-05-18:
> I have copied your command into the terminal and run it at least 3 times. When I then immediately shut down the Orange Pi, remove the SD card and power up again, it does not boot. So are you saying [...]

@Joshua-Riek 

To use the dd command correctly it is necessary to unzip the "XZ".

`sudo xz -d ubuntu*.img.xz
`

`sudo dd bs=1M if=ubuntu-*-orangepi5.img of=/dev/nvme0n1 status=progress
`

Joshua-Riek, 2023-05-18:
It can also be done in a nice one-liner:
```
xz -dc ubuntu-22.04.2-preinstalled-desktop-arm64-orangepi5.img.xz | sudo dd of=/dev/nvme0n1 bs=4k
```

bikingfreak, 2023-05-25:
Joshua,
On an unrelated issue, I bought, in my wisdom, a Brostrend AC1L WiFi
dongle, which supposedly had support for almost all Linux distros,
including Ubuntu 22.04, but got the error when loading the headers:
ERROR (dkms apport):
*kernel package linux-headers-5.10.110-orangepi-rk3588 is not supported*
Error! Bad return status for module build on kernel:
5.10.110-orangepi-rk3588 (a
arch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
dpkg: error processing package rtl88x2bu-dkms (--configure):
 installed rtl88x2bu-dkms package post-installation script subprocess
returned e
rror exit status 10
Errors were encountered while processing:
 rtl88x2bu-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
Aborting, command failed: apt install --yes --no-install-recommends
./rtl88x2bu-dkms.deb

So, they don't support kernel 5.10. Do you have any idea how I can solve
this? I have sent a message to support at Brostrend. Maybe I'll have to
return the dongle - which means I still don't have WiFi on the Orange Pi,
unless I can get a bridge to work.

Regards and sorry to bother you,

Kevin

bikingfreak, 2023-05-25:
However, this is weird. It seems I can connect to WiFi:
lshw -C network
  *-network:0
       description: Ethernet interface
       physical id: 14
       logical name: eth0
       serial: 62:39:75:5c:b5:5c
       size: 100Mbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt-fd 100bt-fd 1000bt-fd
autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=st_gmac
driverversion=Jan_2016 duplex=full link=yes multicast=yes port=twisted pair
speed=100Mbit/s
  *-network:1
       description: Wireless interface
       physical id: 15
       bus info: ***@***.***:1
       logical name: wlx347de44037a5
       serial: 34:7d:e4:40:37:a5
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl88x2bu
driverversion=5.10.110-orangepi-rk3588 ip=192.168.1.24 multicast=yes
wireless=IEEE 802.11AC

Joshua-Riek, 2023-05-25:
No worries, the 5.10 kernel used by this system is not technically 5.10 on a typical Linux system. Also, a lot of drivers are still in development. However, I'm slowly adding more wifi support; for instance, I released a kernel update earlier today with rtl8811cu and rtl8821cu support.

I can't debug or add every wifi module because it's not feasible with the current kernel, I would recommend using a known working wifi adapter such as the one below. 

https://www.amazon.com/Bluetooth-Adapter-Wireless-External-Receiver/dp/B07YDFZWT8/ref=sr_1_4?crid=3CQDAAZ2ZIQY7&keywords=RTL8821CU&qid=1684974699&s=electronics&sprefix=rtl8821cu%2Celectronics%2C158&sr=1-4

Joshua-Riek, 2023-05-25:
Try to enter a command like `ping www.google.com` and see if it works.

bikingfreak, 2023-05-25:
Good idea - why didn't I think of that? If I disconnect the Ethernet cable
I cannot connect - and it also shows in gmail. If I connect the Ethernet I
can ping Google. So, you're right - no Wifi. Let's see what Tech support
says but if they cannot help I'll send the dongle back and get the one you
recommended. Funny, I chose the BrosTrend because it was top of the list
here:
https://www.fosslinux.com/46681/linux-compatible-wireless-network-adapters.htm

Thanks, Joshua.

Kevin

bikingfreak, 2023-05-29:
Joshua,
The adapter I originally bought would not work, even though I had extensive
Tech Support. There were always dkms errors.
I bought the adapter you recommended above and it is not obvious to me how
to install the drivers. I looked at your dialogue with Flavio, who happens
to live near here, but see no resolution there. Please point me to the
installation procedure. The Chinese instructions for Windows don't help,
nor do I see Linux instructions on the CD.

Thanks in advance.
Kevin

bikingfreak, 2023-05-29:
On this website:
https://www.bestlinuxdistro.com/install-rtl8821cu-ubuntu-linux/
after running

apt update && sudo apt install git build-essential

I get errors:

Setting up rtl88x2bu-dkms (5.13.1-20b5) ...
Removing old rtl88x2bu-5.13.1 DKMS files...
Deleting module rtl88x2bu-5.13.1 completely from the DKMS tree.
Loading new rtl88x2bu-5.13.1 DKMS files...
Building for 5.10.110-orangepi-rk3588
Building initial module for 5.10.110-orangepi-rk3588
ERROR (dkms apport): kernel package linux-headers-5.10.110-orangepi-rk3588 is no
t supported
Error! Bad return status for module build on kernel: 5.10.110-orangepi-rk3588 (a
arch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
dpkg: error processing package rtl88x2bu-dkms (--configure):
 installed rtl88x2bu-dkms package post-installation script subprocess returned e
rror exit status 10
Errors were encountered while processing:
 rtl88x2bu-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

This dkms error happens over and over.

bikingfreak, 2023-05-30:
For example:

apt install linux-headers-$(uname -r)

apt install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-headers-5.10.110-orangepi-rk3588 is already the newest version
(5.10.110+20230406.git0d1781e7-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up rtl88x2bu-dkms (5.13.1-20b5) ...
Removing old rtl88x2bu-5.13.1 DKMS files...
Deleting module rtl88x2bu-5.13.1 completely from the DKMS tree.
Loading new rtl88x2bu-5.13.1 DKMS files...
Building for 5.10.110-orangepi-rk3588
Building initial module for 5.10.110-orangepi-rk3588
ERROR (dkms apport): kernel package linux-headers-5.10.110-orangepi-rk3588 is no
t supported
Error! Bad return status for module build on kernel: 5.10.110-orangepi-rk3588 (a
arch64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
dpkg: error processing package rtl88x2bu-dkms (--configure):
 installed rtl88x2bu-dkms package post-installation script subprocess returned e
rror exit status 10
Errors were encountered while processing:
 rtl88x2bu-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

bikingfreak, 2023-05-30:
The make log ends like this:

ERROR: modpost: module 88x2bu uses symbol kernel_write from namespace
VFS_intern
al_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
ERROR: modpost: module 88x2bu uses symbol kernel_read from namespace
VFS_interna
l_I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
ERROR: modpost: module 88x2bu uses symbol filp_open from namespace
VFS_internal_
I_am_really_a_filesystem_and_am_NOT_a_driver, but does not import it.
make[2]: *** [scripts/Makefile.modpost:169:
/var/lib/dkms/rtl88x2bu/5.13.1/build
/Module.symvers] Error 1
make[2]: *** Deleting file
'/var/lib/dkms/rtl88x2bu/5.13.1/build/Module.symvers'
make[1]: *** [Makefile:1816: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.110-orangepi-rk3588'
make: *** [Makefile:2506: modules] Error 2

Joshua-Riek, 2023-05-30:
You don't need to install the drivers for the above wifi adapter. It will work out of the box. If you were trying to install some wifi drivers yourself it most likely messed something up.

Joshua-Riek, 2023-05-30:
I will have a new release coming in 1-2 hours that you might have better luck with, I made a lot of changes to the kernel and added support for additional wifi adapters.

bikingfreak, 2023-05-30:
Oh, great. No pressure on you. That device came with no intelligible
instructions, so I've been winging it.

Thanks a ton,

Kevin

Joshua-Riek, 2023-05-30:
I just published a new release and double checked that the above WiFi adapter will work, have a try and let me know if you still have issues!

bikingfreak, 2023-05-30:
So I run Software Updater and choose upgrade to 22.10?

Joshua-Riek, 2023-05-30:
No, you will need to re-download the image from GitHub. 

There is a way to update without re-downloading, but with the massive kernel changes, I could not make it as simple as I usually have implemented. I will be creating a post on how to do this later tonight.

bikingfreak, 2023-05-30:
Joshua,OK. That will be more than a 1 hour project, so I’ll do it within the next few days.Many, many thanks,KevinSent from my iPad

bikingfreak, 2023-05-30:
Joshua,
I've installed the new release on the SD and, before flashing it onto the
SSD I thought I'd check if the WiFi worked but lsusb does not show it at
all, nor does it appear in settings. But if I disconnect Ethernet, I can
ping Google.com and it now appears in settings and I could configure one
network successfully! Wow! Thank you! That was quite a lot of work but much
less than you did. I would need build the SSD next, then restore all my
previous capabilities, passwords .....

Many thanks,

Kevin

amarmashruwala, 2024-07-27:
Hi Guys, i get the following error when irun this command 
Sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
dd: failed to open '/lib/u-boot-orangepi-rk3588/rkspi_loader.img': No such file or directory

Any idea why? thanks

SuperKali, 2024-07-27:
> Hi Guys, i get the following error when irun this command 
> 
> Sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
> 
> dd: failed to open '/lib/u-boot-orangepi-rk3588/rkspi_loader.img': No such file or directory
> 
> 
> 
> Any idea why? thanks 
> 
> 

Hi,

Are you sure that the file rkspi_loader.img is present on the directory /lib/u-boot-orangepi-rk3588/

SuperKali, 2024-07-27:
@amarmashruwala try to look here

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#install-u-boot-to-the-spi-flash

The process is changed.

kn3d4, 2024-07-29:
I have tried to follow these steps but found a fixed red led light after removing the sd card. Not sure if I miss anything. 

My step were:
1 - download Joshua's image and burn it in an uSD
2 - once I am in: sudo ubuntu-rockchip-install /dev/nvme0n1
3 - once finished: sudo u-boot-install-mtd
4 - rebooting

I have also tried scratching all partitions before installing in both nvme and spi. The strange thing is I have no longer a boot partition in sd (looks like something changed in the last months). I used to have it in my previous installation.

mtdblock0    31:0    0    16M  0 disk 
mmcblk1     179:0    0  59.5G  0 disk 
└─mmcblk1p1 179:1    0  59.5G  0 part /
nvme0n1     259:0    0 476.9G  0 disk 
└─nvme0n1p1 259:1    0 476.9G  0 part

kn3d4, 2024-07-29:
I may have found the problem. The UUID is pointing to my uSD UUID. 

/dev/nvme0n1p1: LABEL="desktop-rootfs" UUID="8d303228-50ab-451b-b968-41a1d29c0a0b" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="af1a47af-5e7c-4d2f-820e-d62708d04ff3"
/dev/mmcblk1p1: LABEL="desktop-rootfs" UUID="914b5a2f-c268-4c88-b12b-202c14395bcd" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="primary" PARTUUID="e386384b-fbd9-44da-bfac-4a1f3ef476be"

and found in extlinux.conf

label l0
	menu label Ubuntu 24.04 LTS 6.1.0-1020-rockchip
	linux /boot/vmlinuz-6.1.0-1020-rockchip
	initrd /boot/initrd.img-6.1.0-1020-rockchip
	fdtdir /lib/firmware/6.1.0-1020-rockchip/device-tree/
	
	append root=UUID=914b5a2f-c268-4c88-b12b-202c14395bcd rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory quiet splash plymouth.ignore-serial-consoles

What would be the most suitable way to fix this?

kn3d4, 2024-07-29:
Well, I have solved my issue by reinstalling everything.... so strange. :( I hate when I don't understand it.

thanhtantran, 2025-02-12:
for anyone still have trouble with flasing SPI, the file rkspi_loader.img has moved to /lib/u-boot

`sudo dd if=/lib/u-boot/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc`