Joshua-Riek/ubuntu-rockchip #143: X Error of failed request

cantalupo555, 2023-05-18:
Hello, I'm using version v1.11

I'm trying to run Steam and several games and it still gives me this error:
`X Error of failed request:  BadWindow (invalid Window parameter)
`

Due to this error it is not possible to run any games.
In Armbian it works perfectly and doesn't show this error.

I'm using box86 and box64.

Joshua-Riek, 2023-05-18:
I think this is because the default display manager uses Wayland, not X11. Do you know how to switch between the two?

closed 2023-05-29 by Joshua-Riek

cantalupo555, 2023-06-26:
> I think this is because the default display manager uses Wayland, not X11. Do you know how to switch between the two?

With X11 this log disappeared.
Thank you, this error log did not prevent box64 from working properly.

I went back to Wayland.