Joshua-Riek/ubuntu-rockchip #153: RTL8821CU wifi doesn't seem to work on 1.15

arun-511, 2023-05-20:
Wifi settings page won't show up, however bluetooth settings and it connectivity is fine. `rfkill list` only shows bluetooth and nothing related to wifi. RTL8821CU dongle used to work great on 1.14.

Joshua-Riek, 2023-05-20:
I changed the WiFi and Bluetooth firmware in v1.15. I most likely forgot to include a file. I'll investigate later today.

Joshua-Riek, 2023-05-24:
I added the kernel drivers for rtl8811cu and rtl8821cu chipsets, I just tested them, and it works perfectly. There should be a kernel update sometime tonight or tomorrow that will include the change. Users will be able to update with apt. 

The chipset commit can be found here for future reference:
https://github.com/Joshua-Riek/linux-rockchip/tree/rockchip-5.10.110

closed 2023-05-24 by Joshua-Riek

arun-511, 2023-05-25:
Awesome!! It works perfectly. Thank you so much.