Joshua-Riek/ubuntu-rockchip #155: NVEM not recogniz: BC511 NVME SK Hynix 256GB

sangjela, 2023-05-21:
always thank you for your hard work.

a few days ago, I try Samsung ssd but failed. (https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/141)
next I purchase Hynix ssd and tried it.

I have NVME error and grep kern.log after boot by micro SD.

you can see
... sudo grep nvme kern.log
May 21 18:07:14 sangje-desktop kernel: [    8.010115] nvme nvme0: pci function 0004:41:00.0
May 21 18:07:14 sangje-desktop kernel: [    8.010201] nvme 0004:41:00.0: enabling device (0000 -> 0002)
May 21 18:07:14 sangje-desktop kernel: [    8.020289] nvme nvme0: missing or invalid SUBNQN field.
May 21 18:07:14 sangje-desktop kernel: [    8.025137] nvme nvme0: 8/0/0 default/read/poll queues
May 21 18:07:14 sangje-desktop kernel: [    8.029488]  nvme0n1: p1 p2
May 21 18:07:14 sangje-desktop kernel: [   40.991192] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS read failed (134)
May 21 18:07:14 sangje-desktop kernel: [   43.992405] blk_update_request: I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
May 21 18:07:14 sangje-desktop kernel: [   44.013048] nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)
May 21 18:07:14 sangje-desktop kernel: [   44.064491] nvme nvme0: Removing after probe failure status: -19
...

whole grep txt for end of this article
is this info enough? or must check another things?

have a nice day

---------------------------------- 
sangje@sangje-desktop:/var/log$ sudo grep nvme kern.log
[sudo] password for sangje: 
May 21 18:07:14 sangje-desktop kernel: [    8.010115] nvme nvme0: pci function 0004:41:00.0
May 21 18:07:14 sangje-desktop kernel: [    8.010201] nvme 0004:41:00.0: enabling device (0000 -> 0002)
May 21 18:07:14 sangje-desktop kernel: [    8.020289] nvme nvme0: missing or invalid SUBNQN field.
May 21 18:07:14 sangje-desktop kernel: [    8.025137] nvme nvme0: 8/0/0 default/read/poll queues
May 21 18:07:14 sangje-desktop kernel: [    8.029488]  nvme0n1: p1 p2
May 21 18:07:14 sangje-desktop kernel: [   40.991192] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS read failed (134)
May 21 18:07:14 sangje-desktop kernel: [   43.992405] blk_update_request: I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
May 21 18:07:14 sangje-desktop kernel: [   44.013048] nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)
May 21 18:07:14 sangje-desktop kernel: [   44.064491] nvme nvme0: Removing after probe failure status: -19
May 21 18:07:14 sangje-desktop kernel: [   44.076587] Buffer I/O error on dev nvme0n1, logical block 62514752, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.118281] Buffer I/O error on dev nvme0n1p1, logical block 16256, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.118295] Buffer I/O error on dev nvme0n1p2, logical block 62510320, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.118460] Buffer I/O error on dev nvme0n1p1, logical block 16257, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.118744] Buffer I/O error on dev nvme0n1p1, logical block 16258, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.118888] Buffer I/O error on dev nvme0n1p1, logical block 16259, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.121865] Buffer I/O error on dev nvme0n1p2, logical block 62510320, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.122045] Buffer I/O error on dev nvme0n1p2, logical block 1, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.122125] Buffer I/O error on dev nvme0n1p1, logical block 16256, async page read
May 21 18:07:14 sangje-desktop kernel: [   44.122329] Buffer I/O error on dev nvme0n1p1, logical block 16257, async page read
May 21 18:09:37 sangje-desktop kernel: [    7.977819] nvme nvme0: pci function 0004:41:00.0
May 21 18:09:37 sangje-desktop kernel: [    7.977883] nvme 0004:41:00.0: enabling device (0000 -> 0002)
May 21 18:09:37 sangje-desktop kernel: [    7.987755] nvme nvme0: missing or invalid SUBNQN field.
May 21 18:09:37 sangje-desktop kernel: [    7.991929] nvme nvme0: 8/0/0 default/read/poll queues
May 21 18:09:37 sangje-desktop kernel: [    7.995285]  nvme0n1: p1 p2
May 21 18:09:37 sangje-desktop kernel: [   40.994688] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS read failed (134)
May 21 18:09:37 sangje-desktop kernel: [   43.439122] blk_update_request: I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
May 21 18:09:37 sangje-desktop kernel: [   43.483478] nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)
May 21 18:09:37 sangje-desktop kernel: [   43.534925] nvme nvme0: Removing after probe failure status: -19
May 21 18:09:37 sangje-desktop kernel: [   43.550457] Buffer I/O error on dev nvme0n1, logical block 62514752, async page read
May 21 18:20:53 sangje-desktop kernel: [    7.971642] nvme nvme0: pci function 0004:41:00.0
May 21 18:20:53 sangje-desktop kernel: [    7.971731] nvme 0004:41:00.0: enabling device (0000 -> 0002)
May 21 18:20:53 sangje-desktop kernel: [    7.981847] nvme nvme0: missing or invalid SUBNQN field.
May 21 18:20:53 sangje-desktop kernel: [    7.985525] nvme nvme0: 8/0/0 default/read/poll queues
May 21 18:20:53 sangje-desktop kernel: [    7.988016]  nvme0n1: p1 p2
May 21 18:20:53 sangje-desktop kernel: [   39.287761] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS read failed (134)
May 21 18:20:53 sangje-desktop kernel: [   41.735588] blk_update_request: I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
May 21 18:20:53 sangje-desktop kernel: [   41.776280] nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)
May 21 18:20:53 sangje-desktop kernel: [   41.827531] nvme nvme0: Removing after probe failure status: -19
May 21 18:20:53 sangje-desktop kernel: [   41.843397] Buffer I/O error on dev nvme0n1, logical block 62514752, async page read
May 21 18:20:53 sangje-desktop kernel: [   41.891161] Buffer I/O error on dev nvme0n1p1, logical block 16256, async page read
May 21 18:20:53 sangje-desktop kernel: [   41.891262] Buffer I/O error on dev nvme0n1p1, logical block 16257, async page read
May 21 18:20:53 sangje-desktop kernel: [   41.891333] Buffer I/O error on dev nvme0n1p1, logical block 16258, async page read
[... 10 lines cut ...]
May 21 18:21:53 sangje-desktop kernel: [    7.991799]  nvme0n1: p1 p2
May 21 18:21:53 sangje-desktop kernel: [   40.994094] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS read failed (134)
May 21 18:21:53 sangje-desktop kernel: [   43.508836] blk_update_request: I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
May 21 18:21:53 sangje-desktop kernel: [   43.542743] nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)
May 21 18:21:53 sangje-desktop kernel: [   43.594151] nvme nvme0: Removing after probe failure status: -19
May 21 18:21:53 sangje-desktop kernel: [   43.609510] Buffer I/O error on dev nvme0n1, logical block 62514752, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.650902] Buffer I/O error on dev nvme0n1p1, logical block 16256, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.651088] Buffer I/O error on dev nvme0n1p1, logical block 16257, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.651228] Buffer I/O error on dev nvme0n1p1, logical block 16258, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.651366] Buffer I/O error on dev nvme0n1p1, logical block 16259, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.653927] Buffer I/O error on dev nvme0n1p2, logical block 62510320, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.660697] Buffer I/O error on dev nvme0n1p1, logical block 16256, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.660865] Buffer I/O error on dev nvme0n1p1, logical block 16257, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.661010] Buffer I/O error on dev nvme0n1p1, logical block 16258, async page read
May 21 18:21:53 sangje-desktop kernel: [   43.661154] Buffer I/O error on dev nvme0n1p1, logical block 16259, async page read
[... cut ...]
May 21 18:22:56 sangje-desktop kernel: [   43.900884] blk_update_request: I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
May 21 18:22:56 sangje-desktop kernel: [   43.924700] nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)
May 21 18:22:56 sangje-desktop kernel: [   43.976145] nvme nvme0: Removing after probe failure status: -19
May 21 18:22:56 sangje-desktop kernel: [   43.991758] Buffer I/O error on dev nvme0n1, logical block 62514752, async page read

Joshua-Riek, 2023-05-21:
This is very strange. I tested it with my Silicon Power M.2 PCIe Gen 3 SSD NVMe 128GB, and it works without issue. 

Here is the NVMe I used for reference: https://www.amazon.com/Silicon-Power-NVMe-Gen3x4-SP001TBP34A60M28/dp/B09HMWH1DG?th=1

```
ubuntu@ubuntu-desktop:~$ dmesg | grep -i nvme
[    7.158734] nvme nvme0: pci function 0004:41:00.0
[    7.158801] nvme 0004:41:00.0: enabling device (0000 -> 0002)
[    7.162005] nvme nvme0: missing or invalid SUBNQN field.
[    7.163437] nvme nvme0: allocated 8 MiB host memory buffer.
[    7.170925] nvme nvme0: 8/0/0 default/read/poll queues
[    7.174245]  nvme0n1: p1 p2 p3 p4

Joshua-Riek, 2023-05-21:
I see in your logs `nvme 0004:41:00.0: can't change power state from D3cold to D0 (config space inaccessible)` maybe power issue. Are you using a 5V4A power adapter?

sangjela, 2023-05-21:
@Joshua-Riek 
Thanks for your reply!

I buy set that power supply includes from
https://a.aliexpress.com/_mLw1V4C

Power 5v 4a , yes. But I test on boot it only consumed 1.1A max by simple usb-c W A V Wh tester. 

So I will try test another power supply
And I purchased nvme usb enclosure I will test it too.

If I failed all try, may be my orange pi 's nvme slot corrupted.... (If it possible kill all my nvme s, I don't try no more dangerous things)

... If all hope gone, may be I must try Usb enclosed nvme boot .T-T (it possible...?)

Have a nice day!

Joshua-Riek, 2023-05-22:
I don't think a usb to nvme boot will work. There is an issue when using usb to sata ssd to boot, and I assume it might also have trouble. 

Also, make sure you use a nvme ssd and not a sata ssd, they can look very similar! 

If you think you are using a sata ssd you can try the below command and then reboot.
```
sudo sed -i 's/overlays=/overlays=sata /g' /boot/firmware/ubuntuEnv.txt
```

sangjela, 2023-05-22:
@Joshua-Riek 
Thanks for your kind & detail reply. 

1  This time I use SK Hynix 256GB BC511 NVMe , i think it is NVME, not SATA ( yes, very similar )
https://www.amazon.co.jp/SK-Hynix-256GB-BC511-NVMe/dp/B08VD15MPY

2  USB boot is not good option? I m sad but that another fate... orz
thanks for your inform . 

I Hope my another power supply s try will work. :)

sangjela, 2023-05-24:
always thank you for your hard work.

this time, I update to v1.15 and NVME reconized Now. Thanks for your work. 
( but, wifi dongle dead. (you already written about wifi dongle, kernel on v1.15 ... ) )

I Test nvme SK Hynix 256GB and Samsung PM991, all two reconized well.

I know now my wifi dongle not work. 
(my wifi dongle : https://ja.aliexpress.com/item/1005001632825136.html?spm=a2g0o.order_list.order_list_main.24.21c9585aH7h12u&gatewayAdapt=glo2jpn with bluetooth version )

----------------------------------------------------------------------------
I attatch v1.14 vs v1.15 compare.
(Default 5V4A Power Supply used.)

ubuntu 22.04 v1.14 (wifi work, nvme fail )

sangje@sangje-desktop:/sys/class/net$ ls /sys/class/net
eth0  lo  wlx4401bb884aec
sangje@sangje-desktop:/sys/class/net$ ls -la
total 0
drwxr-xr-x  2 root root 0  1月  1  2021 .
drwxr-xr-x 76 root root 0  1月  1  2021 ..
lrwxrwxrwx  1 root root 0  1月  1  2021 eth0 -> ../../devices/platform/fe1c0000.ethernet/net/eth0
lrwxrwxrwx  1 root root 0  1月  1  2021 lo -> ../../devices/virtual/net/lo
lrwxrwxrwx  1 root root 0  5月 24 14:03 wlx4401bb884aec -> ../../devices/platform/fc880000.usb/usb2/2-1/2-1:1.2/net/wlx4401bb884aec
sangje@sangje-desktop:/sys/class/net$ sudo lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk1     179:0    0 58.9G  0 disk 
├─mmcblk1p1 179:1    0  512M  0 part /boot/firmware
└─mmcblk1p2 179:2    0 58.4G  0 part /
sangje@sangje-desktop:/sys/class/net$ 

ubuntu 22.04 v1.15  (wifi dongle fail, nvme work )

sangje@sangje-desktop:/sys/class/net$ ls /sys/class/net
eth0  lo
sangje@sangje-desktop:/sys/class/net$ ls /etc/netplan/
sangje@sangje-desktop:/sys/class/net$ sudo lsblk
[sudo] password for sangje: 
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1 476.9G  0 disk 
└─sda1        8:1    1 476.9G  0 part /media/sangje/SanDiskXtrm500_1
mtdblock0    31:0    0    16M  0 disk 
mmcblk1     179:0    0  58.6G  0 disk 
├─mmcblk1p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk1p2 179:2    0  58.1G  0 part /
nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0    16M  0 part 
└─nvme0n1p2 259:2    0 238.5G  0 part 
sangje@sangje-desktop:/sys/class/net$

closed 2023-05-24 by sangjela

sangjela, 2023-05-24:
I'm Very Happy ☺️

sangjela, 2023-05-28:
Hi . finally, I boot from NVME success. v1.15 and ubuntu update.

first I use Lan cable and after plug, Ubuntu update work and WiFi working start.
I Flash Boot and... I copy System to NVME by wiki (xz -dc ubuntu-22.04.2-preinstalled-desktop-arm64-orangepi5.img.xz | sudo dd of=/dev/nvme0n1 bs=4k   <- this command ... must format target before? not certainly.). 
at first it not boot without microsd in this point nvme and microsd needs too.

I Think, nvme before formatted by NTFS that is problem....? any way, 
I try another way. by https://jamesachambers.com/orange-pi-5-ssd-boot-guide/
it copy system to nvme by cat
( cat /dev/mmcblk1 > /dev/nvme0n1 )
but when the step change nvme uuid change failed.
So, I etchinig 1.15 image diretly to NVME in windows 11 ( I Think boot flash already done )
Yes, It work after first ubuntu setting. 

I m Very Happy!
I take some picture for it. https://www.clien.net/service/board/park/18103619

Thanks for your work !　♡💛 (⋈◍＞◡＜◍)。✧♡

Have a nice day & take enough rest plz.

pi benchmark
https://pibenchmarks.com/latest/
sudo curl https://raw.githubusercontent.com/TheRemote/PiBenchmarks/master/Storage.sh | sudo bash

(Optional) Enter alias to use on benchmark results.  Leave blank for completely anonymous.
Alias (leave blank for Anonymous): Alien
Result submitted successfully and will appear live on https://pibenchmarks.com within a couple of minutes.

     Category                  Test                      Result     
HDParm                    Disk Read                 365.47 MB/s              <- NVME show much faster than micro SD, but far from on Windows Pc ^^; but Enough to Love.
HDParm                    Cached Disk Read          360.35 MB/s              
DD                        Disk Write                240 MB/s                 
FIO                       4k random read            86050 IOPS (344201 KB/s) 
FIO                       4k random write           38935 IOPS (155741 KB/s) 
IOZone                    4k read                   130039 KB/s              
IOZone                    4k write                  102174 KB/s              
IOZone                    4k random read            39480 KB/s               
IOZone                    4k random write           68105 KB/s               

                          Score: 20504                                       

Compare with previous benchmark results at:
https://pibenchmarks.com/
https://pibenchmarks.com/user/Alien/

How to Check CPU Temperture.
sangje@sangje-desktop:~$ sensors bigcore1_thermal-virtual-0
bigcore1_thermal-virtual-0
Adapter: Virtual device
temp1:        +49.0°C  

Another things , USB 3.0 Hub on Upper Blue USB 
USB-hub > NVME Enclosed Samsung SSD PM991a 128GB
sangje@sangje-desktop:~$ sudo hdparm -Tt /dev/sdb
/dev/sdb:
 Timing cached reads:   9500 MB in  2.00 seconds = 4755.81 MB/sec
 Timing buffered disk reads: 1084 MB in  3.07 seconds = 352.60 MB/sec

USB-hub > Sandisk Extreme 500 500GB
sangje@sangje-desktop:~$ sudo hdparm -Tt /dev/sda
/dev/sda:
 Timing cached reads:   8792 MB in  2.00 seconds = 4400.74 MB/sec
 Timing buffered disk reads: 672 MB in  3.00 seconds = 223.75 MB/sec