Joshua-Riek/ubuntu-rockchip #157: USB C Screen Flickering

peytonicmaster6, 2023-05-21:
I'm using a dual 4k monitor setup with one connected via the HDMI port and another via a USB-C to HDMI cord. The one connected via USB-C flickers black every now and then. I've tried two different USB-C to HDMI cords and both HDMI ports on each monitor with the same effect. I use the same monitors on a different computer with dual hdmi output and there's no issues, so I have a feeling its an issue with the orange pi 5b. Has anyone else experienced similar issues?

Joshua-Riek, 2023-05-21:
There have been some issues when using a USB-C and HDMI display simultaneously. A while back, the display would crash when trying to use both USB-C and HDMI at the same time so I came up with a hack as a workaround https://github.com/Joshua-Riek/ubuntu-rockchip/issues/32. 

I have no idea if the old issue is related, but my point is using the USB-C and HDMI display can cause some weird behaviors.

peytonicmaster6, 2023-05-21:
I see. I'll do some digging around and see if I can find what may be causing the issue.

Joshua-Riek, 2023-05-24:
The official Orange Pi Ubuntu image has a borked GPU driver implementation.

But are you saying that plugging in a USB-C and HDMI monitor at the same time will cause the system to crash still?

peytonicmaster6, 2023-05-24:
Gotcha. 

I've had it happen twice where the monitor plugged in via USB-C will freeze and the HDMI monitor remains active for a minute or so, and then the entire system freezes up and I have to power cycle it. 

And by active I mean I can move the mouse around and interact with programs already open, but I can't try and toggle back to anything open on the other monitor with keyboard shortcuts.

Joshua-Riek, 2023-05-24:
Interesting, there are a few things I can try to edit in the device tree that might help with some of the usb-c display issues. But there is a good chance it won't work.

Joshua-Riek, 2023-05-27:
There was a new kernel released from Rockchip, and it has about 6000 new code commits. I'm testing it now, and after using it for about 5 minutes, I have not had any issues with HDMI and USB-C simultaneously. 

I need to test some of my other boards, but I hope to issue a kernel update / new release sometime this weekend. It should help improve the stability with HDMI and USB-C since there were many vop2 changes.

peytonicmaster6, 2023-05-27:
That's awesome, I look forward to that update!

Out of curiosity is it linux kernel 6.3? I saw that rk3588 support was added in the official linux 6.3 kernel.

Joshua-Riek, 2023-05-27:
No, the 6.3 kernel still has a lot of work to do. I expect it to be complete in about two years. For more information [here](https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md) is a link to the current status matrix.

The kernel is still from Rockchip with the release tag linux-5.10-gen-rkr4, and they updated the version to 5.10.160. However, it's not real 5.10.160 from kernel.org.

Joshua-Riek, 2023-05-30:
I just uploaded a new release with the 5.10.160 kernel, and it has a lot of changes. I have not been able to do extensive testing with the USB-C display, but it might solve this issue.

peytonicmaster6, 2023-06-01:
I did some quick testing and it appears that the new kernel fixes the flickering and hardware acceleration problem! I can now run the dinosaur jumping game in chrome without any problems and haven't encountered any flickering In my limited testing.

I will say that the new kernel introduces a new problem though. Setting the display mode to fractional scaling causes the monitor connected via HDMI to go completely green, which wasn't an issue before. 

I'll also note that if I unplug the HDMI port and try to plug it back in, the system will freeze up, or if boot up with just USB C plugged in and try to plug in HDMI afterwards, it will freeze and the USB C screen will turn white and noisy. This was an issue before, so nothing new. If you want, I can move these two problems to a different issue so you can close the flickering out since that got solved with the new kernel update.

Joshua-Riek, 2023-06-15:
So I have concluded. The USB-C display driver is very messed up in the kernel and far beyond something I can fix. I've been digging through the Rockchip documentation, and I can make it so one or the other works. But if both are plugged in, things will start to crash.

Joshua-Riek, 2023-06-18:
I just sent a kernel pull request to fix some issues with the USB-C display port. Audio and cursor flickering when using the x11 display manager are now fixed. I think the crash might also be fixed, I need to do more testing.

Joshua-Riek, 2023-06-18:
Hmm looks like there is still an issue. I will open up a new issue in relation to the crashing.

closed 2023-06-18 by Joshua-Riek