Joshua-Riek/ubuntu-rockchip #163: OrangePi 5b - install to eMMC?

craigcarnell, 2023-05-27:
I've written ubuntu-22.04.2-preinstalled-server-arm64-orangepi5b.img.xz to my SD card but now realize this isn't what I want to do!

I want to install Ubuntu onto my Orange Pi 5B eMMC drive, replacing Android that is already installed.

I'm using it as a server.

Once I've booted the pre-installed version can I then install onto eMMC? I realize this is a pre-built image and probably not what I wanted.

Could I write the image directly to eMMC somehow? Or could I install Ubuntu 23.04 from a arm version provided by Ubuntu (I doubt that works)

Joshua-Riek, 2023-05-27:
To install Ubuntu onto your emmc, look at this post: https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/130#discussioncomment-5749373.

Any official version of Ubuntu will not work with this device.

craigcarnell, 2023-05-27:
@Joshua-Riek Thanks, just what I was looking for :+1:

closed 2023-05-27 by craigcarnell