Joshua-Riek/ubuntu-rockchip #165: NPU working?? (feature request)

develperbayman, 2023-05-28:
right now im on debian 11 trying different distros but my current project needs the use of the NPU is that currently operating and if not are there plans on incorporating it?

Joshua-Riek, 2023-05-28:
I have not tested at the NPU yet and don't know where to start. Have you found any distro that works with the NPU? And can you provide me an example of usage?

develperbayman, 2023-05-28:
unfortunately i cannot find a example and the interwebs are quite bare this is all i can find the reddit has links to github a few github repos i just know it would greatly enhance my AI and i know how to make use of it in python unfortunately i know very little on the system side of it:  https://www.reddit.com/r/OrangePI/comments/12b3jmj/accessing_the_npu_on_the_orange_pi/

develperbayman, 2023-05-28:
as far as i know the NPU on this board is a seriously overlooked and ignored piece that makes up the most powerful asset on this board

Joshua-Riek, 2023-05-30:
I found this wiki page detailing NPU usage, and it also has some examples. I tested it out on my Rock 5b, and it works. 

https://wiki.indiedroid.us/Nova/npu

artem-zinnatullin, 2023-06-17:
Another real-world use-case for OrangePi NPU is Frigate object detection, it's a software NVR for home security cameras and there is ongoing work to make it work with OrangePi NPU https://github.com/blakeblackshear/frigate/pull/5733

I highly recommend reading the PR discussion and checking the code, it's mostly a matter of having right system packages it seems

closed 2023-06-25 by Joshua-Riek