Joshua-Riek/ubuntu-rockchip #166: here is a possible upgrade script to enable updating installs without overwriting user data

develperbayman, 2023-05-29:
#!/bin/bash

# Download the custom Ubuntu image
wget -O custom_image.iso <URL_to_custom_image>

# Mount the ISO image
sudo mkdir /mnt/custom_image
sudo mount -o loop custom_image.iso /mnt/custom_image

# Extract the contents of the ISO image
sudo cp -r /mnt/custom_image/* /path/to/destination

# Unmount the ISO image
sudo umount /mnt/custom_image
sudo rm -r /mnt/custom_image

# Perform an upgrade on the installed image
sudo apt update
sudo apt upgrade -y

# Restart the system to apply changes (optional)
sudo reboot

develperbayman, 2023-05-29:
apparently thats what github does with #comments ....my bad

EvilOlaf, 2023-05-29:
https://www.markdownguide.org/cheat-sheet/

develperbayman, 2023-05-29:
also maybe for ease of install? maybe include zenity? 

`#!/bin/bash

# Prompt the user to select an installation location
echo "Select an installation location:"
echo "1. /dev/sda (HDD/SSD)"
echo "2. /dev/sdb (HDD/SSD)"
echo "3. /dev/sdc (HDD/SSD)"
echo "4. /dev/nvme0n1 (M.2 storage)"
read -p "Enter your choice: " location

case $location in
  1)
    install_location="/dev/sda"
    ;;
  2)
    install_location="/dev/sdb"
    ;;
  3)
    install_location="/dev/sdc"
    ;;
  4)
    install_location="/dev/nvme0n1"
    ;;
  *)
    echo "Invalid choice. Exiting."
    exit 1
    ;;
esac

# Prompt the user to confirm the installation location
read -p "Install Ubuntu image to $install_location? (y/n): " confirm_location

if [[ $confirm_location != "y" ]]; then
  echo "Installation canceled. Exiting."
  exit 1
fi

# Download the custom Ubuntu image
wget -O custom_image.iso <URL_to_custom_image>

# Mount the ISO image
sudo mkdir /mnt/custom_image
sudo mount -o loop custom_image.iso /mnt/custom_image

# Extract the contents of the ISO image to the chosen location
sudo cp -r /mnt/custom_image/* $install_location

# Unmount the ISO image
sudo umount /mnt/custom_image
sudo rm -r /mnt/custom_image

# Prompt the user to select the bootloader installation option
echo "Select bootloader installation option:"
echo "1. Install bootloader to SPI flash"
echo "2. Skip bootloader installation"
read -p "Enter your choice: " bootloader_option

case $bootloader_option in
  1)
    # Add commands to install the bootloader to the SPI flash
    echo "Installing bootloader to SPI flash..."
    ;;
  2)
    echo "Skipping bootloader installation."
    ;;
  *)
    echo "Invalid choice. Skipping bootloader installation."
    ;;
esac

echo "Installation complete."
`

develperbayman, 2023-05-29:
idk why again .... but piss on it ....anyway i hope it helps spawn ideas

EvilOlaf, 2023-05-29:
Use triple ` for multiple lines of code
```
Will look
like this
then
```

Joshua-Riek, 2023-05-29:
Most changes will be in the kernel, bootloader, or firmware. I have a PPA to automatically distribute updates with a simple `apt upgrade` command. Other changes will be related to the pre-installed applications, which I don't see a need to push out to user-installed systems.

develperbayman, 2023-05-30:
is this PPA operational

Joshua-Riek, 2023-05-30:
Yes, the PPA is included by default. However, the most recent kernel (5.10.160) will require a new install because there are a lot of changes.

Gluk41, 2023-06-04:
> Yes, the PPA is included by default. However, the most recent kernel (5.10.160) will require a new install because there are a lot of changes.

How can I update the kernel without reinstalling the system?

Joshua-Riek, 2023-06-04:
The kernel can be updated with a simple apt update and upgrade.
```bash
sudo apt update
sudo apt upgrade
```

closed 2023-06-04 by Joshua-Riek

Gluk41, 2023-06-04:
[image]

Joshua-Riek, 2023-06-04:
Oh, updating the kernel from 5.10.110 to 5.10.160 does not work. There were too many changes and installing the new kernel would break the system.

Gluk41, 2023-06-04:
How to upgrade to a new version of the system without reinstalling?

Joshua-Riek, 2023-06-04:
Updates will be distributed for the kernel, bootloader, and firmware to bring each install up to the latest version. However, you have an older version which I can no longer distribute kernel updates to.

Gluk41, 2023-06-04:
I understand it. The question is, how do I update an existing installation without reinstalling? I installed the previous version, configured and launched all the services I needed in it. All this took me about a week of work. After reading the annotation to the system, I was sure that you would update the system with a ppa. What should I do now? Reinstall all services for another week? And where is the guarantee that I won't have to do it again for the next update?

Joshua-Riek, 2023-06-04:
I do not guarantee anything, and this is a side project that I have invested a lot of time into. I don't get paid for anything done here, and I'm not obligated to do anything. If you are unhappy with something, feel free to contribute.

I'm trying to offer an alternative to the poorly designed vendor Operating Systems and will give users a familiar Ubuntu experience. The PPA is a way to distribute updates, but I made a few errors that do not allow updating from 5.10.110 to 5.10.160.

Gluk41, 2023-06-04:
You didn't understand - I don't blame you, I really like your work! You are the only one who has done Ubuntu well for Orange Pi and I am happy to use your build! It works great! I just want to find an easy way to upgrade... Is the inability to update related to the Orange PI loader? Can I somehow change it in the EFI build?