Joshua-Riek/ubuntu-rockchip #167: failed to initial qt

shum-elli, 2023-05-29:
i use opencv_python to display the capture,but i failed and got this message
[image]
i have tried some way ,such as change wayland to X11,but it doesn't work,can you give me some advise about it

Joshua-Riek, 2023-05-29:
Make sure the package `qtwayland5` is installed. Then you need to use the `QT_QPA_PLATFORM=wayland` environment variable to accommodate for wayland applications. If that does not work you can also try using `QT_QPA_PLATFORM=xcb`.
```bash
QT_QPA_PLATFORM=wayland python test.py
```

Also could you share your test script so I could try on my end?

shum-elli, 2023-05-29:
thank you for your suggestion,i will test it tommorow