Joshua-Riek/ubuntu-rockchip #175: Unable to boot via eMMC on Orange Pi 5B

phckopper, 2023-06-01:
Hello,

After flashing the `ubuntu-22.04.2-preinstalled-desktop-arm64-orangepi5b.img` image on my Orange Pi 5B's eMMC, using `rkdeveloptool`. I'm unable to boot the device. The board hangs after a couple of seconds and the green status LED flashes two pulses periodically.

Below is the serial port output of the device. It seems like the correct device tree is loaded, but there're lots of Rockchip specific errors. What could be causing that?

```
 �DDR Version V1.08 20220617
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x1 Samsung
CH0 RX Vref:30.7%, TX Vref:22.8%,22.8%
CH1 RX Vref:34.7%, TX Vref:21.8%,21.8%
CH2 RX Vref:34.7%, TX Vref:21.8%,21.8%
CH3 RX Vref:32.7%, TX Vref:19.8%,19.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi-rk3588 (May 26 2023 - 22:06:33)
Trying to boot from MMC1
spl: mmc init failed with error: -123
Trying to boot from MMC2
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(d21d14728d...) + OK
## Checking uboot 0x00200000 ... sha256(d462e36ddf...) + OK
## Checking fdt 0x0034a1c0 ... sha256(cf0060a3cf...) + OK
## Checking atf-2 0x000f0000 ... sha256(c00c7fd75b...) + OK
## Checking atf-3 0xff100000 ... sha256(71c3a5841b...) + OK
## Checking atf-4 0xff001000 ... sha256(2301cf73be...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 121.428 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-391-g856309329:derrick.huang
NOTICE:  BL31: Built : 14:15:50, Jul 18 2022
INFO:    ext 32k is valid
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2112MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 528MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1068MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1560MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09-orangepi-rk3588 (May 26 2023 - 22:06:33 +0000)

Model: Orange Pi 5B
PreSerial: 2, raw, 0xfeb50000
DRAM:  7.7 GiB
Sysmem: init
Relocation Offset: eda2c000
Relocation fdt: eb9f8e88 - eb9fecc8
CR: M/C/I
Using default environment

[... 100 lines cut ...]
[    3.906391] rockchip-dmc dmc: could not find power_model node
[    3.916796] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdab0000-0xfdabffff]
[    3.916841] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdac0000-0xfdacffff]
[    3.916862] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdad0000-0xfdadffff]
[    3.922237] debugfs: Directory 'fdab0000.npu-rknpu' with parent 'vdd_npu_s0' already present!
[    3.949992] RKNPU fdab0000.npu: failed to find power_model node
[    3.950048] RKNPU fdab0000.npu: RKNPU: failed to initialize power model
[    3.950070] RKNPU fdab0000.npu: RKNPU: failed to get dynamic-coefficient
[    6.295387] [dhd] STATIC-MSG) dhd_static_buf_init : 101.10.361.18 (wlan=r892223-20220519-1)
[    6.295550] [dhd] STATIC-MSG) dhd_init_wlan_mem : prealloc ok for index 0: 8131584(7941K)
[    6.322597] [dhd] _dhd_module_init: in Dongle Host Driver, version 101.10.361.20 (wlan=r892223-20220701-3)
[    6.322597] drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd compiled on May 29 2023 at 19:16:14
[    6.322597] 
[    6.324073] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 3, size 139264
[    6.324092] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 4, size 0
[    6.405272] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 7, size 35072
[    6.405304] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 0, size 1776
[    6.407916] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 5, size 65536
[    6.408240] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 19, size 65712
[    6.408256] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 20, size 262144
[    6.408273] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 22, size 65536
[    6.626077] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 9, size 32896
[    7.511875] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 9, size 32896
[   11.058758] dma-pl330 fea30000.dma-controller: fill_queue:226
```

Joshua-Riek, 2023-06-01:
I see that the kernel loaded which is good. Maybe it is having trouble finding the rootfs UUID. Let me flash my opi5b and take a look.

Joshua-Riek, 2023-06-01:
I tested on my opi5b and booting from the eMMC works fine on my end. My guess is something went wrong when using the rkdeveloptool. I would try to boot from the SD Card and then from inside Ubuntu flash your eMMC as shown below.

```bash
wget https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.16/ubuntu-22.04.2-preinstalled-desktop-arm64-orangepi5b.img.xz
xz -dc ubuntu-22.04.2-preinstalled-desktop-arm64-orangepi5b.img.xz | sudo dd of=/dev/mmcblk0 bs=4k
sync
```

phckopper, 2023-06-01:
Thanks for the quick response. I'll try it out and report back later.

closed 2023-06-06 by Joshua-Riek