Joshua-Riek/ubuntu-rockchip #176: Rockchip 5 won't recognize new Samsung Pro NVME

abapst, 2023-06-01:
I have an NVME I want to use as an extra storage mount for now (not boot). Getting this:
`[    9.966572] nvme nvme0: pci function 0000:01:00.0
[    9.966690] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[    9.969904] nvme nvme0: Shutdown timeout set to 10 seconds
[    9.973810] nvme nvme0: 8/0/0 default/read/poll queues
[   46.610560] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0x10
[   46.645114] blk_update_request: I/O error, dev nvme0n1, sector 1953524992 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   46.660634] nvme 0000:01:00.0: enabling device (0000 -> 0002)
[   46.660750] nvme nvme0: Removing after probe failure status: -19
[   46.674032] Buffer I/O error on dev nvme0n1, logical block 244190624, async page read`

Any ideas? I've considered the possibility that the SSD is defective.

Joshua-Riek, 2023-06-01:
I recently tested booting from NVMe in #172 with no issue. I would try a fresh install, a new NVMe, or make sure your power supply is proper.

abapst, 2023-06-01:
- Power supply has 65W so should be plenty
- I just tried a second identical NVME and got the exact same issue as above
  Model: Samsung 980 Pro 1TB MZ-V8P1T0B
- I will try a fresh install later today

Thanks for the help

HSTe, 2023-06-02:
I think its powersupply issues. The power it can take is 5V only so many powersupplies that gives 65W but uses higher voltage will not work.

Joshua-Riek, 2023-06-02:
I tested on my second Rock 5B, and NVMe works. Also I'm using a fixed 5V 4A power supply.

abapst, 2023-06-02:
Wow yet another power issue on my end! Thanks for checking this. The chip didn't come with a power supply so I've been using an old 2.1A laptop charger I had. Going to order a 5V/4A supply as you said and will update here when I get it.

abapst, 2023-06-02:
Btw, how many watts is the charger you used? Most "orange pi" chargers I'm seeing are 20W but Google says the Rock needs up to 36W with an SSD

gdoteof, 2023-06-03:
these things do great with power over ethernet (not the board directly, unfortunately, but these things + a poe switch is nice)
[image]

closed 2023-06-06 by Joshua-Riek

abapst, 2023-06-08:
Update on this: I tried a 5V/4A power supply as recommended and the NVME mounted without any problems. Thanks for the help!

artem-zinnatullin, 2023-07-02:
Hm, I'm getting severe issues with NVMe over USB-C adapter with 5V/4A power supply no matter which port on OrangePi 5 I plug in to: USB-C, USB-A (Blue, tried both), USB-A (white).

The hardware:

- [NVMe Intel 670p Series M.2 2280 2TB](https://www.amazon.com/gp/product/B08X4XL71S/)
- [NVMe to USB-C adapter SSK USB 3.2](https://www.amazon.com/gp/product/B07MNFH1PX/) 
- [Power Supply iUniker 20W 5V 4A](https://www.amazon.com/gp/product/B097P2NLVH/)
- [USB-C hub with PD plugged into it as separate test Anker 555 USB-C Hub](https://www.amazon.com/gp/product/B087QZVQJX/)

NVMe adapter starts flashing blue instead of constant light (google & amazon review search showed no one with same issue) and drive is not detected by `fdisk -l`, while `lsblk` does sometimes show it but with size 0B despite drive being partitioned and formatted with `ext4`

```ls
lsblk -o NAME,FSTYPE,SIZE,MOUNTPOINT,LABEL
NAME        FSTYPE   SIZE MOUNTPOINT     LABEL
sda                    0B
sdb                    0B
sdc                    0B (#drive in question! shows 0B)
mtdblock0             16M
nvme0n1            119.2G
├─nvme0n1p1 vfat     512M /boot/firmware system-boot
└─nvme0n1p2 ext4   118.7G /              writable
```

I've tried unplugging everything optional: USB Zigbee Stick and gpio fan.

---

What does work but not reliably is plugging the NVMe via separate USB Hub with PD power supply plugged into it, however after reboot I drive is often not recognized. When I plug it into my macbook it all works every time.

And what's worse over the USB-C hub with all the right cables USB 3.2 I only get 7000KiB/s direct read/write performance! tested with `fio` (that's over USB-C port on OrangePi 5)

```js
[global]
size=2G
direct=1
ioengine=libaio
rw=randrw

[partition]
filename=/mnt/sdc1/testfile

---

Run status group 0 (all jobs):
   READ: bw=7475KiB/s (7655kB/s), 7475KiB/s-7475KiB/s (7655kB/s-7655kB/s), io=1023MiB (1073MB), run=140165-140165msec
  WRITE: bw=7487KiB/s (7666kB/s), 7487KiB/s-7487KiB/s (7666kB/s-7666kB/s), io=1025MiB (1075MB), run=140165-140165msec

Disk stats (read/write):
  sdc: ios=261468/265189, merge=0/61, ticks=58664/55045, in_queue=113710, util=99.24%
```

I'm running ubuntu-rockchip v1.18, will try v1.19 soon

artem-zinnatullin, 2023-07-02:
So here is what I've found, hope it saves someone hours I've wasted

1. I've tried plugging into all ports again and doing `lsusb -v | less` to find info on the USB connection status to the NVMe
2. Did same thing on my macbook with `system_profiler SPUSBDataType`, the drive was connected fine and the mA power draw is what's important! `Current Available (mA): 900          Current Required (mA): 896`
3. I could get the drive to show up with `lsusb` on OrangePi5 only via Type-C to Type-C connection, but it wasn't reading any capacity on it and what turns out is that device was capped at 500mA according to `lsusb -v`!
4. With USB-C hub and PD connected to it I found that again it only worked with one combination of ports and cables I have at home most of which are rated for PD100 and USB3.2, go figure, once I found correct combo Linux also showed `896mA` and drive immediately got recognized by `fdisk -l`!

I don't know what's the root cause is, but I think OrangePi5 is completely under-delivering the power on my 5V 4A brick with just system NVMe plugged in and minimal CPU load.

artem-zinnatullin, 2023-07-02:
Oh and to add to this `fio` sequential reads and writes w/o `direct=1` are showing reasonable results over USB-C hub:

buffered sequential sync **reads**:

```js
Run status group 0 (all jobs):
   READ: bw=218MiB/s (228MB/s), 218MiB/s-218MiB/s (228MB/s-228MB/s), io=2048MiB (2147MB), run=9408-9408msec

Disk stats (read/write):
  sdc: ios=8180/204, merge=0/3, ticks=14002/399, in_queue=14402, util=98.75%
```

(Change `rw` to `read` and add `direct=1` for your tests)
```js
size=2G
ioengine=sync
runtime=30s

[sequential-write]
rw=write
filename=/mnt/sdc1/seq_write_testfile"
```