Joshua-Riek/ubuntu-rockchip #180: boot from usb, but still bring up nvme?

gdoteof, 2023-06-03:
I tried following your guide about booting from usb; but I think you are really only talking about a sata-over-usb rather than usb flash drive?   i should probably learn u-boot.  

i was able to get it to boot from usb by setting the root partition by id; but then the nvme doesn't show up.  

```sh
# lsblk
NAME      MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda         8:0    1 119.5G  0 disk 
├─sda1      8:1    1   512M  0 part /boot/firmware
└─sda2      8:2    1   119G  0 part /
mtdblock0  31:0    0    16M  0 disk 
```

```sh
# cat /boot/firmware/ubuntuEnv.txt 
bootargs=root=UUID=e9486bee-cebe-4c05-8a07-c4aa133eff16 rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 systemd.unified_cgroup_hierarchy=0 
fdtfile=rk3588s-orangepi-5.dtb
overlay_prefix=orangepi-5
overlays=sata
```

```sh
# fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mtdblock0: 16 MiB, 16777216 bytes, 32768 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: AB932B75-593F-4E50-8805-D3E0834EF882

Device           Start   End Sectors  Size Type
/dev/mtdblock0p1    64  7167    7104  3.5M Linux filesystem
/dev/mtdblock0p2  7168  7679     512  256K Linux filesystem
/dev/mtdblock0p3  7680  8063     384  192K Linux filesystem
/dev/mtdblock0p4  8064  8127      64   32K Linux filesystem
/dev/mtdblock0p5  8128  8191      64   32K Linux filesystem
/dev/mtdblock0p6  8192 16383    8192    4M Linux filesystem
/dev/mtdblock0p7 16384 32734   16351    8M Linux filesystem

Disk /dev/sda: 119.51 GiB, 128320801792 bytes, 250626566 sectors
Disk model: Flash Drive FIT 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 7C96D236-0413-4301-A684-77FC87DC7FB3

Device       Start       End   Sectors  Size Type
/dev/sda1    32768   1081343   1048576  512M Linux extended boot
/dev/sda2  1081344 250626532 249545189  119G Linux filesystem
```

gdoteof, 2023-06-03:
closing because my example is broken, will re-open if i can get back to what i thought was happening.

Joshua-Riek, 2023-06-03:
If you have a sata ssd plugged into the m.2 slot then you need to follow the sata instructions. The system can only use NVMe or SATA, not both at the same time. If you are using a sata ssd over USB its not required as it's a USB device.

Joshua-Riek, 2023-06-03:
I updated the Orange Pi 5 wiki page with a lot more information, and it should clear up some confusion. 

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5

closed 2023-06-03 by Joshua-Riek