Joshua-Riek/ubuntu-rockchip #186: WireGuard

billzhong, 2023-06-06:
Hi,

I am trying to install WireGuard but failed:

```
# apt install wireguard
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  build-essential bzip2 cpp-12 dctrl-tools dkms dpkg-dev fakeroot g++ g++-11 gcc-12 libalgorithm-diff-perl libalgorithm-diff-xs-perl
  libalgorithm-merge-perl libasan8 libdpkg-perl libfakeroot libfile-fcntllock-perl libgcc-12-dev libjansson4 libnftables1 libnftnl11
  libstdc++-11-dev libtsan2 linux-headers-5.15.0-73 linux-headers-5.15.0-73-generic linux-headers-generic lto-disabled-list nftables
  wireguard-dkms wireguard-tools
Suggested packages:
  bzip2-doc gcc-12-locales debtags menu debian-keyring gcc-11-doc gcc-12-doc bzr libstdc++-11-doc firewalld openresolv | resolvconf
The following NEW packages will be installed:
  build-essential bzip2 cpp-12 dctrl-tools dkms dpkg-dev fakeroot g++ g++-11 gcc-12 libalgorithm-diff-perl libalgorithm-diff-xs-perl
  libalgorithm-merge-perl libasan8 libdpkg-perl libfakeroot libfile-fcntllock-perl libgcc-12-dev libjansson4 libnftables1 libnftnl11
  libstdc++-11-dev libtsan2 linux-headers-5.15.0-73 linux-headers-5.15.0-73-generic linux-headers-generic lto-disabled-list nftables
  wireguard wireguard-dkms wireguard-tools
0 upgraded, 31 newly installed, 0 to remove and 0 not upgraded.
Need to get 168 MB of archives.
After this operation, 606 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 cpp-12 arm64 12.1.0-2ubuntu1~22.04 [61.7 MB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libasan8 arm64 12.1.0-2ubuntu1~22.04 [2,392 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libtsan2 arm64 12.1.0-2ubuntu1~22.04 [2,508 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libgcc-12-dev arm64 12.1.0-2ubuntu1~22.04 [1,169 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 gcc-12 arm64 12.1.0-2ubuntu1~22.04 [70.0 MB]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libdpkg-perl all 1.21.1ubuntu2.2 [237 kB]                         
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 bzip2 arm64 1.0.8-5build1 [34.6 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 lto-disabled-list all 24 [12.5 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 dpkg-dev all 1.21.1ubuntu2.2 [922 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libstdc++-11-dev arm64 11.3.0-1ubuntu1~22.04.1 [2,075 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 g++-11 arm64 11.3.0-1ubuntu1~22.04.1 [11.1 MB]
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 g++ arm64 4:11.2.0-1ubuntu1 [1,394 B]
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 build-essential arm64 12.9ubuntu3 [4,740 B]
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 dctrl-tools arm64 2.24-3build2 [65.2 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 dkms all 2.8.7-2ubuntu2.1 [70.0 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libjansson4 arm64 2.13.1-1.1build3 [33.6 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libnftnl11 arm64 1.2.1-1build1 [64.5 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libnftables1 arm64 1.0.2-1ubuntu3 [323 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 nftables arm64 1.0.2-1ubuntu3 [67.4 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libfakeroot arm64 1.28-1ubuntu1 [31.5 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 fakeroot arm64 1.28-1ubuntu1 [60.5 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libalgorithm-diff-perl all 1.201-1 [41.8 kB]
Get:23 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libalgorithm-diff-xs-perl arm64 0.04-6build3 [11.7 kB]                   
Get:24 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libalgorithm-merge-perl all 0.08-3 [12.0 kB]
Get:25 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libfile-fcntllock-perl arm64 0.22-3build7 [33.7 kB]
Get:26 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-5.15.0-73 all 5.15.0-73.80 [12.3 MB]
Get:27 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-5.15.0-73-generic arm64 5.15.0-73.80 [2,531 kB]
Get:28 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 linux-headers-generic arm64 5.15.0.73.71 [2,312 B]
Get:29 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 wireguard-dkms all 1.0.20210606-1 [279 kB]
Get:30 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 wireguard-tools arm64 1.0.20210914-1ubuntu2 [92.5 kB]
Get:31 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 wireguard all 1.0.20210914-1ubuntu2 [3,114 B]
Fetched 168 MB in 28s (6,022 kB/s)     
Extracting templates from packages: 100%
Selecting previously unselected package cpp-12.
(Reading database ... 64011 files and directories currently installed.)
Preparing to unpack .../00-cpp-12_12.1.0-2ubuntu1~22.04_arm64.deb ...
Unpacking cpp-12 (12.1.0-2ubuntu1~22.04) ...
Selecting previously unselected package libasan8:arm64.
Preparing to unpack .../01-libasan8_12.1.0-2ubuntu1~22.04_arm64.deb ...
Unpacking libasan8:arm64 (12.1.0-2ubuntu1~22.04) ...
Selecting previously unselected package libtsan2:arm64.
Preparing to unpack .../02-libtsan2_12.1.0-2ubuntu1~22.04_arm64.deb ...
Unpacking libtsan2:arm64 (12.1.0-2ubuntu1~22.04) ...
Selecting previously unselected package libgcc-12-dev:arm64.
Preparing to unpack .../03-libgcc-12-dev_12.1.0-2ubuntu1~22.04_arm64.deb ...
[... 102 lines cut ...]
Setting up dctrl-tools (2.24-3build2) ...
Setting up libalgorithm-merge-perl (0.08-3) ...
Setting up libnftables1:arm64 (1.0.2-1ubuntu3) ...
Setting up nftables (1.0.2-1ubuntu3) ...
Setting up g++-11 (11.3.0-1ubuntu1~22.04.1) ...
Setting up dpkg-dev (1.21.1ubuntu2.2) ...
Setting up libgcc-12-dev:arm64 (12.1.0-2ubuntu1~22.04) ...
Setting up g++ (4:11.2.0-1ubuntu1) ...
update-alternatives: using /usr/bin/g++ to provide /usr/bin/c++ (c++) in auto mode
Setting up build-essential (12.9ubuntu3) ...
Setting up gcc-12 (12.1.0-2ubuntu1~22.04) ...
Setting up dkms (2.8.7-2ubuntu2.1) ...
Setting up wireguard-dkms (1.0.20210606-1) ...
Loading new wireguard-1.0.20210606 DKMS files...
Building for 5.10.160-rockchip-rk3588
Building initial module for 5.10.160-rockchip-rk3588
Error! The /var/lib/dkms/wireguard/1.0.20210606/5.10.160-rockchip-rk3588/aarch64/dkms.conf for module wireguard includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.
This indicates that it should not be built.
Skipped.
Setting up wireguard (1.0.20210914-1ubuntu2) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1+3) ...
```

Any idea how to workaround it? Thanks.

Joshua-Riek, 2023-06-06:
It might be built into the kernel, what is the output of the command `modinfo wireguard`?

billzhong, 2023-06-06:
```
# modinfo wireguard
filename:       /lib/modules/5.10.160-rockchip-rk3588/kernel/drivers/net/wireguard/wireguard.ko
alias:          net-pf-16-proto-16-family-wireguard
alias:          rtnl-link-wireguard
version:        1.0.0
author:         Jason A. Donenfeld <Jason@zx2c4.com>
description:    WireGuard secure network tunnel
license:        GPL v2
srcversion:     F577A9071D7F4D757D53E2C
depends:        libcurve25519-generic,ipv6,udp_tunnel,libchacha20poly1305,ip6_udp_tunnel
intree:         Y
name:           wireguard
vermagic:       5.10.160-rockchip-rk3588 SMP mod_unload modversions aarch64
```

Joshua-Riek, 2023-06-06:
I see, I checked the kernel config and saw that wireguard is built as a module. You should be able to load it with `sudo modprobe wireguard` I don't believe downloading the wireguard package is required.

closed 2023-06-06 by billzhong