Joshua-Riek/ubuntu-rockchip #190: The /dev/video0 device does not exist in 1.17.

ironiris, 2023-06-08:
The /dev/video0 device does not exist in 1.17.(orange pi 5 plus)
Is it possible to create one?

Joshua-Riek, 2023-06-08:
Weird, I don't have time to look into this right now, but will when I have time later today. Since /dev/video0 shows when you tried the rock5b image on the opi5 plus, I guess this might be related to the device tree.

Joshua-Riek, 2023-06-08:
I see the issue, `hdmirx_ctrler` is disabled in the device tree by default. 

However enabling it will start to flood the [kernel log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11685340/dmesg.log).
```
[  128.697344] fdee0000.hdmirx-controller: hdmirx_wait_lock_and_get_timing HDMI pull out, return!
[  128.769708] rk_hdmirx fdee0000.hdmirx-controller: hdmirx_audio_interrupts_setup: 1
```

I will need to do further investigation.

Joshua-Riek, 2023-06-08:
I came up with a patch for the above-flooded logs. It's a hack, but Orange Pi did the same in their kernel. I'll send a pull request to get this merged and fixed.
https://github.com/Joshua-Riek/linux-rockchip/commit/f612a4330d33fd226764ac050d487fa87869cff4
https://github.com/Joshua-Riek/linux-rockchip/commit/523cacf66535b537fbc2c178db1cc61a4e4035c8

Joshua-Riek, 2023-06-09:
The changes have been merged, and I'm uploading the changes to my PPA now. A new kernel should be available to download with a simple `apt update && apt upgrade` in about 2-3 hours, which will fix this issue.

closed 2023-06-09 by Joshua-Riek