Joshua-Riek/ubuntu-rockchip #194: Wireless connection problem

Natisfaction, 2023-06-11:
Yesterday I tried to install the 1.17 version, but then I realized that my wifi USB adapter wasn't working. In the previous version till 1.14 (the latest I was using before installing 1.17) the adapter was working perfectly. The adapter is tp link TL-WN722N.

Joshua-Riek, 2023-06-11:
There were massive changes between those two versions. The kernel was updated to 5.10.160, and some firmware has changed. Could you provide me with the kernel logs? I think it may relate to missing firmware. 

If you could please do the following then upload the file kernel.txt:
1. Plug in your WiFi USB adapter. 
2. Reboot your system.
3. Login and open up a terminal window.
4. Enter the command `dmesg > kernel.txt` to dump kernel logs into the kernel.txt file.

Natisfaction, 2023-06-11:
Sure, thank you for your response, I really appreciate your work.

Here's the log:
`[    1.645268] kernel: Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    1.645294] kernel: Linux version 5.10.160-rockchip (root@bos02-arm64-003) (gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Wed Jun 7 21:10:50 UTC 2023
[    1.655743] kernel: Machine model: Orange Pi 5
[    1.655899] kernel: efi: UEFI not found.
[    1.659430] kernel: OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
[    1.659448] kernel: OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
[    1.659537] kernel: Reserved memory: created CMA memory pool at 0x0000000010000000, size 256 MiB
[    1.659541] kernel: OF: reserved mem: initialized node cma, compatible id shared-dma-pool
[    1.928916] kernel: Zone ranges:
[    1.928923] kernel:   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    1.928934] kernel:   DMA32    empty
[    1.928940] kernel:   Normal   [mem 0x0000000100000000-0x00000003ffefffff]
[    1.928948] kernel: Movable zone start for each node
[    1.928951] kernel: Early memory node ranges
[    1.928956] kernel:   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    1.928964] kernel:   node   0: [mem 0x0000000100000000-0x00000003fbffffff]
[    1.928973] kernel:   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]
[    1.928979] kernel: Initmem setup node 0 [mem 0x0000000000200000-0x00000003ffefffff]
[    1.928985] kernel: On node 0 totalpages: 4126720
[    1.928990] kernel:   DMA zone: 15352 pages used for memmap
[    1.928995] kernel:   DMA zone: 0 pages reserved
[    1.928999] kernel:   DMA zone: 982528 pages, LIFO batch:63
[    1.929004] kernel:   Normal zone: 49148 pages used for memmap
[    1.929009] kernel:   Normal zone: 3144192 pages, LIFO batch:63
[    2.024564] kernel: On node 0, zone Normal: 256 pages in unavailable ranges
[    2.024740] kernel: psci: probing for conduit method from DT.
[    2.024751] kernel: psci: PSCIv1.1 detected in firmware.
[    2.024755] kernel: psci: Using standard PSCI v0.2 function IDs
[    2.024761] kernel: psci: MIGRATE_INFO_TYPE not supported.
[    2.024768] kernel: psci: SMC Calling Convention v1.2
[    2.025201] kernel: percpu: Embedded 32 pages/cpu s93480 r8192 d29400 u131072
[    2.025243] kernel: pcpu-alloc: s93480 r8192 d29400 u131072 alloc=32*4096
[    2.025251] kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
[    2.025389] kernel: Detected VIPT I-cache on CPU0
[    2.025438] kernel: CPU features: detected: GIC system register CPU interface
[    2.025444] kernel: CPU features: detected: Virtualization Host Extensions
[    2.025455] kernel: CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    2.025468] kernel: alternatives: patching kernel code
[    2.025853] kernel: Built 1 zonelists, mobility grouping on.  Total pages: 4062220
[    2.025863] kernel: Kernel command line: root=UUID=e62eafd0-2540-4bf1-bca1-55d018f422ae rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 systemd.unified_cgroup_hierarchy=0 quiet splash plymouth.ignore-serial-consoles
[    2.028002] kernel: Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes, linear)
[    2.028775] kernel: Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    2.028782] kernel: mem auto-init: stack:off, heap alloc:off, heap free:off
[    2.035097] kernel: software IO TLB: mapped [mem 0x00000000ec000000-0x00000000f0000000] (64MB)
[    2.182482] kernel: Memory: 15809764K/16506880K available (18624K kernel code, 3748K rwdata, 7088K rodata, 7232K init, 670K bss, 434972K reserved, 262144K cma-reserved)
[    2.182633] kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    2.182670] kernel: ftrace: allocating 69650 entries in 273 pages
[    2.298855] kernel: ftrace: allocated 273 pages with 3 groups
[    2.299164] kernel: rcu: Hierarchical RCU implementation.
[    2.299170] kernel: rcu:         RCU event tracing is enabled.
[    2.299175] kernel:         Rude variant of Tasks RCU enabled.
[    2.299179] kernel:         Tracing variant of Tasks RCU enabled.
[    2.299185] kernel: rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    2.304773] kernel: NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    2.307772] kernel: GICv3: GIC: Using split EOI/Deactivate mode
[    2.307779] kernel: GICv3: 480 SPIs implemented
[    2.307783] kernel: GICv3: 0 Extended SPIs implemented
[    2.307814] kernel: GICv3: Distributor has no Range Selector support
[    2.307823] kernel: GICv3: 16 PPIs implemented
[    2.307870] kernel: GICv3: CPU0: found redistributor 0 region 0:0x00000000fe680000
[    2.308001] kernel: ITS [mem 0xfe640000-0xfe65ffff]
[    2.308056] kernel: ITS@0x00000000fe640000: allocated 8192 Devices @1001f0000 (indirect, esz 8, psz 64K, shr 0)
[    2.308085] kernel: ITS@0x00000000fe640000: allocated 32768 Interrupt Collections @100200000 (flat, esz 2, psz 64K, shr 0)
[    2.308096] kernel: ITS: using cache flushing for cmd queue
[    2.308138] kernel: ITS [mem 0xfe660000-0xfe67ffff]
[    2.308185] kernel: ITS@0x00000000fe660000: allocated 8192 Devices @100220000 (indirect, esz 8, psz 64K, shr 0)
[... cut ...]
[  177.316861] kernel: squashfs: version 4.0 (2009/01/31) Phillip Lougher
[  177.509816] kernel: dma-pl330 fea30000.dma-controller: pl330_submit_req:1737 Try increasing mcbufsz (258/256)
[  177.509822] kernel: dma-pl330 fea30000.dma-controller: fill_queue:2263 Bad Desc(4)
[  182.042647] kernel: ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
[  182.052207] kernel: [BT_RFKILL]: ENABLE UART_RTS`

I tried installing the rtl8188eus drivers, and by now it seems to be working as before. Don't know if this can be helpful tho.

Joshua-Riek, 2023-06-11:
Awesome, glad you where able to get it working! Ill close this for now, but I will try and re-visit when I have time to add the drivers by default.

closed 2023-06-11 by Joshua-Riek

jeredbrent, 2023-06-20:
Similar issue here and having issues with rtl8188eus drivers... are you installing from kelebek333?
Can you post linky to the drivers you found to work?  Thanks!

Natisfaction, 2023-06-21:
Sure, I found them in the aircrack-ng GitHub: https://github.com/aircrack-ng/rtl8812au

Just read the instructions and should work like a charm.

portavales, 2023-07-26:
It seems I have the same wifi adapter as Natisfaction.
(this one: 
https://www.amazon.com/Orange-Pi-Support-Compatible-Computers/dp/B0BZRNM6HR/?_encoding=UTF8&pd_rd_w=ueEuj&content-id=amzn1.sym.5f7e0a27-49c0-47d3-80b2-fd9271d863ca%3Aamzn1.symc.e5c80209-769f-4ade-a325-2eaec14b8e0e&pf_rd_p=5f7e0a27-49c0-47d3-80b2-fd9271d863ca&pf_rd_r=M5PDA41CS715MH0FXJMX&pd_rd_wg=YrZZk&pd_rd_r=598f03b1-c6f5-4a80-8d19-59dc3a638e5d&ref_=pd_gw_ci_mcx_mr_hp_atf_m )

I am having a similar problem but I can't seem to fix it.

Here is the relevant part of the dmesg:

```
[    8.306623] rk_gmac-dwmac fe1c0000.ethernet: Using 32 bits DMA width
[    8.429749] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x3
[    8.447529] usbcore: registered new interface driver rndis_wlan

[    8.447535] =======================================================
[    8.447537] ==== Launching Wi-Fi driver! (Powered by Rockchip) ====
[    8.447539] =======================================================
[    8.447542] Realtek 8821CS SDIO WiFi driver (Powered by Rockchip,Ver v5.14.2-28-g6011b0372.20220328) init.
[    8.447545] [WLAN_RFKILL]: rockchip_wifi_power: 1
[    8.447548] [WLAN_RFKILL]: rockchip_wifi_power: rfkill-wlan driver has not Successful initialized
[    8.447551] RTW: module init start
[    8.447553] RTW: rtl8821cs v5.14.2-28-g6011b0372.20220328
[    8.447556] RTW: build time: Jul 25 2023 12:52:31
[    8.447558] RTW: rtl8821cs BT-Coex version = COEX20210319-5555
[    8.447590] RTW: module init ret=0
[    8.447609] usbcore: registered new interface driver asix
[    8.447625] usbcore: registered new interface driver cdc_ether

```

I tried the aircrack-ng GitHub: https://github.com/aircrack-ng/rtl8812au

```
$ git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
$ sudo make dkms_install

mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=5.10.160-rockchip KSRC=/lib/modules/5.10.160-rockchip/build...................
cleaning build area...
dkms install -m 8812au -v 5.6.4.2_35491.20191025

88XXau.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.160-rockchip/updates/dkms/

depmod....
dkms status -m 8812au
8812au/5.6.4.2_35491.20191025, 5.10.160-rockchip, aarch64: installed

$lsmod | grep 88
ipv6                  438272  88

$ sudo modprobe 88XXau
$ lsmod | grep 88
88XXau               2236416  0
ipv6                  438272  88

```
Still after that, can't see the wifi adapter with ifconfig or iwconfig

portavales, 2023-07-26:
Nevermind, my adapter is different from the original post.
My card is AP6275P
Found this:
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5

More especifically:
sudo sed -i 's/overlays=/overlays=ap6275p /g' /boot/firmware/ubuntuEnv.txt