Joshua-Riek/ubuntu-rockchip #195: Ver 1.17: NVME SSD on Orange Pi 5 runs at half speed (PCIe Gen1?)

dilibom, 2023-06-11:
Yesterday I installed version 1.17 on my Orange Pi 5.
Found that the NVME SSD (Netac G930ES 512 Gb) was running at half speed:
`
sudo hdparm -t-direct /dev/nvme0n1p2

/dev/nvme0n1p2:
 Timing O_DIRECT disk reads: 572 MB in  3.00 seconds = 190.53 MB/sec
`
[kernel.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11714374/kernel.txt)
On official Ubuntu Jammy image, this benchmark reports 360-390 MB/sec.
Is it possible, that the PCIe interface is limited to only Gen 1?
But SSD support PCIe Gen 3 and RK3588s - PCIe Gen 2.

Joshua-Riek, 2023-06-11:
The RK3588S uses pcie 2.0 speeds. Write speeds of 200 MB/sec are expected. 

Last I checked, hdparm is meant for hard drives, not NVMe SSDs, and can give skewed results.

dilibom, 2023-06-11:
PCIe 2.0 x1 expected parameters: 5 GT/s with 500 MB/sec limit (https://minerstat.com/help/what-is-the-difference-between-pcie-gen-1-pcie-gen-2-and-pcie-gen-3).
In attached file kernel.txt, on 541 line says:
[    4.773173] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)
Standard Ubuntu utility "Disks" (https://help.ubuntu.com/stable/ubuntu-help/disk-benchmark.html.en) report 205 MB/sec, which roughly corresponds to hdparm result.
Usual disk speed on RK3588s is 360-380 MB/sec (for example - this video: https://www.youtube.com/watch?v=dDIMJowwpZk)

Joshua-Riek, 2023-06-11:
I see. My guess is this happened when I updated the kernel from 5.10.110 to 5.10.160. I'll go back to some older builds and take a look at the speeds.

dilibom, 2023-06-11:
Thank you.
I boot the image with the official kernel 5.10.110 - the kernel log says:
[    6.932612] pci 0004:41:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0004:40:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)

[kernel_xulong_minimal_5.10.110.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11714481/kernel_xulong_minimal_5.10.110.txt)

Joshua-Riek, 2023-06-11:
Hmm, I do see the issue for sure. This might take a while to fix. The 5.10.110 and 5.10.160 differ by over 6000 commits, and I don't know much about PCIe bandwidth in the kernel. But ill see what I can find.

Joshua-Riek, 2023-06-11:
I talked with another dev, and we found out the issue. Having the wireless and bluetooth device tree nodes enabled by default will conflict with the PCIe lane, lowering the bandwidth. I'll be able to submit a patch sometime later today to resolve this issue.

Joshua-Riek, 2023-06-13:
The fix has been merged into the kernel, and I'll release a kernel update in a few days. Users will be able to get the change with a simple apt update and apt upgrade.

dilibom, 2023-06-16:
Release 1.18 (cli ubuntu-server): problem not resolved.
hdparm report 190 Mb/s.

Joshua-Riek, 2023-06-16:
I just checked on my end, and I see no issue. hdparm is for hard drives, not for NVMe drives, it could be showing invalid results

[image]

dilibom, 2023-06-16:
Yes, on Ubuntu desktop release 1.18 i got the same results and hdparm report 360 Mb/s. Everything is fine.
But on the server version, the results are 2 times worse. I use dmesg for kernel log, PCI-E still with 2,5 GT/s (gen.1 only):
[    4.764004] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)
Could it be that desktop and server images work differently? My kernel log in attachment
[kernel.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11772746/kernel.txt)
.

Joshua-Riek, 2023-06-16:
It should be the exact same. Make sure you did not download the Orange Pi 5B image. Ill test the server image later today.

Joshua-Riek, 2023-06-16:
It looks fine on my end. 

```
ubuntu@ubuntu:/mnt$ uname -a
Linux ubuntu 5.10.160-rockchip #3 SMP Thu Jun 15 01:13:53 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:/mnt$ dmesg | grep -i "pci 0004:41:00.0"
[    7.423403] pci 0004:41:00.0: [1987:5012] type 00 class 0x010802
[    7.423482] pci 0004:41:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    7.424063] pci 0004:41:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0004:40:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    7.445791] pci 0004:41:00.0: BAR 0: assigned [mem 0xf4200000-0xf4203fff 64bit]
ubuntu@ubuntu:/mnt$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0    16M  0 disk 
mmcblk1     179:0    0 119.4G  0 disk 
├─mmcblk1p1 179:1    0   512M  0 part /boot/firmware
└─mmcblk1p2 179:2    0 118.9G  0 part /
nvme0n1     259:0    0 476.9G  0 disk 
├─nvme0n1p1 259:1    0   256M  0 part 
└─nvme0n1p2 259:2    0   6.4G  0 part /mnt
ubuntu@ubuntu:/mnt$ sync; dd if=/dev/zero of=tempfile bs=1M count=1024; sync
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 2.6922 s, 399 MB/s
ubuntu@ubuntu:/mnt$ sudo /sbin/sysctl -w vm.drop_caches=3
vm.drop_caches = 3
ubuntu@ubuntu:/mnt$  dd if=tempfile of=/dev/null bs=1M count=1024
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 3.04935 s, 352 MB/s
```

dilibom, 2023-06-16:
```
ubuntu@zuzi:~$ uname -a
Linux zuzi 5.10.160-rockchip #3 SMP Thu Jun 15 01:13:53 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@zuzi:~$ dmesg | grep -i "pci 0004:41:00.0"
[    4.763120] pci 0004:41:00.0: [1f40:2263] type 00 class 0x010802
[    4.763251] pci 0004:41:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    4.764004] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)
[    4.776531] pci 0004:41:00.0: BAR 0: assigned [mem 0xf4200000-0xf4203fff 64bit]
ubuntu@zuzi:~$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0    16M  0 disk
nvme0n1     259:0    0 476,9G  0 disk
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0 476,4G  0 part /
ubuntu@zuzi:~$ sync; dd if=/dev/zero of=tempfile bs=1M count=1024; sync
1024+0 records in
1024+0 records out
1073741824 bytes (1,1 GB, 1,0 GiB) copied, 5,86203 s, 183 MB/s
ubuntu@zuzi:~$ sudo /sbin/sysctl -w vm.drop_caches=3
vm.drop_caches = 3
ubuntu@zuzi:~$ dd if=tempfile of=/dev/null bs=1M count=1024
1024+0 records in
1024+0 records out
1073741824 bytes (1,1 GB, 1,0 GiB) copied, 5,25199 s, 204 MB/s
```
 Can you send your dmesg log? I will compare and look for the problem myself.

dilibom, 2023-06-16:
Perhaps it depends on the SSD model or on the boot device. If you boot from an SD card, then the PCIe speed will be full.

Joshua-Riek, 2023-06-16:
Let me flash the os to my NVMe; ill give that a try.

watou, 2023-06-16:
Maybe unrelated and different board, but this commit from March 30 has a description that caught my eye: "rockchip: dts: rk3588 rock 5b: enable pcie30 for NVMe boot"

https://github.com/radxa/u-boot/commit/ed87342b06f67f82123becd0b3e4d90811a8456d

Joshua-Riek, 2023-06-16:
My results are a little strange. But the key thing I see is `2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1`. I will re-open this issue and do some testing.

```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.10.160-rockchip #3 SMP Thu Jun 15 01:13:53 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:~$ dmesg | grep -i "pci 0004:41:00.0"
[    4.863072] pci 0004:41:00.0: [1987:5012] type 00 class 0x010802
[    4.863160] pci 0004:41:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    4.863777] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    4.879138] pci 0004:41:00.0: BAR 0: assigned [mem 0xf4200000-0xf4203fff 64bit]
ubuntu@ubuntu:~$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0    16M  0 disk 
nvme0n1     259:0    0 476.9G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0 476.4G  0 part /
ubuntu@ubuntu:~$ dd if=/dev/zero of=tempfile bs=1M count=1024 oflag=dsync
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 6.0902 s, 176 MB/s
ubuntu@ubuntu:~$ sudo /sbin/sysctl -w vm.drop_caches=3
vm.drop_caches = 3
ubuntu@ubuntu:~$ dd if=tempfile of=/dev/null bs=1M count=1024 oflag=dsync
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 3.0423 s, 353 MB/s
```

Joshua-Riek, 2023-06-16:
```dtb
```

dilibom, 2023-06-16:
The dd command sometimes produces too large results for me. Sometimes when writing it was 670 Mb/s, sometimes when reading (caches?). Therefore I use hdparm -t-direct.

Joshua-Riek, 2023-06-17:
I think this is by design. If I enable PCIe20x1_1 it will conflict with the on-board USB _3.0._
[image]

Joshua-Riek, 2023-06-17:
I checked out the latest Ubuntu image from Orangepi, which has the same issue.
```bash
orangepi@orangepi5:~$ uname -a
Linux orangepi5 5.10.110-rockchip-rk3588 #1.1.6 SMP Thu Jun 1 21:23:54 CST 2023 aarch64 aarch64 aarch64 GNU/Linux
orangepi@orangepi5:~$ dmesg | grep -i "pci 0004:41:00.0"
[    5.002070] pci 0004:41:00.0: [1987:5012] type 00 class 0x010802
[    5.002173] pci 0004:41:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    5.002853] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    5.024023] pci 0004:41:00.0: BAR 0: assigned [mem 0xf4200000-0xf4203fff 64bit]

```

dilibom, 2023-06-17:
Yes, I checked - the same.
Unclear. Why then, in the case of booting from an SD card, USB3 is available and NVME is full speed? And only in the case of booting from NVME do we have a problem.
There was no such problem on the old kernel.
Is it possible to revert to the old 5.10.110 kernel while leaving current root partition?

Joshua-Riek, 2023-06-17:
It's possible, but I think this issue was always there. Will need to go back to older versions to test.

Joshua-Riek, 2023-06-17:
This issue seems to also be on the old 5.10.110 kernel for sure.
```bash
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.10.110-orangepi-rk3588 #1 SMP Thu May 4 02:31:47 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:~$ dmesg | grep -i "pci 0004:41:00.0"
[    4.886169] pci 0004:41:00.0: [1987:5012] type 00 class 0x010802
[    4.886257] pci 0004:41:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    4.886874] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[    4.902276] pci 0004:41:00.0: BAR 0: assigned [mem 0xf4200000-0xf4203fff 64bit]
```

dilibom, 2023-06-17:
It's strange that this PCIe gen. reduction only works when booting from NVME. Perhaps the problem is in the bootloader on the SPI NOR FLASH, in which something happens due to the boot order. I will try to look about the issue at [xunlong repo](https://github.com/orangepi-xunlong/)

theflyingape, 2023-07-05:
> The dd command sometimes produces too large results for me. Sometimes when writing it was 670 Mb/s, sometimes when reading (caches? compression?). Therefore I use hdparm -t-direct.

Yes, did you use its `dsync` option, i.e., a simple 210mb partition copy into a file yields:
```
$ sudo dd if=/dev/sda1 of=boot.img status=progress
209715200 bytes (210 MB, 200 MiB) copied, 2.79608 s, 75.0 MB/s

$ sudo dd if=/dev/sda1 of=boot.img bs=4k oflag=dsync status=progress
209715200 bytes (210 MB, 200 MiB) copied, 181.185 s, 1.2 MB/s
```

Joshua-Riek, 2023-07-20:
I did some research into RK3588 PCIe, and this is by design. We only have half the available bandwidth here. So I will be closing this for now.

However, I have no idea why we get the full bandwidth when not booting from the NVMe.

closed 2023-07-20 by Joshua-Riek