Joshua-Riek/ubuntu-rockchip #196: What is the password for the www-data user?

iRaizenn, 2023-06-11:
Or I want the password for the root user

Joshua-Riek, 2023-06-11:
www-data is not a default user or group from my understanding. Maybe you need to create the user?

iRaizenn, 2023-06-12:
> www-data is not a default user or group from my understanding. Maybe you need to create the user?

I am using a web server that NEEDS to execute a php file with the www-data user

closed 2023-06-15 by Joshua-Riek

iRaizenn, 2023-06-17:
> Try running your php file like this `sudo -u www-data php script.php`, the password should be the same as root.

no it is not.

Joshua-Riek, 2023-06-17:
I have the same result for the www-data user on my Ubuntu laptop. You need to configure the www-data user yourself. 
```bash
joshua@GS66-Stealth-11UE:~$ uname -a
Linux GS66-Stealth-11UE 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
joshua@GS66-Stealth-11UE:~$ sudo passwd -S www-data
www-data L 08/19/2021 0 99999 7 -1
```

iRaizenn, 2023-06-18:
www-data is for Apache web servers only. you just created a new user named www-data

Joshua-Riek, 2023-06-18:
I did not create any www-data user to my knowledge. If there are user permission issues, delete the user and create a new one with the expected permissions you require.