Joshua-Riek/ubuntu-rockchip #198: OrangePI5Plus audio input error.

ironiris, 2023-06-12:
In Orange Pi 5 Plus official Ubuntu,
There is a sound input option in Settings - Sound - Input - Input Device
but in Ubuntu-rockchip(orange pi 5 plus), there is nothing.

Thank you.

Joshua-Riek, 2023-06-12:
Hmmm, is this for HDMI IN? There should be HDMI0, HDMI1, HDMI-In, DP0, and ES8388 options available.

Joshua-Riek, 2023-06-12:
Or do you mean the microphone?

arun-511, 2023-06-12:
Yeah, there is no entry for the onboard microphone. The input field is empty until a device is added.

[image]
[image]

But this is an excepted behaviour, right?

ironiris, 2023-06-12:
First of all, Orange Pi 5 Plus has a built-in microphone
and audio jack input*output.
If I run arecord command, The output is shown below.
`arecord -l
**** CAPTURE 하드웨어 장치 목록 ****
3 카드: rockchipes8388 [rockchip,es8388], 0 장치: dailink-multicodecs ES8323.7-0011-0 [dailink-multicodecs ES8323.7-0011-0]
  하위장치: 0/1
  하위장치 #0: subdevice #0
4 카드: rockchiphdmiin [rockchip,hdmiin], 0 장치: fddf8000.i2s-dummy_codec hdmiin-dc-0 [fddf8000.i2s-dummy_codec hdmiin-dc-0]
  하위장치: 1/1
  하위장치 #0: subdevice #0
`

However, if I try the command below to use video and sound together, it will say busy.
`gst-launch-1.0 v4l2src device=/dev/video0 io-mode=dmabuf ! video/x-raw,format=NV12,width=3840,height=2160,framerate=60/1 ! videoscale ! video/x-raw,width=2560,height=1440,framerate=60/1,key-int-max=4 ! mpph264enc ! h264parse ! mux. alsasrc device=hw:4,0 ! voaacenc bitrate=128000 ! aacparse ! mux. flvmux streamable=true name=mux ! queue ! rtmpsink location="rtmp://a.rtmp.youtube.com/live2/xy27-su6m-gs83-hkhj-036e"
Setting pipeline to PAUSED ...
ERROR: from element /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: Could not open audio device for recording. Device is being used by another application.
Additional debug info:
../ext/alsa/gstalsasrc.c(785): gst_alsasrc_open (): /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0:
Device 'hw:4,0' is busy
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...`

Joshua-Riek, 2023-06-12:
I did a quick test using an old hack of alsa-ucm-conf that I used a while ago for the Opi5 audio (no longer used because it's a hack and had some other issues). The microphone is detected on the Opi5 / 5B but not the Opi5 Plus. I will need to look into this more.

njvergaraa, 2023-06-21:
With headphones plugged into the jack on PI5 Plus no sound is played either. In the PI5a there are no problems, everything is ok.

Could it be part of the same bug?

Joshua-Riek, 2023-06-21:
Most likely. There are some audio configs from OrangePi that I need to play with. I have not had much time in the past few days, and work has been very busy. Thanks for letting me know the audio jack also has an issue, I would not have thought to test it.

Joshua-Riek, 2023-06-22:
I did a quick test with my old [v1.13](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.13) release, and the microphone and audio jack both work. I will need to re-work a few things to fix this.

ironiris, 2023-06-22:
It would be nice to be able to hear the sound coming in through the HDMI IN as well.
Is there any way to do this?

Joshua-Riek, 2023-06-22:
I don't know. I need time to reverse engineer how Orange Pi has their sound system setup in their distro.

cantalupo555, 2023-06-25:
[image]

@Joshua-Riek Here the problem is solved.

Joshua-Riek, 2023-06-25:
Wow! How was it solved???

cantalupo555, 2023-06-25:
> Wow! How was it solved???

Previously had this problem that was reported.
Nowadays my INPUT works both in Armbian and Ubuntu.
I didn't have to make any changes.

njvergaraa, 2023-06-25:
> > Wow! How was it solved???
> 
> Previously had this problem that was reported. Nowadays my INPUT works both in Armbian and Ubuntu. I didn't have to make any changes.
> 
> *Latest versions of the system (kernel).

@cantalupo555 How did you update the kernel ?... an apt upgrade does not bring me anything

cantalupo555, 2023-06-25:
> @cantalupo555 How did you update the kernel ?... an apt upgrade does not bring me anything

It was necessary to reinstall the operating system to the latest available version.

https://github.com/Joshua-Riek/ubuntu-rockchip/releases
or
https://www.armbian.com/orangepi-5/

Joshua-Riek, 2023-06-25:
This is very weird, and I have not updated anything to fix this.

cantalupo555, 2023-06-26:
> This is very weird, and I have not updated anything to fix this.

Yes, I'm just using the latest kernel version you implemented.
Problem here has been resolved.

cantalupo555, 2023-06-27:
Telegram
[image]

Discord
[image]

njvergaraa, 2023-06-28:
@Joshua-Riek Work OK... Mic and Headset. Thanks!!!

Joshua-Riek, 2023-06-28:
Oh, I was confused, I thought this was for the on-board microphone for some reason haha. But yes the audio jack is now fixed and audio input including headset and microphones will now work.

closed 2023-06-28 by Joshua-Riek

bmecdev, 2023-07-05:
[image]
no input device on orange pi 5, the onboard microphone is not detect.