Joshua-Riek/ubuntu-rockchip #200: Approximate effort of supporting Ubuntu 20 again on OPi 5

RuhanSA079, 2023-06-13:
Hello Joshua,

I'm really interested in this project of yours, and I used your Jammy image before (Ubuntu Server), but since I kinda need Ubuntu 20.04 for building snaps natively, I downloaded your older Ubuntu 20 server image, which works well. (Had to install snapd as a apt package)

I don't have a need for a desktop, as I only use the CPU's raw power to compile kernels and snaps.
What effort would it be to get Ubuntu 20 Server supported again as a release-build?

Joshua-Riek, 2023-06-13:
I don't plan on adding support for Ubuntu 20.04 again. Already nearing my limits with personal computational power and storage resources now that I have support for nine boards. 

Can you use a docker image or deboostrap and chroot to create an emulated Ubuntu 20.04 environment for compilation? 

It's not that hard to create a 20.04 image yourself, I would be happy to help in doing this. But the hard part is the desktop and GPU-related drivers, which it sounds like you do not need.

RuhanSA079, 2023-06-14:
Hello,

No need, I am currently using an older release, so no need to recreate/add support for this board for Ubuntu 20 Server/Desktop.
I just updated the software components, installed few missing packages, but nothing apt can't sort out.

Great project by the way.
Closing issue.
All the best.

closed 2023-06-14 by RuhanSA079