Joshua-Riek/ubuntu-rockchip #202: Rock 5B: v1.17 image on SD card will boot NVMe if Armbian installed on it

watou, 2023-06-13:
With the SPI NOR flash erased, v1.17 ubuntu-rockchip image on SD Card and Armbian 23.02 on NVMe, Armbian is booted. Expected ubuntu-rockchip to be booted.  With SD card removed and NVMe left installed, reboot does not boot anything, as expected.  This means that ubuntu-rockchip image on the SD card is facilitating the booting of the NVMe.

This prevents switching to ubuntu-rockchip, except maybe for those who have some other way to erase their NVMe cards.

Joshua-Riek, 2023-06-13:
I updated u-boot and added the pr to adjust boot priority. I will give this a test later today.

Joshua-Riek, 2023-06-13:
I just finished testing the change, and it works as expected, booting from the sdcard first. I'll be coming out with a new release in a few days that will include this change. Thanks for letting me know about this.

watou, 2023-06-13:
I'm very happy with that result; many thanks!

watou, 2023-06-15:
Just to report that v1.18 on my Rock5B and NVMe worked exactly as described.  Thank you!