Joshua-Riek/ubuntu-rockchip #206: [OPi 5 Plus] NVME unsafe_shutdowns count keeps increasing every reboot

arun-511, 2023-06-15:
Device: OPi 5 Plus
System: v1.18

`nvme smart-log /dev/nvme0n1` reported `unsafe_shutdowns` count is incrementing with every reboot (both power-on and soft) by 1. `power_cycles` count also increments by 2. This behaviour is only limited to **Orange Pi 5 Plus.**

OPi 5 on the other hand is fine with no `unsafe_shutdowns` increment and `power_cycles` count increments by 1 only on power on reboot.

I have tested with 2 NVME ssds both report the same behaviour.

1. Samsung p991a (Gen 3.0 x4)
2. Crucial P5 Plus (Gen 4.0 x4)

I also tested these ssds with my laptop and they report normaly like OPi 5.

Joshua-Riek, 2023-06-15:
Very interesting find. I can compare this actually with the Rock 5B and NanoPC T6 that I have now. Hopefully, they don't have this issue and I can figure out what the differences are between them and the Opi5+.

Joshua-Riek, 2023-06-16:
I took a quick look at this on my Rock 5B, and the unsafe_shutdowns value does increase every reboot. I have a few more RK3588 devices with NVMe ports I can test. PCIe-related issues are something I'm not familiar with. I can do some basic testing and troubleshooting, but if this is more than a simple fix I might not be able to do much here. I will need help from someone with more knowledge in this area.

arun-511, 2023-06-16:
A Google search brings up a few similar issues, it appears the kernel is not sending a proper shutdown command or waiting a bit before cutting power to the nvme.

While this probably wouldn't do any physical damage to the Controller, it can certainly corrupt data on the drive if a chunk of cached data was pending to be flushed, especially for devices with onboard ram like Crucial P5 Plus. In the long run, this could reduce the disk life as this happens on every reboot.

The `power_cycles` count also seems to increase by 2, not a good sign either.

Joshua-Riek, 2023-06-17:
I've been looking into the NVMe issues and I don't think this is something I have the knowledge to fix. The kernel is very old and originally for Android, so the NVMe not receiving a proper shutdown signal does not surprise me. 

I would expect this to be fixed when the mainline kernel becomes available, however this might be in 1-2 years.

arun-511, 2023-06-17:
I tried official OPi 5 Plus ubuntu images and they all have the same issue. I wonder why it properly shuts down on OPi 5, maybe for the PCIe generation difference?

I searched some forums and similar issues popped up, some resolved with a kernel update, some BIOS and some with nvme firmware update.

I also looked at the nvme speed issue. Like you said this kernel is old and probably wasn't even meant to have a nvme in a android device. Looks like nothing can be done at this point except waiting for the machine kernel availability.

Joshua-Riek, 2023-06-17:
Yeah, I'm just glad to have these issues documented, maybe another developer will come along with an idea on how to solve this.

closed 2023-06-18 by Joshua-Riek