Joshua-Riek/ubuntu-rockchip #208: Kernel 6.3 as alternative?

omikron88, 2023-06-18:
Is it possible to use new kernel 6.3 as alternative?

Joshua-Riek, 2023-06-18:
I have not looked into the mainline 6.x kernel at all. I have been focused on fixing issues with the BSP 5.10.x kernel from Rockchip. I have not looked into it because many features are missing, and in development, I estimate one or two years that we will have a fully stable 6.x kernel available. 

https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md

Joshua-Riek, 2023-06-20:
I might start to release beta builds with the 6.x kernel in 2-3 months. I will need some more storage and computing power. It takes about 3 hours to fully build everything on my server. If I release an image for each board using the 6.x kernel, it will take around 6 hours to build everything.

closed 2023-06-20 by Joshua-Riek

omikron88, 2023-06-20:
This feature Is not needed for everyone.
Just include the script, that installs needed packages, downolads source
tree, builds the kernel and makes the packages.

The build runs quickly on devices with SSD.

Joshua-Riek, 2023-06-26:
Rockchip just uploaded a 6.1 kernel. This will be interesting to look at.
https://gitlab.com/rk3588_linux/rk/kernel/-/tree/develop-6.1?ref_type=heads