Joshua-Riek/ubuntu-rockchip #209: missing linux-modules-extra package

drakan725, 2023-06-18:
Hello,

i wanted to ask if it would a lot of hassle to add package linux-modules-extra for rockchip?
problem is that mainland Ubuntu release is using kernel 5.15.0 and we are using 5.10. so it have to be built sepearatly because we cannot use one from Ubuntu repos.

Thanks for your work and have nice day 😄

Joshua-Riek, 2023-06-18:
I have been trying to get the kernel similar to what Ubuntu has for its kernel packaging. But I can't create the same thing or use the same kernel config to build the modules because the Rockchip kernel is a mess and would require much effort. 

The kernel is technically based on 2.6 with many Rockchip patches, making it hell to deal with. Most people like me are only trying to fix essential kernel bugs while we wait for mainline 6.x to become available and stable for rk3588. I estimate this will take 1-2 years.

Joshua-Riek, 2023-06-19:
I will be working on improving the kernel packaging. It will take time, but I will try and make it close to Ubuntu's mainline kernel packaging within reason.

closed 2023-06-19 by Joshua-Riek