Joshua-Riek/ubuntu-rockchip #210: USB not working

awir3, 2023-06-18:
Im using NanoPC T6 , both USB-C and USB-A not working cant use keyboard or mouse

Joshua-Riek, 2023-06-18:
I'll look at this now.

Joshua-Riek, 2023-06-18:
On my NanoPC T6 both USB-C and USB-A ports are working and can find my mouse and keyboard. Have you test if the two ports work using another OS?

awir3, 2023-06-18:
[image]

awir3, 2023-06-18:
> On my NanoPC T6 both USB-C and USB-A ports are working and can find my mouse and keyboard. Have you test if the two ports work using another OS?

Both USB-C and USB-A work just fine with another OS. It seems that both ports are disconnected nothing work

Joshua-Riek, 2023-06-18:
Both USB-C and USB-A work fine for me. Maybe you have a hardware issue with the NanoPC T6?

awir3, 2023-06-18:
> Both USB-C and USB-A work fine for me. Maybe you have a hardware issue with the NanoPC T6?

I've tested different types of official FriendlyElec OS and both USB ports work fine.

Joshua-Riek, 2023-06-18:
I will look at the source code from FriendlyElec and see if I can find any differences. Maybe there are some changes I forgot to add.

awir3, 2023-06-19:
> changes

I just fix it and it works just fine 

Edit the file ubuntuEnv.txt and change rk3588s-nanopi-r6s.dtb to rk3588-nanopc-t6.dtb

Joshua-Riek, 2023-06-19:
I'm glad it's working for you now!

closed 2023-06-19 by Joshua-Riek