Joshua-Riek/ubuntu-rockchip #212: How to configure UART to device connected to rx and tx ports

bulganbaev, 2023-06-19:


Joshua-Riek, 2023-06-19:
Could you provide some information here? What device are you using, and do you want to get a serial console output?

bulganbaev, 2023-06-19:
I am using Orange Pi 5 plus,  I have connected device to ports 6,8 and 10. In Raspberry PI previously i just enabled uart in raspi-config. Is there ways to enable it in Orange pi 5 plus using your image?

Joshua-Riek, 2023-06-19:
You must add a device tree overlay to enable UART for your pins. Please take a look at the wiki below on how to do that.
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

I don't know the pinout of the GPIO, so I don't know which overlay you will need to use. Take a look at the schematics and you should be able to find which overlay you need to use.
[image]

closed 2023-06-23 by Joshua-Riek