Joshua-Riek/ubuntu-rockchip #213: Orange Pi 5B - VLAN in promisc mode not supported

brussell-ac, 2023-06-20:
Hi, I tripped over the following kernel driver bug while trying to set up VLANs on an Orange Pi 5B running release 1.16.

[ 3202.686172] rk_gmac-dwmac fe1c0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[ 3964.457596] 8021q: 802.1Q VLAN Support v1.8
[ 3964.457643] 8021q: adding VLAN 0 to HW filter on device eth0
[ 3964.457667] rk_gmac-dwmac fe1c0000.ethernet eth0: Adding VLAN in promisc mode not supported
[ 3964.460842] rk_gmac-dwmac fe1c0000.ethernet eth0: Adding VLAN in promisc mode not supported

A patch for this is described here. Could it please be added to the next release?

https://forum.armbian.com/topic/26518-network-bridge-fails-to-initialize-vlan-in-promisc-mode-not-supported/

Joshua-Riek, 2023-06-20:
Can you test and see if the patch works? I use a fork of Armbian's kernel, but I'm sending all my changes over with pull requests.

wilb, 2023-07-05:
Encountered this myself - compiled your kernel with the above patch and can confirm the bridge now works :partying_face:

Joshua-Riek, 2023-07-05:
Thank you, I can send a pull request and get the patch merged.

brussell-ac, 2023-07-05:
Thank you both very much - sorry I hadn't done this yet, and I couldn't work out all the steps to test it. Not core to this case, but would be great to have a pointer for build+patch+install OPi kernels for next time.

Joshua-Riek, 2023-07-06:
I would like to improve on my documentation so users can do that easily. 

But the patch has been merged into the kernel. I plan to release a kernel update sometime this week or next. It will be available with a simple apt update and apt upgrade command.

wilb, 2023-07-06:
The majority of the steps are already outlined in the [Kernel wiki](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Kernel) - additional steps to patch & build below if you want to try yourself @brussell-ac 

```
# create a workspace for build 
mkdir -p ~/git && cd ~/git 

# use your favourite text editor to copy / paste patch file into a file called testing.patch (or download directly if available)
git clone -b linux-5.10-gen-rkr4 https://github.com/Joshua-Riek/linux-rockchip.git

# change to the kernel dir and apply the patch
cd linux-rockchip
patch -p1 < ../testing.patch

# build the packages
dpkg-buildpackage -a "$(cat debian/arch)" -d -b -nc -uc

# install the packages
dpkg -i linux-headers-5.10.160-rockchip_5.10.160-5_arm64.deb  linux-image-5.10.160-rockchip_5.10.160-5_arm64.deb

# temporarily freeze the packages so they don't get overwritten 
sudo apt-mark hold linux-image-5.10.160-rockchip linux-headers-5.10.160-rockchip
```

brussell-ac, 2023-07-06:
Thank you both - patch is in but I'll go teach myself how to do that anyway for next time :)

closed 2023-07-19 by Joshua-Riek