Joshua-Riek/ubuntu-rockchip #214: Gnome settings panel no longer opens

watou, 2023-06-20:
I was using 1.18 release on Rock 5B with no issues.  I've since performed a sudo apt update + upgrade and installed a few other components including a menu editor, and now the Settings panel will no longer load.  My /var/log/syslog gets lines like these on my attempts:
```
Jun 20 17:49:50 rock5b gnome-shell[1719]: Settings panel for desktop file gnome-bluetooth-panel.desktop could not be loaded!
Jun 20 17:50:42 rock5b gnome-shell[1719]: Settings panel for desktop file gnome-bluetooth-panel.desktop could not be loaded!
Jun 20 17:50:56 rock5b gnome-shell[1719]: Settings panel for desktop file gnome-control-center.desktop could not be loaded!
Jun 20 17:51:02 rock5b gnome-shell[1719]: Settings panel for desktop file gnome-display-panel.desktop could not be loaded!
```

Joshua-Riek, 2023-06-20:
My guess is some of the packages you installed or the menu editor is messing something up.

watou, 2023-06-20:
The problem was indeed not with your image.  In trying to install an AppImage program I was instructed to 
```
sudo apt-get install fuse libfuse2
```
which caused the removal of gnome-control-center .  Sorry for the time wasting!