Joshua-Riek/ubuntu-rockchip #216: [Documentation] Add explainer of this repo vs "official" OrangePi Ubuntu distro

Chambana, 2023-06-20:
The README could use a couple sentences on the differences between this distro and the Ubuntu images that OrangePi links to on their official site [here](https://drive.google.com/drive/folders/1i5zQOg1GIA4_VNGikFl2nPM0Y2MBw2M0).  Or maybe they're the same?

Joshua-Riek, 2023-06-20:
They are **_very_** different distros. I'm trying to make this as close to an actual Ubuntu 22.04 distro that a user might find on the Raspberry Pi4 or a modern Desktop, all within reason of course. 

OrangePi uses an old Armbian fork with a Ubuntu base, everyone's definition of Ubuntu is different. But I wish OrangePi would link my distro on their site, it feels like I'm doing them a major favor with this whole project. I might need to give them a poke about that.

Chambana, 2023-06-20:
This is great context and what you're doing is exactly what people are looking for.  I think most users that land on your README are probably Raspberry Pi 4 users looking for a performance-bump akin to a "Raspberry Pi 5" and found the Orange Pi 5b.  I think a vanilla Ubuntu image is really important for this community as they migrate from the aging Pi4 to the OrangePi 5b.

closed 2023-06-20 by Chambana

Joshua-Riek, 2023-06-20:
I'll work on adding some more information. Thanks for the idea. I was not expecting this project to get a lot of attention as I was only doing this so I could learn more. But now I have more than nine supported boards and more coming.