Joshua-Riek/ubuntu-rockchip #218: Severe network jitter in Moonlight-QT compared to official Orange Pi Ubuntu image on OP 5B

deathbyguitar, 2023-06-23:
When running Moonlight-QT for in-home game streaming, the stream quickly becomes a choppy mess, regularly showing network jitter values around 4-7%, give or take and thus causing lots of stuttering. This is not an issue when running the same via OrangePi's very own Ubuntu image, in-fact Moonlight will show 0% jitter the entire time it's running...so maybe it's a question of what's different networking-wise between this distro and the official one? 

Moonlight-QT that I'm using can be installed from the repo mentioned here.

https://launchpad.net/~liujianfeng1994/+archive/ubuntu/rockchip-multimedia

 If it makes any difference, it's currently set to stream 1440p, 120hz while the TV I'm connected to is displaying in 4k at 120 hz.

Joshua-Riek, 2023-06-23:
There are massive differences between my distro and the Ubuntu image provided by Orange Pi. They have a botched and very broken GPU driver implementation. If you used that ppa on their image, things will break, and you most likely won't realize it. The kernel is also very old from Orange Pi, but it should have minimal impact on networking.

I would try two things. Make sure you use a wired ethernet connection, then dial back your settings to stream at 60Hz. Streaming at above 60Hz is exceptionally unreasonable and will save on computational and networking resources.

deathbyguitar, 2023-06-24:
Thanks for the quick response. 

The Orange Pi can actually handle this sort of thing pretty easily thanks to its powerful GPU. The reason I run it at 120 hz is because my TV runs at that refresh rate and games like Doom can stream from my desktop well over 120 FPS. It makes a massive difference with smoothness when playing games optimized for high framerates. Moonlight 

Are you incorporating different sound drivers or settings in your distro vs the stock Orange Pi Ubuntu release? Yours works perfectly in Moonlight except for little stutters. The stock OS works perfectly except for weird behavior with audio over HDMI. I imagine the latter would be easier for me to find a fix for.

Joshua-Riek, 2023-06-25:
Running a display at 120Hz differs from streaming at 120fps. It would require a high-speed internet connection and a GPU to decode the stream. The RK3588 SOC is made for either 8k@30fps or 4k@60fps. Anything higher could lead to software or performance issues.

The Orange Pi Ubuntu image uses many custom audio configs for pulsaudio and alsa. I used to use them, but it would break when packages like pulseaudio or alsa updated and had some weird audio bugs. So now I use the latest pulseaudio and alsa with a udev rule defined to map audio devices correctly. 

https://github.com/Joshua-Riek/ubuntu-rockchip/blob/39390a65fa69cbae49c895a81a14b69254abd5f8/scripts/config-image.sh#L108-L110

deathbyguitar, 2023-06-26:
Thank you for pointing me in the right direction with those rules in the config-image.sh. I'll see if I can implement something like that in the stock image because it honestly works perfectly for my needs apart from weird HDMI audio bugs. I don't really know what I'm doing but I can certainly learn, especially with the help of some friends who know their way around Linux.

And regarding the streaming, you might be surprised what the Orange Pi is capable of in that regard. I bought it because of the HDMI 2.1 port and the HDMI 2.1 spec specifically supports 4K at 120 HZ. The video signal streamed from my desktop to the Orange Pi via Sunlight/Moonlight is 1440p at 120 HZ and the Orange Pi's system resolution is set to 4k120. Both devices are connected to my router via ethernet. This produces a super fluid gaming experience that's about as nice as connecting my desktop directly to my TV. I promise you - it works and it works well, even moreso than my laptop as a Moonlight client because that could only output 1080p120hz.

Joshua-Riek, 2023-07-02:
My focus is more on system-related issues, I performed some testing, and the network speed seems fine. This looks to be application or use-case specific, which is out of scope for this project. If you want to try and fix the official image I would suggest looking into the alsa audio config files from the BSP and the device tree.

closed 2023-07-02 by Joshua-Riek