Joshua-Riek/ubuntu-rockchip #219: Orange PI 5 Plus Random Freezing

ggdk123, 2023-06-23:
Ubuntu freezes from time to time for no reason... no issue while running stress test, but sudden freezing while doing nothing or just browsing web

ggdk123, 2023-06-23:
Original charger BTW

arun-511, 2023-06-23:
Does the UI freezes and you have to restart manually with the power button?

Joshua-Riek, 2023-06-23:
Can you provide the syslog when the crash occurs?

Joshua-Riek, 2023-06-23:
I'll run my Opi5+ for a few hours playing youtube with Chromium. But if I can not reproduce the crash you are experiencing, I can do nothing.

ggdk123, 2023-06-23:
No issue with playing youtube videos, but with some other basic stuffs....

ggdk123, 2023-06-23:
I got lots of this kind of errors in the system log

==
Jun 23 20:31:30 abcde chromium-browser.desktop[2149]: message repeated 21 times: [ error: Too many contexts created!]
Jun 23 20:34:11 abcde chromium-browser.desktop[2149]: error: Too many contexts created!
Jun 23 20:34:18 abcde chromium-browser.desktop[2149]: message repeated 5 times: [ error: Too many contexts created!]
Jun 23 20:40:27 abcde chromium-browser.desktop[2149]: error: Too many contexts created!
Jun 23 20:40:52 abcde chromium-browser.desktop[2149]: message repeated 65 times: [ error: Too many contexts created!]
Jun 23 20:40:52 abcde chromium-browser.desktop[2150]: [2150:2156:0623/204052.900753:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -100
Jun 23 20:40:52 abcde chromium-browser.desktop[2150]: [2150:2156:0623/204052.905175:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -100
Jun 23 20:40:53 abcde chromium-browser.desktop[2150]: [2150:2156:0623/204053.372371:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -100
Jun 23 20:40:53 abcde chromium-browser.desktop[2150]: [2150:2156:0623/204053.376065:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -100
Jun 23 20:40:56 abcde chromium-browser.desktop[2149]: error: Too many contexts created!
Jun 23 20:43:36 abcde chromium-browser.desktop[2149]: message repeated 73 times: [ error: Too many contexts created!]
Jun 23 20:43:36 abcde chromium-browser.desktop[2150]: [2150:2156:0623/204336.533580:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -100
Jun 23 20:43:37 abcde chromium-browser.desktop[2150]: [2150:2156:0623/204337.023906:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -100
Jun 23 20:43:40 abcde chromium-browser.desktop[2149]: error: Too many contexts created!

ggdk123, 2023-06-23:
> Does the UI freezes and you have to restart manually with the power button?

Yup, I had to hold the power button to manually reboot.

Joshua-Riek, 2023-06-23:
> No issue with playing youtube videos, but with some other basic stuffs....

I need a specific reproducible example. I can't test every scenario by myself I'm maintaining 9 of these boards, with more on the way. Additionally, I need the full system log, the excerpt from your syslog above look normal (normal for rockchip standards which are sadly very low).

ggdk123, 2023-06-24:
it froze again just now while browsing a simple web page in chromium.

Here is the log for the last 1 hour until freezing

==
Jun 24 17:26:53 abcde com.obsproject.Studio.desktop[19593]: info: [pipewire] Stream 0xaaaaf67703d0 state: "connecting" (error: none)
Jun 24 17:26:53 abcde com.obsproject.Studio.desktop[19593]: info: [pipewire] Stream 0xaaaaf67703d0 state: "paused" (error: none)
Jun 24 17:26:53 abcde com.obsproject.Studio.desktop[19593]: info: [pipewire] Stream 0xaaaaf67703d0 state: "streaming" (error: none)
Jun 24 17:29:06 abcde gnome-shell[1631]: libinput error: event8  - Logitech USB Receiver: client bug: event processing lagging behind by 25ms, your system is too slow
Jun 24 17:36:10 abcde ubuntu-appindicators@ubuntu.com[1631]: Impossible to send about-to-show to menu: Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.Failed: error occurred in AboutToShow
Jun 24 17:36:12 abcde ubuntu-appindicators@ubuntu.com[1631]: message repeated 3 times: [ Impossible to send about-to-show to menu: Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.Failed: error occurred in AboutToShow]
Jun 24 17:39:10 abcde gnome-shell[1631]: libinput error: event8  - Logitech USB Receiver: client bug: event processing lagging behind by 29ms, your system is too slow
Jun 24 17:40:39 abcde mpp[22038]: mpp: error found on mpp initialization
Jun 24 17:40:39 abcde mpp[22038]: mpp: error found on mpp initialization
Jun 24 17:40:39 abcde mpp[22038]: mpp: error found on mpp initialization
Jun 24 17:40:39 abcde mpp[22038]: mpp: error found on mpp initialization
Jun 24 17:40:39 abcde mpp[22038]: mpp: error found on mpp initialization
Jun 24 17:40:42 abcde com.obsproject.Studio.desktop[19593]: info: [pipewire] Stream 0xaaaaf67703d0 state: "paused" (error: none)
Jun 24 17:40:42 abcde com.obsproject.Studio.desktop[19593]: info: [pipewire] Stream 0xaaaaf67703d0 state: "unconnected" (error: none)
Jun 24 17:40:42 abcde ubuntu-appindicators@ubuntu.com[1631]: While reading menu layout on proxy :1.607: Gio.DBusError: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: No such object path '/MenuBar'
Jun 24 17:41:03 abcde kernel: [16940.639251] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.653823] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.670484] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.687053] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.703923] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.720635] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.737191] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.753894] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.770521] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.787269] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.804288] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.821182] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.837254] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.853867] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.870579] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.887302] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.903839] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.920611] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.937099] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.954261] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.971194] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16940.987163] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16941.003823] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16941.020562] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
Jun 24 17:41:03 abcde kernel: [16941.037227] [drm:rockchip_vop2_mod_supported] *ERROR* Unsupported format modifier 0x800000100000351
[... cut ...]
Jun 24 18:23:57 abcde chromium-browser.desktop[3932]: [3932:3937:0624/182357.774670:ERROR:ssl_client_socket_impl.cc(985)] handshake failed; returned -1, SSL error code 1, net_error -101
Jun 24 18:23:57 abcde chromium-browser.desktop[3930]: error: Too many contexts created!
Jun 24 18:24:05 abcde chromium-browser.desktop[3930]: message repeated 29 times: [ error: Too many contexts created!]
Jun 24 18:24:07 abcde chromium-browser.desktop[3930]: error: Too many contexts created!
Jun 24 18:24:28 abcde chromium-browser.desktop[3930]: message repeated 29 times: [ error: Too many contexts created!]

Joshua-Riek, 2023-06-24:
Type journalctl > sys.log and upload the file please.

ggdk123, 2023-06-24:
it froze again just now while browsing a simple web page in chromium.

Here is the log

[sys.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11856984/sys.log)

Joshua-Riek, 2023-06-24:
Do you have any media plugged into he device? Like a hard drive, usb, or other storage?

ggdk123, 2023-06-24:
I have one nvme ssd. There was 1 x USB thumb drive plugged in for a while yesterday, but only for a while.

Joshua-Riek, 2023-06-24:
I also see a lot of gnome-shell errors. Have you made any modifications? Like themes or extensions?

ggdk123, 2023-06-24:
zero modification, zero theme or extension

Joshua-Riek, 2023-06-24:
You might need a system re-install, to me gnome-shell looks broken for some odd reason.

But there is a new kernel that I have not tested much, I was planning on doing it this weekend. It has some stability enhancements.

ggdk123, 2023-06-24:
i actually did reinstall the system after experiencing the first one or two freezes. The system log I uploaded was the entire log after the "reinstallation".

Hopefully the new kernel could help improve, thanks

Joshua-Riek, 2023-06-24:
Also, make sure you are using a 5V 4A power supply. I would try to find another if your current is 5V 4A.

ggdk123, 2023-06-24:
I'm using the power supply shipped with the board from the official store

Joshua-Riek, 2023-07-01:
I will be closing this for now, I have not been able to reproduce it, and there have been no reports of crashing by other users.

closed 2023-07-01 by Joshua-Riek