Joshua-Riek/ubuntu-rockchip #222: Enabling user mode access to GPIO not working on Orange Pi 5b

Chambana, 2023-06-23:
I've tried a few methods to enable user level access to GPIO and they're not working with the latest image of 22.04 from this repo running on an Orange Pi 5b.  

I'm only able to run wiringPi python scripts as sudo.  If I run as user, this is the error:

```
wiringPiSetup: Unable to open /dev/mem or /dev/gpiomem: No such file or directory.
  Aborting your program because if it can not access the GPIO
  hardware then it most certianly won't work
  Try running with sudo?
```

I followed [this guide](https://opi-gpio.readthedocs.io/en/latest/install.html#non-root-access) from the OPi.GPIO docs:  

```
Non Root Access
If you want to be able to use the library as a non root user, you will need to setup a UDEV rule to grant you permissions first. This can be accomplished as follows:

$ sudo usermod -aG gpio <current_user>

$ sudo nano /etc/udev/rules.d/99-gpio.rules

That should add your user to the GPIO group, create a new UDEV rule, and open it in the Nano text editor.

Enter the following into Nano

SUBSYSTEM==”gpio”, KERNEL==”gpiochip*”, ACTION==”add”, PROGRAM=”/bin/sh -c ‘chown root:gpio /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport’” SUBSYSTEM==”gpio”, KERNEL==”gpio*”, ACTION==”add”, PROGRAM=”/bin/sh -c ‘chown root:gpio /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value’”
press ctrl-x, Y, and ENTER to save and close the file.

Finally, reboot and you should be ready to use OPi.GPIO as a non root user.
```

I've tried a few other suggestions, but nothing has worked.  Please post a method for enabling user-level access to GPIO on the OrangePi using this repo.  Happy to wipe my install and do a clean test if there are documented steps to do so.

Joshua-Riek, 2023-06-23:
Your steps look correct at a glance. Have you rebooted your system or reloaded the udev rules? Try a simple reboot and the udev rule to grant GPIO access should be loaded.

Joshua-Riek, 2023-06-23:
This is something I can not fix, sadly. WiringOP must access /dev/mem, which requires root because it can directly read the system memory. From what I see, Orange Pi needs to update the wiringOP to read from /dev/gpiochip*.

https://github.com/orangepi-xunlong/wiringOP/issues/28

Chambana, 2023-06-25:
Understood, was just going to provide an update that this is reproducible -- see below.  I added the udev rule and rebooted (no dice).  I added my user to the gpio group and even made that group owner of /dev/mem.  I was surpised that even adding the **gpio** group as the group owner of the /dev/mem file didn't do the trick. 

```
ubuntu@orangepi:~/scratch$ more /etc/udev/rules.d/99-gpio.rules 
SUBSYSTEM==”gpio”, KERNEL==”gpiochip*”, ACTION==”add”, PROGRAM=”/bin/sh -c ‘chown root:gpio /sys/class/gpio/export /sys/clas
s/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport’” SUBSYSTEM==”gpio”, KERNEL==”gpio*”, ACTION==”a
dd”, PROGRAM=”/bin/sh -c ‘chown root:gpio /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/act
ive_low /sys%p/direction /sys%p/edge /sys%p/value’”
ubuntu@orangepi:~/scratch$ groups ubuntu
ubuntu : ubuntu adm dialout cdrom floppy sudo audio dip video plugdev netdev lxd gpio
ubuntu@orangepi:~/scratch$ ls -la /dev/mem
crw-rw---- 1 root gpio 1, 1 Jun 25 22:21 /dev/mem
ubuntu@orangepi:~/scratch$ python3
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSetupPhys()
wiringPiSetup: Unable to open /dev/mem or /dev/gpiomem: No such file or directory.
  Aborting your program because if it can not access the GPIO
  hardware then it most certianly won't work
  Try running with sudo?
ubuntu@orangepi:~/scratch$ sudo python3
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wiringpi
>>> wiringpi.wiringPiSetupPhys()
0
>>> 
```

LoyalServant, 2024-04-26:
After chasing around a little with this myself, I came to the conclusion that another solution to this is to use libgpiod and move on.
I am using C, so I don't know about Python, etc.
The GPIOs I have gotten to work are split between gpiochip1 and gpiochip3
You just have to add a udev rule to sort out access to /dev/gpiochipX
pin 40 on the connector is gpiochip3, pin 3
38 is gpiochip3, pin 4
36 is gpiochip3, pin 5
32 is gpiochip1, pin 3
I only needed a few - these are the ones I worked out but it seems like the rest are following a similar convention.
according to wiringpi, 'gpio readall' the name of pin 40 is GPIO3_A3 and that is gpiochip3 pin 3.
Hope this helps someone trying to work around this.
Cheers.

closed 2025-02-17 by github-actions[bot]