Joshua-Riek/ubuntu-rockchip #223: [Feature request] NanoPC T6 - boot from SSD

vladkozlov69, 2023-06-24:
Hi Joshua,

First of all, thanks for your great work. Your ubuntu image works on T6 like a charm.

As Orange Pi 5 RK3588 board can boot from SSD with proper firmware in SPI, and NanoPC T6 shares the same SoC and has the SPI, I believe we can make NanoPC T6 booting from SSD without placing /boot partition on SD-card or EMMC? If so, have you tried it, or any chance to have it done?

Thanks in advance.

Joshua-Riek, 2023-06-24:
I don't think the NanoPC T6 has an SPI chip, but the eMMC can be used instead. Someone added support to boot from an NVMe for the NanoPi R6; the boards are very similar. I will need to look at the schematics and see if there are any changes specific to the NanoPC T6, then update the u-boot device tree accordingly.

https://github.com/MoeGuoH/nanopi_r6c-uboot/commit/9e65415395757f8347086778f862f32ebcdd793a

Joshua-Riek, 2023-06-24:
Booting from the NVMe on the NanoPi R6 is now working. It should only require a few changes to get working on the NanoPC T6.

Joshua-Riek, 2023-06-25:
Oh, interesting. The NanoPC T6 has a new option with the SPI NOR flash.
https://www.friendlyelec.com/index.php?route=product/product&product_id=292

Joshua-Riek, 2023-06-25:
Booting from the NVMe should now work on the NanoPC T6. I will do some additional tests before the next release to make sure there are no issues with this.

vladkozlov69, 2023-06-26:
> The NanoPC T6 has a new option with the SPI NOR flash

Yes, and owners of lower options still can achieve it as there is a SOIC-8 placeholder easy to solder.

vladkozlov69, 2023-06-26:
> If you have an SPI NOR chip on your board

Unfortunately I don't have it. I would like to get a chip photo from the person who has it, or contact a vendor to get the chip requirements. 
I have a lot of ESP32 boards collecting dust and some of them have 8MB or 32MB SPI chips... I bet one of them can be used there but I need a chip specs.

vladkozlov69, 2023-06-26:
> I might send an email out later today to FriendlyELEC asking where to find this information.

I found it on circuit diagram, seems to be a 32MB, but 1.8v VCC

[image]

Joshua-Riek, 2023-06-26:
I see, that is interesting.

vladkozlov69, 2023-06-26:
Actually, having /boot on eMMC (or even on SD) isn't a big burden as if root is mounted to NVMe there isn't any writing load on /boot, but as a purist I wonder if we can avoid any additional storages )))

Joshua-Riek, 2023-06-28:
I'll be closing this as completed for now. I think everything should work. I only need the SPI chip to confirm. If I find one, I will post it here to let you know where to find one.

closed 2023-06-28 by Joshua-Riek

vladkozlov69, 2023-06-29:
Great, thanks. Could you please add a bit more details on Wiki regarding writing u-boot to the emmc (right now it contains info only about writing u-boot to SPI, but it would be good to know whether it requires specific partitioning etc.)? Thanks in advance.

Joshua-Riek, 2023-06-29:
I will update the wiki later, but as a quick example you can write the bootloader to the emmc like below. However if you format the emmc it will remove the bootloader. 
```
sudo dd if=/lib/u-boot-nanopi-rk3588/idbloader.img of=/dev/mmcblk2 seek=64 conv=notrunc
sudo dd if=/lib/u-boot-nanopi-rk3588/u-boot.itb of=/dev/mmcblk2 seek=16384 conv=notrunc
```

Joshua-Riek, 2023-06-30:
@vladkozlov69 I think that I found a compatible SPI chip, W25Q32JWSSIQ.

https://www.mouser.com/ProductDetail/454-W25Q32JWSSIQ

https://www.aliexpress.us/item/3256805393487921.html?spm=a2g0o.productlist.main.3.43a72d80TDFtdg&algo_pvid=bfec0dcf-8946-4712-b3f5-ac816447902e&aem_p4p_detail=202306301633453449697654617500005688740&algo_exp_id=bfec0dcf-8946-4712-b3f5-ac816447902e-1&pdp_npi=3%40dis%21USD%215.8%215.8%21%21%21%21%21%40211bf55216881680258482444d080d%2112000033628127356%21sea%21US%210&curPageLogUid=CiVulA6AQPPU&search_p4p_id=202306301633453449697654617500005688740_2

cjkzwe, 2023-07-08:
@Joshua-Riek Could you provide a spi boot loader for me to test? I had soldered the flash in the schematic called W25Q256JWEIQ, But when I boot into my system, I could not find any /dev/mtdblk* device. What may be the reason?

Joshua-Riek, 2023-07-08:
The reason `/dev/mtdblk*` does not show is that the device tree does not have it defined. I don't have a NanoPC T6 with the SPI soldered. Currently, you will need to use the RKDevTool https://github.com/Joshua-Riek/ubuntu-rockchip/issues/236#issuecomment-1614830621.

If you want, I can send you a few device trees that you can try. And `/dev/mtdblk*` should work, I will need feedback on this as again I do not have one with the SPI soldered.

cjkzwe, 2023-07-08:
@Joshua-Riek OK, thanks a lot. I will send you my testing result as soon as possible. But I am not sure if this is a hardware issue or a software issue because I used a temperature higher than recommended to solder the SPI Flash.

Joshua-Riek, 2023-07-08:
Right now, it's for sure a software issue. Do you know how to swap out device trees on the system?

I have two device trees attached in the below zip archives. I think one of them should work. On your running system please try the below commands and after the reboot let me know if `/dev/mtdblk*` exists.

```
wget https://github.com/Joshua-Riek/ubuntu-rockchip/files/11991343/rk3588-nanopc-t6-fspim2_pins.zip
unzip rk3588-nanopc-t6-fspim2_pins.zip 
sudo cp rk3588-nanopc-t6.dtb /boot/firmware/rk3588-nanopc-t6.dtb
sync
sudo reboot now
```
If the above does not work to show `/dev/mtdblk*` then try this device tree. 
```
wget https://github.com/Joshua-Riek/ubuntu-rockchip/files/11991345/rk3588-nanopc-t6-fspim1_pins.zip
unzip rk3588-nanopc-t6-fspim1_pins.zip 
sudo cp rk3588-nanopc-t6.dtb /boot/firmware/rk3588-nanopc-t6.dtb
sync
sudo reboot now
```

Again I'm not sure which will work, or if either will work. Please let me know of the results.

[rk3588-nanopc-t6-fspim1_pins.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11991345/rk3588-nanopc-t6-fspim1_pins.zip)
[rk3588-nanopc-t6-fspim2_pins.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11991343/rk3588-nanopc-t6-fspim2_pins.zip)

cjkzwe, 2023-07-08:
@Joshua-Riek I am using the rk3588-sd-ubuntu-jammy-desktop-arm64-YYYYMMDD.img image provided by FriendlyELEC, where there seems to be no /boot/firmware/ directory at all, so I need to install the system under this project first. After that, I will test it. Thank you!

Joshua-Riek, 2023-07-08:
Right, you will need to use the image from this project. If one of the device trees works from above, I can send a kernel PR to friendlyelec so they have the patch too.

cjkzwe, 2023-07-08:
@Joshua-Riek Unfortunately, these two device tree files still cannot show the /dev/mtdblk* device, and I have successfully written rkspi_loader through RKDevTool, proving that this is probably not a hardware problem.

Joshua-Riek, 2023-07-08:
I see. I will need to do more research and get back to this. It isn't easy without having the hardware on hand.

cjkzwe, 2023-07-08:
Okay, thank you for your efforts!

cjkzwe, 2023-07-08:
@Joshua-Riek If you want to test the new device tree file, you can leave it to me and I will give you feedback.

Joshua-Riek, 2023-07-08:
Thank you, I will keep that in mind!

cjkzwe, 2023-07-09:
@Joshua-Riek 
[RK SpiNor and  SLC Nand SupportList Ver1.11_20181228.pdf](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11994686/RK.SpiNor.and.SLC.Nand.SupportList.Ver1.11_20181228.pdf)
This file may be helpful.

Joshua-Riek, 2023-07-24:
Hello @cjkzwe, can you test this new device tree like before? I think it should be working, but I need someone to confirm if `/dev/mtdblk*` exists with this new device tree.

```
wget https://github.com/Joshua-Riek/ubuntu-rockchip/files/12152840/rk3588-nanopc-t6.zip
unzip rk3588-nanopc-t6.zip 
sudo cp rk3588-nanopc-t6.dtb /boot/firmware/rk3588-nanopc-t6.dtb
sync
sudo reboot now
```
[rk3588-nanopc-t6.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12152840/rk3588-nanopc-t6.zip)

Here is the code change. I map out fspim1 according to the schematics.
```diff
diff --git a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
index 47a1030a4c73..5a62a440c243 100644
--- a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
@@ -478,6 +478,37 @@ &pcie3x4 {
        status = "okay";
 };
 
+&sfc {
+       status = "okay";
+       max-freq = <50000000>;
+       #address-cells = <1>;
+       #size-cells = <0>;
+       pinctrl-names = "default";
+       pinctrl-0 = <&fspim1_pins>;
+
+       spi_flash: spi-flash@0 {
+               #address-cells = <1>;
+               #size-cells = <0>;
+               compatible = "jedec,spi-nor";
+               reg = <0x0>;
+               spi-max-frequency = <50000000>;
+               spi-tx-bus-width = <1>;
+               spi-rx-bus-width = <4>;
+               status = "okay";
+
+               partitions {
+                       compatible = "fixed-partitions";
+                       #address-cells = <1>;
+                       #size-cells = <1>;
+
+                       loader@0 {
+                               label = "loader";
+                               reg = <0x0 0x1000000>;
+                       };
+               };
+       };
+};
+
 &pinctrl {
        gpio-leds {
                sys_led_pin: sys-led-pin {
@@ -537,6 +568,25 @@ pcie_m2_1_pwren: pcie-m21-pwren {
                };
        };
 
+       fspi {
+               /omit-if-no-ref/
+               fspim1_pins: fspim1-pins {
+                       rockchip,pins =
+                               /* fspi_clk_m1 */
+                               <2 RK_PB3 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_cs0n_m1 */
+                               <2 RK_PB4 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d0_m1 */
+                               <2 RK_PA6 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d1_m1 */
+                               <2 RK_PA7 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d2_m1 */
+                               <2 RK_PB0 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d3_m1 */
+                               <2 RK_PB1 2 &pcfg_pull_up_drv_level_2>;
+               };
+    };
+
        sdmmc {
                sd_s0_pwr: sd-s0-pwr {
                        rockchip,pins = <4 RK_PA5 RK_FUNC_GPIO &pcfg_pull_down>;
```

vladkozlov69, 2023-08-04:
> ```
> sudo dd if=/lib/u-boot-nanopi-rk3588/idbloader.img of=/dev/mmcblk2 seek=64 conv=notrunc
> sudo dd if=/lib/u-boot-nanopi-rk3588/u-boot.itb of=/dev/mmcblk2 seek=16384 conv=notrunc
> ```

@Joshua-Riek looks like these files disappeared in the last release?

Joshua-Riek, 2023-08-04:
I renamed nanopi to friendlyelec. I thought I had a symlink, so the old path will still work, but I guess I forgot to add it.

cjkzwe, 2023-08-05:
@Joshua-RiekI am sorry for not being able to respond to you in time, but I will still be busy in the next few days. I will reply to you within five days.

young-bob, 2023-08-05:
> Right now, it's for sure a software issue. Do you know how to swap out device trees on the system?
>
> I have two device trees attached in the below zip archives. I think one of them should work. On [...]

rk3588-nanopc-t6-fspim1_pins   working
[image]

rk3588-nanopc-t6-fspim2_pins  not working
[image]

young-bob, 2023-08-05:
@Joshua-Riek Thanks for your fantastic work!

seua, 2023-08-05:
I tried the 1.21 img with the tk3588-nanopc-t6.dtb in firmware folder , and  /dev/mtdblk* not shown. 
I tries the 3 attachment this thread, all not work.

Joshua-Riek, 2023-08-05:
> I tried the 1.21 img with the tk3588-nanopc-t6.dtb in firmware folder , and /dev/mtdblk* not shown. I tries the 3 attachment this thread, all not work.

I am aware, I don't have the hardware, so I can debug and fix this. I created a new issue for this #290.

peclik, 2023-08-07:
@Joshua-Riek I also confirm rk3588-nanopc-t6-fspim1_pins is working.
There is a wrong path in the instructions (missing dtbs dir), the correct command is
```
sudo cp rk3588-nanopc-t6.dtb /boot/firmware/dtbs/rk3588-nanopc-t6.dtb
```
I have flashed u-boot and booting form eMMC without SD card.

Joshua-Riek, 2023-08-07:
Oh, and with that dtb, you can see /dev/mtdblock0?

peclik, 2023-08-07:
Yes, I can.

Joshua-Riek, 2023-08-07:
Perfect! Thanks for catching the typo!

young-bob, 2023-08-07:
Yes, rk3588-nanopc-t6-fspim1_pins is working. I can see /dev/mtdblock0 and have successfully flashed u-boot to SPI NOR but its boots failed. 

Then, I flashed idbloader.img and u-boot.itb to eMMC. The system boots from NVME without a TF card.

sudo dd if=/lib/u-boot-friendlyelec-rk3588/idbloader.img of=/dev/mmcblk2 seek=64 conv=notrunc
sudo dd if=/lib/u-boot-friendlyelec-rk3588/u-boot.itb of=/dev/mmcblk2 seek=16384 conv=notrunc

Joshua-Riek, 2023-08-07:
> Yes, rk3588-nanopc-t6-fspim1_pins is working. I can see /dev/mtdblock0 and have successfully flashed u-boot to SPI NOR but its boots failed.
>
> Then, I flashed idbloader.img and u-boot.itb to eMMC. [...]

That is strange. I have a NanoPC T6 with the SPI NOR and eMMC coming in a few weeks, so I can get this working properly. There were some generous donations recently to help me with the cost. 

Can someone try the device tree mentioned here https://github.com/Joshua-Riek/ubuntu-rockchip/issues/223#issuecomment-1648690312? The rk3588-nanopc-t6-fspim1_pins device tree has a few pins not mapped correctly and may cause issues.

young-bob, 2023-08-07:
> > Yes, rk3588-nanopc-t6-fspim1_pins is working. I can see /dev/mtdblock0 and have successfully flashed u-boot to SPI NOR but its boots failed.
> > Then, I flashed idbloader.img and u-boot.itb to eMMC. The system boots from NVME without a TF card.
> > sudo dd if=/lib/u-boot-friendlyelec-rk3588/idbloader.img of=/dev/mmcblk2 seek=64 conv=notrunc sudo dd if=/lib/u-boot-friendlyelec-rk3588/u-boot.itb of=/dev/mmcblk2 seek=16384 conv=notrunc
> 
> That is strange. I have a NanoPC T6 with the SPI NOR and eMMC coming in a few weeks, so I can get this working properly. There were some generous donations recently to help me with the cost.
> 
> Can someone try the device tree mentioned here [#223 (comment)](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/223#issuecomment-1648690312)? The rk3588-nanopc-t6-fspim1_pins device tree has a few pins not mapped correctly and may cause issues.

Wow, I tested this, and it works. Now completely detached from the TF card and eMMC and successfully loaded the system from NVME, perfect!

young-bob, 2023-08-07:
> Wow, I tested this, and it works. Now completely detached from the TF card and eMMC and successfully loaded the system from NVME, perfect!

[image]

[image]

Joshua-Riek, 2023-08-07:
> Wow, I tested this, and it works. Now completely detached from the TF card and eMMC and successfully loaded the system from NVME, perfect!

Sweet! Could you ensure there is no bootloader in the eMMC and clear the first few MB with zeros? Something like the below command should work.
```
sudo dd if=/dev/zero of=/dev/mmcblk0 count=4096 bs=512
```

Joshua-Riek, 2023-08-07:
@young-bob, thank you very much for the testing! I'll send a pull request to the kernel later to get this added to the system!

young-bob, 2023-08-08:
@Joshua-Riek Thank you, as a T6 user, this is a best gift!

cjkzwe, 2023-08-14:
@Joshua-Riek I am sorry but unfortunatly, the device tree file in the latest release still does not work for my nanopc with my spi flash, which is recommended by their official schematic. I can not see any mtdblk* in /dev. I have tried the guide in #290 but still not working. This might still be a problem. @young-bob Which spi flash do you use?

Joshua-Riek, 2023-08-14:
@cjkzwe I have a T6 coming today with spi flash. I can finally do some testing later today.

Joshua-Riek, 2023-08-14:
I see the issue. Here is the diff between the working device tree and the current one. I need to check the freq and pins.
```diff
4835c4835
< 		max-freq = <0x2faf080>;
---
> 		max-freq = <0x5f5e100>;
4845c4845
< 			spi-max-frequency = <0x2faf080>;
---
> 			spi-max-frequency = <0x5f5e100>;
7193c7193
< 				rockchip,pins = <0x02 0x0b 0x02 0x175 0x02 0x0c 0x02 0x175 0x02 0x06 0x02 0x175 0x02 0x07 0x02 0x175 0x02 0x08 0x02 0x175 0x02 0x09 0x02 0x175>;
---
> 				rockchip,pins = <0x02 0x0b 0x03 0x175 0x02 0x0c 0x03 0x175 0x02 0x06 0x03 0x175 0x02 0x07 0x03 0x175 0x02 0x08 0x03 0x175 0x02 0x09 0x03 0x175>;
```

Joshua-Riek, 2023-08-14:
I sent a PR to the kernel repo for a fix, and I plan to issue a kernel update once the change is merged. 

Below are the corresponding device tree change.
```diff
diff --git a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
index c71e8e79c13d..ec9acd528797 100644
--- a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
@@ -520,25 +520,6 @@ usr_led_pin: usr-led-pin {
                };
        };
 
-       fspi {
-               /omit-if-no-ref/
-               fspim1_pins: fspim1-pins {
-                       rockchip,pins =
-                               /* fspi_clk_m1 */
-                               <2 RK_PB3 2 &pcfg_pull_up_drv_level_2>,
-                               /* fspi_cs0n_m1 */
-                               <2 RK_PB4 2 &pcfg_pull_up_drv_level_2>,
-                               /* fspi_d0_m1 */
-                               <2 RK_PA6 2 &pcfg_pull_up_drv_level_2>,
-                               /* fspi_d1_m1 */
-                               <2 RK_PA7 2 &pcfg_pull_up_drv_level_2>,
-                               /* fspi_d2_m1 */
-                               <2 RK_PB0 2 &pcfg_pull_up_drv_level_2>,
-                               /* fspi_d3_m1 */
-                               <2 RK_PB1 2 &pcfg_pull_up_drv_level_2>;
-               };
-       };
-
        headphone {
                hp_det: hp-det {
                        rockchip,pins = <1 RK_PC4 RK_FUNC_GPIO &pcfg_pull_none>;

```

young-bob, 2023-08-15:
> @Joshua-Riek I am sorry but unfortunatly, the device tree file in the latest release still does not work for my nanopc with my spi flash, which is recommended by their official schematic. I can not [...]

@cjkzwe
I use mtdblock0.
Chip: Winbond 25Q128JWSQ 2311

[image]

cjkzwe, 2023-08-15:
@young-bob I'm using W25Q256JWEIQ.