Joshua-Riek/ubuntu-rockchip #228: Armcord - Share Screen

cantalupo555, 2023-06-25:
The "Share Screen" function does not work.
I tested it in Armbian and it works perfectly.
Version .deb: https://github.com/ArmCord/ArmCord/releases/download/v3.2.0/ArmCord_3.2.0_arm64.deb

Log:
```
@orangepi5:~$ armcord 
[Config manager] doneSetup: undefined
[Config manager] performanceMode: none
ArmCord has been run before. Skipping setup.
No performance modes set
[Config manager] windowStyle: default
[Config manager] armcordCSP: true
[Config manager] doneSetup: undefined
[Config manager] customIcon: undefined
Setting up CSP unstricter...
[Config manager] trayIcon: default
[Config manager] windowStyle: default
[Config manager] clientName: undefined
[Config manager] spellcheck: true
[Config manager] windowStyle: default
[Config manager] startMinimized: false
[Config manager] mods: none
[Config manager] ignoreProtocolWarning: undefined
[Mod loader] Skipping
[Config manager] 0: undefined
[Config manager] mobileMode: false
[Config manager] useLegacyCapturer: false
Starting screenshare module...
[Config manager] trayIcon: default
[Config manager] alternativePaste: false
undefined
[Config manager] inviteWebsocket: true
[arRPC] arRPC v3.1.0 ArmCord
[arRPC > ipc] checking /run/user/1000/discord-ipc-0
[Config manager] skipSplash: undefined
[arRPC > ipc] checked if socket is available: true 
[arRPC > ipc] listening at /run/user/1000/discord-ipc-0
[arRPC > websocket] trying port 6463
[arRPC > websocket] listening on 6463
[arRPC > process] started
[Config manager] channel: stable
[Config manager] mods: none
[Config manager] automaticPatches: false
[Config manager] useLegacyCapturer: false
[Config manager] startMinimized: false
[Config manager] trayIcon: default
{ width: 2560, height: 1053, isMaximized: true, x: 0, y: 27 }
[Window state manager] [object Object]
{ width: 2560, height: 1053, isMaximized: true, x: 0, y: 27 }
[Window state manager] [object Object]
{ width: 2560, height: 1053, isMaximized: true, x: 0, y: 27 }
[Window state manager] [object Object]
{ width: 2560, height: 1053, isMaximized: true, x: 0, y: 27 }
[Window state manager] [object Object]
{ width: 2560, height: 1053, isMaximized: true, x: 0, y: 27 }
[Window state manager] [object Object]
[Config manager] startMinimized: false
[Config manager] channel: stable
[Config manager] mods: none
[Config manager] automaticPatches: false
[Config manager] useLegacyCapturer: false
[Config manager] mobileMode: false
[3887:0624/212749.482545:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[Config manager] trayIcon: default
[Config manager] trayIcon: default
[Config manager] trayIcon: default
[Config manager] disableAutogain: undefined
[Config manager] trayIcon: default
{
  frame: EventEmitter {},
  securityOrigin: 'https://discord.com/',
  userGesture: true,
  videoRequested: true,
  audioRequested: true
}
Segmentation fault (core dumped)

```

Joshua-Riek, 2023-06-25:
When you tested this on Armbian, where you using wayland or x11?

cantalupo555, 2023-06-26:
Problem encountered when switching to X11, chromium has constant crashes, very slow.
Any open video becomes stop motion.

closed 2023-06-26 by cantalupo555

cantalupo555, 2023-06-26:
https://github.com/SpacingBat3/WebCord

WebCord currently supports Wayland.
Problem solved.
@Joshua-Riek