Joshua-Riek/ubuntu-rockchip #229: Enabling i2c on 22.04?

petewarden, 2023-06-26:
First, thanks for your great work on these images, they've been a big help.

I'm porting an application that uses an I2C sensor, and I haven't been able to access it on the Ubuntu 22.04 Server image for Rock 5B, and I wondered if you had any suggestions for debugging the issue?

I've previously been using a [Radxa Rock 5B Ubuntu Focal Server image](https://github.com/radxa/debos-radxa/releases/download/20221031-1045/rock-5b-ubuntu-focal-server-arm64-20221031-1328-gpt.img.xz) but I've switched to [one of your images](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.18/ubuntu-22.04.2-preinstalled-server-arm64-rock5b.img.xz.sha256) because it seems to run our compute-intensive code significantly faster.

I've been able to get everything working except I2C. I have checked that the sensor is wired in correctly, but when I run `i2cdetect -y 0` I see the following output:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- UU UU -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

I'm not entirely certain how to interpret it, but here's what I expect to see, and do see when using the Radxa image:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- 39 -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 4f 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- 62 -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```
(0x62 is the address of the sensor)

I have made sure that I've set up the user permissions for the ubuntu user, and `i2cdetect` and the accessibility of `/dev/i2c-0` seems to indicate that isn't the problem.

One thing I'm suspicious about is that in the Radxa image I had to take the following steps to enable I2C:

 - Add `dtoverlay=rk3588-i2c0-m1` to /boot/config.txt.
 - Run `sudo update_extlinux.sh`.
 - `sudo reboot`.

Before I took those steps, I saw similar results on the Radxa OS to those I see now with your image (the `UU`s), so I suspect I need to do something to enable the I2C support at the kernel level. I haven't been able to find anything similar to the `/boot/config.txt` or `update_extlinux.sh` files on your image though, and I don't understand kernel magic well enough to figure out how to get similar results, since I was just cargo-culting those steps from a Stack Overflow suggestion.

Can you point me towards any terms or docs I should be digging into to make progress on this? Any help is much appreciated as I'm quite baffled right now.

Joshua-Riek, 2023-06-26:
You should need to do something similar on this Ubuntu image to enable I2C. I will quickly check the raxda kernel source and my own. I think I know what needs to be done, but I need to confirm.

Joshua-Riek, 2023-06-26:
I have the same overlay file available, please do the following to enable I2C.
1. Open the `ubuntuEnv.txt` file: Usually in the `/boot/firmware/` directory. You can use a text editor like nano to open the file with root permission.
2. You will see the bottom line says `overlays=`
3. Add the device tree overlay to the previously mentioned line as follows: `overlays=rk3588-i2c0-m1`
4. Save the file and reboot your system, I2C should be enabled.

petewarden, 2023-06-26:
Thanks so much @Joshua-Riek, that worked like a charm! Closing the issue.

closed 2023-06-26 by petewarden

kphanipavan, 2024-03-03:
Adding overlays seems to block CPUFreq from loading, which deals with cpu frequency scaling. If CPUFreq doesn't load, the cpu gets stuck at its lowest power state(by default), as you observed. I had the same issue and reported the same [in the Radxa's discord chat](https://discord.com/channels/855634073376260096/1063181753122037841/1140141032231673937). This seems to be unfixed yet.

Jibun-no-Kage, 2024-09-18:
Is not the i2c pins 3,5 on i2c-8 bus?  The IndieDroid NOVA GPIO information states that i2c SDA pin 3 and i2c SCL pin 5 are not on the 0 bus, but the 8 bus.  So are you not loading the wrong overlay?

Reference...
https://wiki.indiedroid.us/Nova/gpio

[image]