Joshua-Riek/ubuntu-rockchip #231: [User Request] Config Partition

raphaelscholle, 2023-06-27:
Hello, I started to use your images a few weeks ago and am quite impressed how well they work.

But there is one thing I really miss and this is the configuration Partition which Radxa included in their new release and Raspberry's use since the beginning.
This makes it much easier to add/edit overlays and configure the images.

Maybe it's possible to add this functionality into your images

Joshua-Riek, 2023-06-27:
I have been using the Raspberry Pi as an inspiration for this project, and I would love to implement a similar system to their config.txt file. I have yet to look into this as I have been focused on fixing bugs. I'll be sure to look at how Radxa has implemented this. I won't make any promises because I need to do more research.

Right now, overlays can be specified in the ubuntuEnv.txt file under the path /boot/firmware, and this is documented [here](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays). It could be more elegant, but it works for now.

raphaelscholle, 2023-06-27:
Thanks for the link, but I knew that before ;)

The cool thing about the config partition is that you would be able to edit the file without a linux device or using the sbc directly.

So while flashing you can just edit that and plug in the sd-card/emmc/ssd and it'll just work

Joshua-Riek, 2023-06-29:
I think this might exist already now i think about it, after flashing your media you will be able to see a partition called system-boot. Which can be accessed on any device because the partition is FAT16. The server image will come with could-init configuration files in this partition which can be used to setup complex configurations.

Joshua-Riek, 2023-06-29:
Wow, that is strange. I can look into using FAT32 for the boot partition.

raphaelscholle, 2023-07-20:
I actually invested a little time into that and tested a few things with the build script. I don't think the main issue is the FAT16 format, it's just something strange, each of my windows devices  doesn't auto mount the partition even if it is fat32. 

Also I'm that interested in this partition, because in OpenHD we do use the partition to configure some stuff while flashing our custom image, so that the user doesn't need to interact at all, and settings can be changed within the device that flashes the image (windows/linux/mac). Without the partition I can't write anything to the device itself which renders it quite useless.

Joshua-Riek, 2023-08-20:
Sorry about taking so long to get back to you on this. I prefer the boot partition to be a Linux boot partition for now.

closed 2023-08-20 by Joshua-Riek