Joshua-Riek/ubuntu-rockchip #232: [BUG] camera_engine_rkaiq (Rockship ISP) fails and doesn't start *tested on Rock5A

raphaelscholle, 2023-06-27:
When using your image the camera_engine_rkaiq doesn't work. Tested on Rock5A + imx415.
I enabled the overlay and the camera displays an untuned image (without using the ISP, which is started with camera_engine_rkaiq)

Here is a debug-log:

rkaiq log level ff0
media get entity by name: stream_cif_dvp_id0 is null
media get entity by name: stream_cif_dvp_id1 is null
media get entity by name: stream_cif_dvp_id2 is null
media get entity by name: stream_cif_dvp_id3 is null
media get entity by name: rkcif-mipi-luma is null
media get entity by name: rkcif-lvds-subdev is null
media get entity by name: rkcif-lite-lvds-subdev is null
media get entity by name: rockchip-mipi-dphy-rx is null
media get entity by name: stream_cif is null
media get entity by name: rkcif-dvp-sof is null
media get entity by name: rkisp-csi-subdev is null
media get entity by name: rkisp-mpfbc-subdev is null
media get entity by name: rkisp_rawwr0 is null
media get entity by name: rkisp_rawwr1 is null
media get entity by name: rkisp_rawwr2 is null
media get entity by name: rkisp_rawwr3 is null
media get entity by name: rkisp_dmapath is null
media get entity by name: rkisp-mipi-luma is null
media get entity by name: rockchip-mipi-dphy-rx is null
media get entity by name: rockchip-csi2-dphy0 is null
media get entity by name: rkcif_dvp is null
media get entity by name: rkcif_dvp is null
media get entity by name: rkcif_lite_mipi_lvds is null
media get entity by name: rkcif_mipi_lvds is null
media get entity by name: rkcif_mipi_lvds1 is null
media get entity by name: rkcif_mipi_lvds2 is null
media get entity by name: rkcif_mipi_lvds3 is null
media get entity by name: rkcif_mipi_lvds4 is null
media get entity by name: rkcif_mipi_lvds5 is null
media get entity by name: rkcif-mipi-lvds is null
media get entity by name: rkcif-mipi-lvds1 is null
media get entity by name: rkcif-mipi-lvds3 is null
media get entity by name: rkcif-mipi-lvds4 is null
media get entity by name: rkcif-mipi-lvds5 is null
media get entity by name: rkisp-isp-subdev is null
media get entity by name: rkisp-input-params is null
media get entity by name: rkisp-statistics is null
media get entity by name: rkisp_mainpath is null
Cound not find rkisp dev names, skipped /dev/media0
ERR: Bad media topology for: /dev/media0
DBG: get rkisp-isp-subdev devname: /dev/v4l-subdev3
DBG: get rkisp-input-params devname: /dev/video19
DBG: get rkisp-statistics devname: /dev/video18
DBG: get rkisp_mainpath devname: /dev/video11
ERR: Bad media topology for: /dev/media2
ERR: Bad media topology for: /dev/media3
ERR: Bad media topology for: /dev/media4
ERR: Bad media topology for: /dev/media5
ERR: Bad media topology for: /dev/media6
ERR: Bad media topology for: /dev/media7
ERR: Bad media topology for: /dev/media8
ERR: Bad media topology for: /dev/media9
ERR: Bad media topology for: /dev/media10
ERR: Bad media topology for: /dev/media11
ERR: Bad media topology for: /dev/media12
ERR: Bad media topology for: /dev/media13
ERR: Bad media topology for: /dev/media14
ERR: Bad media topology for: /dev/media15
Segmentation fault (core dumped)

Joshua-Riek, 2023-06-27:
I don't have the hardware but I can think of a few things that might be the problem. My first thought is the device tree overlay is outdated from radxa's. Which overlay are you using exactly? And what is the overlay used for this device on radxa's image?

raphaelscholle, 2023-06-27:
I can test and compare your and radxa's image tomorrow

if you have anything more to test I can do Rock5A and Rock5B tests

Joshua-Riek, 2023-06-27:
The Rock 5A and 5B images should be the same software-wise. I would also try to switch the display manager to x11 instead of wayland.

raphaelscholle, 2023-06-28:
I didn't use x at all and just used gstreamer,
the radxa images use : Linux rock-5a 5.10.110-6-rockchip

and there the isp is working.

I tested a lot with your kernel today and noticed that some wifi-drivers break compilation.
Also I noticed that a simple install via apt fails.

Joshua-Riek, 2023-06-28:
I can't do much without any logs, especially because I don't have the camera hardware. And what do you mean by a simple install via apt fails?

raphaelscholle, 2023-06-28:
logs will follow soon, I've just build the complete image with the build-script and testing if the overlay radxa uses is simple to integrate.

raphaelscholle, 2023-06-28:
Okey .. investigated a little bit more, the problem doesn't look to be the dtb file,

got the isp started when using this camera enginge

 https://github.com/radxa/debos-radxa/raw/main/rootfs/packages/arm64/camera/camera-engine-rkaiq_3.0.2_arm64.deb

tuning files are off .. a lot, but it's better then before

Joshua-Riek, 2023-06-29:
> I also noticed with a clean flash I can install a kernel via apt, must have been a bug which i introduced while tinkering around.

I uploaded a new kernel to my ppa, so this is not a bug, it's a feature :)

 My guess is I have an old version of rkaiq included.

raphaelscholle, 2023-06-29:
I've actually seen this ;) But I build my own kernel based on yours with a modified camera overlay, just doesn't change anything there

raphaelscholle, 2023-06-29:
[image]
[image]

I just made a few screenshots, both with the isp(rkaiq enabled). For these I also installed your newest kernel

Joshua-Riek, 2023-06-29:
Very interesting, are you using the server or desktop image? By default, the server image does not contain any GPU / VPU-related drivers, but they can be downloaded with apt.

raphaelscholle, 2023-06-29:
I noticed that, so I use the desktop image.

raphaelscholle, 2023-06-29:
I also tried to apply the same tuning files from the official image, which doesn't change anything. Next step I want to test is replacing the rkaiq_3A_server. Maybe that helps

Joshua-Riek, 2023-06-29:
I will need to track down where I got the rkaiq files from, it's been a while and they are totally outdated. For sure try updating  rkaiq_3A_server.

raphaelscholle, 2023-06-29:
there is something really weird, even replacing the binary with the one from the original image doesn't change a thing

Joshua-Riek, 2023-06-29:
Try switching the desktop to x11 or run your application with the prefix `malirun` like `malirun ./myapp`. The malirun will switch the GPU-related libs for the running application.

raphaelscholle, 2023-06-29:
No difference, I'm not even in X, I stream the vamera directly via gstreamer to a external device, which is decoding. So all display/decode stuff is out of the picture.

Since you asked yesterday, the official images use this overlay:
https://github.com/radxa/overlays/blob/main/arch/arm64/boot/dts/rockchip/overlays/rock-5a-radxa-camera-4k.dts

raphaelscholle, 2023-06-29:
This is my pipeline, which is the same with the original and your image : 

gst-launch1.0 v4l2src device=/dev/video11 io-mode=auto do-timestamp=true ! video/x-raw,format=NV12, framerate=30/1 ! mpph264enc rc-mode=cbr bps=2000000 width=1920 height=1080 gop=5 ! queue ! h264parse config-interval=-1 ! rtph264pay mtu=1024 !  appsink drop=true name=out_appsink]

Joshua-Riek, 2023-06-29:
Interesting, could you provide me a link to the radxa image you are using? I will take a look at the system files and see if I can find anything.

raphaelscholle, 2023-06-29:
sure here it is 👍 
https://github.com/radxa-build/rock-5a/releases/download/b16/rock-5a_ubuntu_jammy_cli_b16.img.xz

Joshua-Riek, 2023-06-29:
Thanks, ill take a look

raphaelscholle, 2023-06-29:
... I think I got it repaired now .. the issue was on my side I actually used a far to high bitrate ... bps is refering to kbps in mpp .. good to know

Joshua-Riek, 2023-06-29:
Haha don't worry about it, I do the same thing all the time. I'm glad to hear it's working!

peytonicmaster6, 2023-07-09:
Hate to reopen this issue, but I'm having the same problem on an Orange Pi 5B. The camera feed on your ubuntu image is green and doesn't look like the ISP is running, like shown earlier by @raphaelscholle. On the Orange Pi Arch Linux image, the same camera looks normal and definitely has the ISP running. 

I'm also using the exact same gstreamer command as the Arch Linux image, which doesn't specify a bitrate as in the previous case. It's also the same case if I use OpenCV in python, so I think it may be a case with the Rockchip ISP rather than the command.

Joshua-Riek, 2023-07-09:
@peytonicmaster6 make a new issue for the 5b. I will take a look.

raphaelscholle, 2023-07-19:
After a swift reboot the issue returned. I also can't reconstruct the "fix". Also it looks like mpp doesn't accept cbr now. Pretty strange. I installed all the packages you install to your ubuntu-version to the official one and the isp/rkaiq works normal. (but also not accepting cbr)

It's a really strange behaviour. I also did not get the rkaiq version from radxa or your repo working on the ubuntu image at all.

Joshua-Riek, 2023-09-11:
Hello @raphaelscholle, sorry for only getting back to this now, but I have some progress. I updated the camera software packages with commit d206c00d1af117b65d82f7cbf24077423c992c5a. Try to install them on your system, and it should work, I tested this out on the Orange Pi 5 Plus, and it works well, also there is no green tint to the captured image.

https://github.com/Joshua-Riek/ubuntu-rockchip/blob/d206c00d1af117b65d82f7cbf24077423c992c5a/scripts/build-rootfs.sh#L258-L260

Joshua-Riek, 2024-06-24:
Closing see https://github.com/Joshua-Riek/ubuntu-rockchip/pull/690

closed 2024-06-24 by Joshua-Riek