Joshua-Riek/ubuntu-rockchip #234: [OrangePi 5] System hangs after connect OTG hub

0312birdzhang, 2023-06-29:
Dmesg:
```
...
[    9.469448] rk_gmac-dwmac fe1c0000.ethernet eth0: registered PTP clock
[    9.469586] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[   12.513123] rk_gmac-dwmac fe1c0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   12.513175] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   15.143438] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.143442] Bluetooth: BNEP filters: protocol multicast
[   15.143451] Bluetooth: BNEP socket layer initialized
[   15.577217] mali fb000000.gpu: Loading Mali firmware 0x1010000
[   15.580440] mali fb000000.gpu: Mali firmware git_sha: a9b9d413d30169284dfbc5c68e8c65fdaeca28b7 
[   34.707664] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
[  660.694888] perf: interrupt took too long (2507 > 2500), lowering kernel.perf_event_max_sample_rate to 79500

[ 1046.400568] usb 7-1: new high-speed USB device number 2 using xhci-hcd
[ 1046.541696] usb 7-1: New USB device found, idVendor=1a40, idProduct=0801, bcdDevice= 1.00
[ 1046.541715] usb 7-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 1046.541721] usb 7-1: Product: USB 2.0 Hub
[ 1046.589564] hub 7-1:1.0: USB hub found
[ 1046.589721] hub 7-1:1.0: 4 ports detected

Socket error Event: 32 Error: 10053.
Connection closing...Socket close.

Connection closed by foreign host.

Disconnected from remote host(orangepi) at 11:43:48.

```
Full log:
[dmsg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11901801/dmsg.txt)

Joshua-Riek, 2023-06-29:
Not enough information here. Please explain in a reproducible manner. Also, your logs don't look complete, I just see a usb hub being plugged in.

0312birdzhang, 2023-06-29:
How to reproduce:  just connect a usb otg hub to usb-c port, then the system will stuck( remote ssh and GUI), no more logs.

[image]

Joshua-Riek, 2023-06-29:
I tested this on my end and my system does not hang. Are you using a 5v4a power adapter?

0312birdzhang, 2023-06-30:
Okay, i can confirm it's the third party panel problem. 

I paste the patch to [gist](https://gist.github.com/0312birdzhang/4cfd2653c095565aa7f26383b5c556c5) if you have some time to look at, or you can ignore this.

Joshua-Riek, 2023-06-30:
I see. Have you tested this patch? I can send a kernel pr if you can confirm it works.

0312birdzhang, 2023-06-30:
Oh, you misunderstood me. This patch is from the screen panel vendor, they modified the kernel to make the panel work, and this is the reason why system hangs.

I asked them, they said it's working well on the stock Orange Pi ubuntu system, so they don't fix this bug.

Joshua-Riek, 2023-07-01:
I see. I saw a patch for the innolux afj101 ba213 panel in the current kernel. Does that panel work for a display?

0312birdzhang, 2023-07-02:
not fully working, only touch screen and black light, nothing show

Joshua-Riek, 2023-07-04:
I don't think there is much I can do here. This looks like an issue with the panel kernel driver.

closed 2023-07-22 by Joshua-Riek