Joshua-Riek/ubuntu-rockchip #236: NanoPC-T6 can not boot from nvme without sd card

lee86, 2023-06-30:
my t6 can boot from nvme with sd card,but sometime mount disks like this 
/dev/mmcblk0p2  /
/dev/nvme0n1p1 /boot/firmware
or 
/dev/nvme0n1p2 /
/dev/mmcblk0p1  /boot/firmware
i know it caused because the uuid is same
so the boot in sd is necessary?

Joshua-Riek, 2023-06-30:
The sdcard is not required. There are two ways to boot from the NVMe. 

You can write the bootloader to the eMMC. However, if you format the eMMC it will remove the bootloader.
```
sudo dd if=/lib/u-boot-nanopi-rk3588/idbloader.img of=/dev/mmcblk2 seek=64 conv=notrunc
sudo dd if=/lib/u-boot-nanopi-rk3588/u-boot.itb of=/dev/mmcblk2 seek=16384 conv=notrunc
```

Or you can write the bootloader to the SPI NOR. Some devices don't have this.
```
sudo dd if=/lib/u-boot-nanopi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
sync
```

lee86, 2023-06-30:
> 
Note: In China, all new factory 16G RAM versions of the T6 have SPI NOR, but you may not find it in /dev.
[image]

So the commad `sudo dd if=/lib/u-boot-nanopi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc` means copy /lib/u-boot-nanopi-rk3588/rkspi_loader.img to /dev/mtdblock0,This is the reason why I failed.

Others may also encounter this situation, the solution is to use `RKDevTool` to write `rkspi_loader.img` onto SPI NOR.
[image]

Now I can boot from Nvme.

Forgive my broken English and thank you for your help!

closed 2023-06-30 by lee86

Joshua-Riek, 2023-06-30:
I'm happy it's working! Your T6 has the SPI NOR, and you don't see it in /dev? It must be a device tree bug. I will investigate.

Joshua-Riek, 2023-06-30:
I might have a patch to fix the SPI NOR. Can you please download and unzip the attached device tree file and place rk3588-nanopc-t6.dtb to /boot/firmware/rk3588-nanopc-t6.dtb, then reboot? [rk3588-nanopc-t6.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/11919850/rk3588-nanopc-t6.zip)  You should be able to see /dev/mtdblock0 after the reboot.

```diff
diff --git a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
index 47a1030a4c73..8240a142fb5f 100644
--- a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
@@ -422,6 +422,37 @@ &i2s7_8ch {
        status = "okay";
 };
 
+&sfc {
+       status = "okay";
+       max-freq = <100000000>;
+       #address-cells = <1>;
+       #size-cells = <0>;
+       pinctrl-names = "default";
+       pinctrl-0 = <&fspim2_pins>;
+
+       spi_flash: spi-flash@0 {
+               #address-cells = <1>;
+               #size-cells = <0>;
+               compatible = "jedec,spi-nor";
+               reg = <0x0>;
+               spi-max-frequency = <100000000>;
+               spi-tx-bus-width = <1>;
+               spi-rx-bus-width = <4>;
+               status = "okay";
+
+               partitions {
+                       compatible = "fixed-partitions";
+                       #address-cells = <1>;
+                       #size-cells = <1>;
+
+                       loader@0 {
+                               label = "loader";
+                               reg = <0x0 0x1000000>;
+                       };
+               };
+       };
+};
+
 &mdio1 {
        status = "disabled";
 };
```

lee86, 2023-07-02:
My situation may be unique. Replacing /boot/firmware/dtbs/rk3588-nanopc-t6.dtb and placing rk3588-nanopc-t6.dtb in /boot/firmware/rk3588-nanopc-t6.dtb did not work. Additionally, placing it in /usr/lib/linux-image-5.10.160-rockchip/rockchip/rk3588-nanopc-t6.dtb also did not work. Device was restarted after each replacement.
[image]

Joshua-Riek, 2023-07-02:
I see, thank you for trying. I will reach out to FriendlyELEC for the hardware so I can do some testing.